elFarto/nvidia-vaapi-driver #267: DRM_IOCTL_NVIDIA_GEM_IMPORT_NVKMS_MEMORY failed

am-silex, 2024-01-16:
Hi
Sorry to bother you, just playing with your project and have this issue. My config:
- NVidia GT710 1Gb DDR3
- nvidia-driver-470
- Ubuntu 22.04.3 LTS
- Linux htpc 6.5.0-14-generic #14~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 20 18:15:30 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
- Firefox Nightly 123.0a1 (2024-01-15) (64-bit)

lshw -class display 

> *-display
>        description: VGA compatible controller
>        product: GK208B [GeForce GT 710]
>        vendor: NVIDIA Corporation
>        physical id: 0
>        bus info: pci@0000:01:00.0
>        version: a1
>        width: 64 bits
>        clock: 33MHz
>        capabilities: vga_controller bus_master cap_list rom
>        configuration: driver=nvidia latency=0
>        resources: irq:29 memory:fd000000-fdffffff memory:f0000000-f7ffffff memory:f8000000-f9ffffff ioport:cc00(size=128) memory:c0000-dffff

Firefox vaapi log

>        125.898066459 [4103-4117] ../src/vabackend.c:2154       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
>        125.898118959 [4103-4117] ../src/vabackend.c:2163       __vaDriverInit_1_0 Now have 0 (0 max) instances
>        125.898126764 [4103-4117] ../src/vabackend.c:2189       __vaDriverInit_1_0 Selecting Direct backend
>        125.931795047 [4103-4117] ../src/direct/nv-driver.c: 254            init_nvdriver Initing nvdriver...
>        125.931980530 [4103-4117] ../src/direct/nv-driver.c: 272            init_nvdriver NVIDIA kernel driver version: , major version: 0, minor version: 0
>        125.931993698 [4103-4117] ../src/direct/nv-driver.c: 279            init_nvdriver Got dev info: 100 1 0 fe
>        126.464563754 [4103-4117] ../src/vabackend.c:1412      nvQueryImageFormats In nvQueryImageFormats
>        126.610861410 [4103-4117] ../src/vabackend.c: 640           nvCreateConfig got profile: 6 with 0 attributes
>        126.610924314 [4103-4117] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) (nil) 0
>        126.610933870 [4103-4117] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) 0x7f43add2ba00 8
>        126.617297065 [4103-4117] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
>        126.618958447 [4103-4117] ../src/vabackend.c: 640           nvCreateConfig got profile: 7 with 0 attributes
>        126.619017522 [4103-4117] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) (nil) 0
>        126.619026936 [4103-4117] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) 0x7f43add2b340 8
>        126.628160487 [4103-4117] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
>        126.628215578 [4103-4117] ../src/vabackend.c: 640           nvCreateConfig got profile: 13 with 0 attributes
>        126.628230198 [4103-4117] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) (nil) 0
>        126.628238063 [4103-4117] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) 0x7f43add2b340 8
>        126.650278959 [4103-4117] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
>        126.844342943 [4103-4117] ../src/vabackend.c: 640           nvCreateConfig got profile: 7 with 0 attributes
>        126.844406136 [4103-4117] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) (nil) 0
>        126.844415850 [4103-4117] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) 0x7f43add2b4c0 8
>        126.850936680 [4103-4117] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
>        126.851002911 [4103-4117] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) (nil) 32579
>        126.851011912 [4103-4117] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) 0x7f43add2b340 8
>        126.857757841 [4103-4117] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
>        126.857839063 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5c480)
>        126.857850311 [4103-4117] ../src/vabackend.c:1506            nvDeriveImage In nvDeriveImage
>        126.857865545 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5cac0)
>        126.857875361 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5c5c0)
>        126.857885044 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5c840)
>        126.857895276 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5c8e0)
>        126.857920161 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5ca20)
>        126.857930156 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5cb60)
>        126.857939693 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5cca0)
>        126.857977585 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5d420)
>        126.857987460 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5d4c0)
>        126.857997339 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5d600)
>        126.858007432 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5da60)
>        126.858025142 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5d6a0)
>        126.858035171 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5d560)
>        126.858047690 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5d740)
>        126.858057490 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5d7e0)
>        126.858068367 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5d880)
>        126.858077803 [4103-4117] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 960x544, format 1 (0x7f43ade5d920)
[... 32 lines cut ...]
>        127.768467816 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5dec0)
>        127.768479379 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5d9c0)
>        127.768487871 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5de20)
>        127.768496005 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5d920)
>        127.768503919 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5d880)
>        127.768511821 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5d7e0)
>        127.768519986 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5d740)
>        127.768528031 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5d560)
>        127.768535952 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5d6a0)
>        127.768543675 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5da60)
>        127.768551146 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5d600)
>        127.768559014 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5d4c0)
>        127.768566962 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5d420)
>        127.768574568 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5cca0)
>        127.768582275 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5cb60)
[... cut ...]
>        127.768614462 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5c5c0)
>        127.768622320 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5cac0)
>        127.768630166 [4103-4117] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7f43ade5c480)
>        127.768642257 [4103-4117] ../src/vabackend.c:2064              nvTerminate Terminating 0x7f43da7dfb60
>        127.768946163 [4103-4117] ../src/vabackend.c:2078              nvTerminate Now have 0 (0 max) instances
> 
>

elFarto, 2024-01-21:
Was there anything logged to dmesg when that fails?

am-silex, 2024-01-22:
Unfortunately, I haven't captured it if there were any.

drygdryg, 2024-01-23:
I faced the same issue with GT 710 (1 GB, DDR3) on Arch Linux.
Graphics devices on my system:
```
# lshw -class display
  *-display                 
       description: VGA compatible controller
       product: GK208B [GeForce GT 710]
       vendor: NVIDIA Corporation
       physical id: 0
       bus info: pci@0000:03:00.0
       version: a1
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress vga_controller bus_master cap_list rom
       configuration: driver=nvidia latency=0
       resources: irq:49 memory:fa000000-faffffff memory:e8000000-efffffff memory:f0000000-f1ffffff ioport:e000(size=128) memory:fb000000-fb07ffff
  *-graphics
       product: EFI VGA
       physical id: 2
       logical name: /dev/fb0
       capabilities: fb
       configuration: depth=32 resolution=1920,1080
```
VA-API info:
```
$  NVD_LOG=1 vainfo
Trying display: wayland
Trying display: x11
libva error: vaGetDriverNames() failed with unknown libva error
      2112.181239814 [46312-46312] ../src/vabackend.c:2154       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      2112.181261395 [46312-46312] ../src/vabackend.c:2163       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2112.181268728 [46312-46312] ../src/vabackend.c:2189       __vaDriverInit_1_0 Selecting Direct backend
      2112.188908482 [46312-46312] ../src/direct/direct-export-buf.c:  66      direct_initExporter Searching for GPU: 0 0 128
      2112.188941796 [46312-46312] ../src/direct/direct-export-buf.c:  88      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
      2112.188950457 [46312-46312] ../src/direct/nv-driver.c: 254            init_nvdriver Initing nvdriver...
      2112.189014851 [46312-46312] ../src/direct/nv-driver.c: 272            init_nvdriver NVIDIA kernel driver version: , major version: 0, minor version: 0
      2112.189026794 [46312-46312] ../src/direct/nv-driver.c: 279            init_nvdriver Got dev info: 300 1 0 fe
vainfo: VA-API version: 1.20 (libva 2.20.1)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      2112.301236097 [46312-46312] ../src/vabackend.c:2064              nvTerminate Terminating 0x5fd132601660
      2112.301388423 [46312-46312] ../src/vabackend.c:2078              nvTerminate Now have 0 (0 max) instances
```

I tried to run mpv:
```
$ NVD_LOG=1 mpv --hwdec=vaapi test.mp4
 (+) Video --vid=1 (*) (h264 704x528 25.000fps)
 (+) Audio --aid=1 (*) (aac 2ch 48000Hz)
[vaapi] libva: vaGetDriverNames() failed with unknown libva error
      2373.426648459 [55115-55126] ../src/vabackend.c:2154       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      2373.426665640 [55115-55126] ../src/vabackend.c:2163       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2373.426675278 [55115-55126] ../src/vabackend.c:2189       __vaDriverInit_1_0 Selecting Direct backend
      2373.426692878 [55115-55126] ../src/direct/direct-export-buf.c:  66      direct_initExporter Searching for GPU: 0 0 128
      2373.426716345 [55115-55126] ../src/direct/direct-export-buf.c:  88      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
      2373.426722771 [55115-55126] ../src/direct/nv-driver.c: 254            init_nvdriver Initing nvdriver...
      2373.426767609 [55115-55126] ../src/direct/nv-driver.c: 272            init_nvdriver NVIDIA kernel driver version: , major version: 0, minor version: 0
      2373.426776968 [55115-55126] ../src/direct/nv-driver.c: 279            init_nvdriver Got dev info: 300 1 0 fe
      2373.487599443 [55115-55126] ../src/vabackend.c:1412      nvQueryImageFormats In nvQueryImageFormats
      2373.548550777 [55115-55126] ../src/vabackend.c: 640           nvCreateConfig got profile: 0 with 0 attributes
      2373.548568656 [55115-55126] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 1 (8) (nil) 0
      2373.548576130 [55115-55126] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 1 (8) 0x7b5494a50300 8
[... 293 lines cut ...]
[  +0,002803] [drm:nv_drm_gem_import_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000300] Failed to import NVKMS memory to GEM object
[  +0,000274] [drm:nv_drm_gem_import_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000300] Failed to import NVKMS memory to GEM object
[  +0,000267] [drm:nv_drm_gem_import_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000300] Failed to import NVKMS memory to GEM object
[  +0,000281] [drm:nv_drm_gem_import_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000300] Failed to import NVKMS memory to GEM object
[  +0,002804] [drm:nv_drm_gem_import_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000300] Failed to import NVKMS memory to GEM object
[  +0,000304] [drm:nv_drm_gem_import_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000300] Failed to import NVKMS memory to GEM object
[... cut ...]
[  +0,002814] [drm:nv_drm_gem_import_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000300] Failed to import NVKMS memory to GEM object
[  +0,000277] [drm:nv_drm_gem_import_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000300] Failed to import NVKMS memory to GEM object
[  +0,000297] [drm:nv_drm_gem_import_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000300] Failed to import NVKMS memory to GEM object
[  +0,000320] [drm:nv_drm_gem_import_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000300] Failed to import NVKMS memory to GEM object
```

am-silex, 2024-01-23:
Although, I haven't saved dmesg logs, I can confirm that error was the same 

> Failed to import NVKMS memory to GEM object

rkoot, 2024-01-24:
As far as I can tell the GT710 isn't supported by nvenc/nvdec.
(at least the matrix posted on the nvidia site doesn't mention the GT710)

am-silex, 2024-01-24:
NVENC: No Support
NVDEC: **1st Gen**
PureVideo HD: VP5
VDPAU: Feature Set D

https://www.techpowerup.com/gpu-specs/geforce-gt-710.c1990

rkoot, 2024-01-24:
https://developer.nvidia.com/video-encode-and-decode-gpu-support-matrix-new#geforce
doesn't mention it. I'd trust nvidia as a source more than a third party for this.

am-silex, 2024-01-24:
That page displays only currently supported GPUs
GT710 is outdated and only supported by 470 driver series

elFarto, 2024-01-27:
Does mpv work with `--hwdev=nvdec`?

am-silex, 2024-02-01:
When starting Firefox Nightly with all the parameters **dmesg** produces this

> [  185.750207] [drm:nv_drm_gem_import_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to import NVKMS memory to GEM object
> [  185.750633] [drm:nv_drm_gem_import_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to import NVKMS memory to GEM object
> [  185.753194] [drm:nv_drm_gem_import_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to import NVKMS memory to GEM object

am-silex, 2024-02-01:
Yes, mpv works as well as other apps utilizing vdpau and nvdec directly, like Kodi

> mpv --hwdec=nvdec 720.mp4
>  (+) Video --vid=1 (*) (h264 1280x720 25.000fps)
>  (+) Audio --aid=1 (*) (aac 2ch 48000Hz)

mirh, 2024-02-01:
Shouldn't this be titled "problems with driver 470" perhaps?
Does that 2021 driver still actually work with other cards and newer nvidia-vaapi?

elFarto, 2024-02-01:
Ok, can you try rerunning with `NVD_BACKEND=egl` set?

am-silex, 2024-02-02:
> Ok, can you try rerunning with `NVD_BACKEND=egl` set?

No errors and no hw offload, cpu utilization is as high as usual

am-silex, 2024-02-02:
> Shouldn't this be titled "problems with driver 470" perhaps? Does that 2021 driver still actually work with other cards and newer nvidia-vaapi?

Although, the root cause might be to the NVIDIA driver version, the driver itself works perfectly. All the other apps can offload supported video decoding through VDPAU API and NVDEC.

drygdryg, 2024-02-02:
> Does mpv work with --hwdev=nvdec

Yes, it works. NVENC is also working (limited by H.264), I tested NVENC in the NoMachine remote access app. The latest Linux driver version for GT 710 is 470.223.02.

am-silex, 2024-02-02:
It's interesting, this [specs ](https://www.techpowerup.com/gpu-specs/geforce-gt-710.c1990)claims that NVENC not supported on GK208B

mirh, 2024-02-02:
Just stop using those awful compilation websites
https://www.nvidia.com/en-us/geforce/forums/geforce-graphics-cards/5/280359/does-the-geforce-gt-710-have-nvenc/

With this said, I just installed 470.223.02 on my 2080S and yes I'm getting the same error OP has. 
Meanwhile the egl backend is giving me #218

mirh, 2024-02-02:
Actually this is a regression caused by bb7752d37412b05f0151e60904f5e743d08ec297 too

am-silex, 2024-02-02:
@mirh all right, such mashup sites are not reliable. 
However, NVENC support for the matter of this project is irrelevant, I believe. Main use case implementing VA-API is to comfortably watch online videos in Firefox.

elFarto, 2024-02-03:
Ok, found the issue. It seems the 470 series has changed how it returns the version number. I've push a fix for it to master.

am-silex, 2024-02-03:
Should I worry about this?

> [   41.790153] NVRM: API mismatch: the client has the version , but
>                NVRM: this kernel module has the version 470.223.02.  Please
>                NVRM: make sure that this kernel module and all NVIDIA driver
>                NVRM: components have the same version.

am-silex, 2024-02-03:
And I can't confirm that it works. Here is a comparison in CPU usage while playing the same sample in h264 format
| mpv| Firefox Nightly| Firefox (snap)|
|--------|--------|--------|
| ~5%| ~15% | ~20% |

Firefox Nightly uses slightly less in CPU in other scenarios (YouTube with h264ify, other streaming web sites)

elFarto, 2024-02-03:
Hmm, the error isn't really a problem, it's a result of me having to call the check version function in a different way. I don't think I'll be able to work around it on the 470 drivers, but I should be able to do on the 500+ series.

To check if the GPU is being used for decode, use something like `nvidia-smi pmon`, and look in the `dec` column.

am-silex, 2024-02-04:
With you last commit I can confirm that the driver works, despite dmesg log.
Even Kodi now switched to `ff-h264-vaapi (HW)` when playing videos or watching DVB channels, CPU utilization is the same as with `vdpau`.
However, FIrefox uses a higher percentage playing the same sample. I reckon it is due to some overhead in its internal implementation and has nothing to do with the driver. When starting playing something, dmesg spits this info, too.
`nvidia-smi pmon` isn't supported on GT710

> Not supported on the device(s)
> Failed to process command line

Thank you for you work!

drygdryg, 2024-02-05:
I also confirm that VA-API is now functional on my GT 710 on the latest (master) driver version built with [nv-codec-headers](https://github.com/FFmpeg/nv-codec-headers) v11.1.
I check the GPU usage using NVIDIA's nvidia-settings GUI tool and get about the same video engine utilization when using both NVDEC and the VA-API, when playing a H.264 video using the mpv player.
How did I install the driver with the right nv-codec-headers version on Arch Linux:
```
paru -S aur/ffnvcodec-headers-11-1 aur/libva-nvidia-driver-git
```

Thank you for your hard work!

closed 2024-02-10 by am-silex