elFarto/nvidia-vaapi-driver #268: Ubuntu 22.04 - ERROR: Build directory has been generated with Meson version 1.1.1, which is incompatible with the current version 0.61.2.

sgofferj, 2024-01-19:
I'm trying to build on Ubuntu 22.04 but I can't get it to work. I have never even heard of meson, so unfortunately, I'm at a brick wall now. Any help would be greatly appreciated!

Here's some output:
```
┌─[sgofferj@enterprise][~/source/nvidia-vaapi-driver]
└──▪ meson setup build
The Meson build system
Version: 1.1.1
Source dir: /home/sgofferj/source/nvidia-vaapi-driver
Build dir: /home/sgofferj/source/nvidia-vaapi-driver/build
Build type: native build
WARNING: failed to process netrc file: ~/.netrc access too permissive: access permissions must restrict access to only the owner (/home/sgofferj/.netrc, line 3).
Project name: nvidia-vaapi-driver
Project version: 0.1
C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C linker for the host machine: cc ld.bfd 2.38
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library m found: YES
Library dl found: YES
Found pkg-config: /usr/bin/pkg-config (0.29.2)
Run-time dependency egl found: YES 1.5
Run-time dependency ffnvcodec found: YES 11.1.5.1
Run-time dependency libdrm found: YES 2.4.113
Run-time dependency threads found: YES
Run-time dependency libva found: YES 1.14.0
Run-time dependency gstreamer-codecparsers-1.0 found: YES 1.20.3
Compiler for C supports arguments -Wno-missing-field-initializers: YES 
Compiler for C supports arguments -Wno-unused-parameter: YES 
Compiler for C supports arguments -Werror=format: YES 
Compiler for C supports arguments -Werror=format-security: YES 
Compiler for C supports arguments -Werror=incompatible-pointer-types: YES 
Compiler for C supports arguments -Werror=init-self: YES 
Compiler for C supports arguments -Werror=int-conversion: YES 
Compiler for C supports arguments -Werror=missing-declarations: YES 
Compiler for C supports arguments -Werror=missing-prototypes: YES 
Compiler for C supports arguments -Werror=pointer-arith: YES 
Compiler for C supports arguments -Werror=undef: YES 
Compiler for C supports arguments -Werror=vla: YES 
Compiler for C supports arguments -Wsuggest-attribute=format: YES 
Compiler for C supports arguments -Wwrite-strings: YES 
Build targets in project: 1

Found ninja-1.10.1 at /usr/bin/ninja

┌─[sgofferj@enterprise][~/source/nvidia-vaapi-driver]                                               
└──▪ meson install -C build
ninja: Entering directory `/home/sgofferj/source/nvidia-vaapi-driver/build'
[16/16] Linking target nvidia_drv_video.so
Installing nvidia_drv_video.so to /usr/lib/x86_64-linux-gnu/dri
Installation failed due to insufficient permissions.
Attempt to use /usr/bin/sudo to gain elevated privileges? [y/n] y

ERROR: Build directory has been generated with Meson version 1.1.1, which is incompatible with the current version 0.61.2.
```

drhashes, 2024-01-20:
1. The Meson build system
Version: 1.1.1

Here is one problem. You have two versions of meson installed.
I bet if you issue the command: meson --version
you get:
0.61.2

but the build configure script found version 1.1.1 in the end.

2. WARNING: failed to process netrc file: ~/.netrc access too permissive: access permissions must restrict access to only the owner (/home/sgofferj/.netrc, line 3).

Not sure why you have .netrc file but it is confusing the build it seems.
Why not temporary rename that: mv .netrc .netrc-
Also check what is wrong with it on line 3
+ ls -l .netrc to see permissions.

elFarto, 2024-01-21:
Actually that build completed successfully it seems, only failing when switching to root to do the install. You can check if it actually got installed by doing: `ls -l /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so`.

It does seem like you have 2 different versions of meson installed, and when switching to root you're picking up the old one (both versions are new enough to build this project).

sgofferj, 2024-01-22:
Thank you. It didn't get installed but built, so I cp'd the .so by hand and it works.
I have not the slightest idea how I could have 2 versions of meson installed though. I don't recall having ever used it before...

closed 2024-01-22 by sgofferj