elFarto/nvidia-vaapi-driver #271: `eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: could not bind to DMA buffer` after commit f9777663a9519161ad3320819170e039fe909a4f

thesword53, 2024-01-30:
Commit f9777663a9519161ad3320819170e039fe909a4f breaks video decoding.

```
 (+) Video --vid=1 (*) (h264 1280x720 30.151fps)
 (+) Audio --aid=1 --alang=eng (*) (aac 2ch 44100Hz)
      4869.341313916 [121133-121144] ../src/vabackend.c:2154       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
      4869.341324726 [121133-121144] ../src/vabackend.c:2163       __vaDriverInit_1_0 Now have 0 (0 max) instances
      4869.341327237 [121133-121144] ../src/vabackend.c:2189       __vaDriverInit_1_0 Selecting Direct backend
      4869.341341667 [121133-121144] ../src/direct/nv-driver.c: 254            init_nvdriver Initing nvdriver...
      4869.341370617 [121133-121144] ../src/direct/nv-driver.c: 272            init_nvdriver NVIDIA kernel driver version: 545.29.06, major version: 545, minor version: 29
      4869.341375247 [121133-121144] ../src/direct/nv-driver.c: 279            init_nvdriver Got dev info: 2d00 1 2 6
      4869.428608152 [121133-121144] ../src/vabackend.c:1412      nvQueryImageFormats In nvQueryImageFormats
      4869.458858196 [121133-121144] ../src/vabackend.c: 640           nvCreateConfig got profile: 0 with 0 attributes
      4869.458867066 [121133-121144] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 1 (8) (nil) 0
      4869.458870246 [121133-121144] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 1 (8) 0x7e0451705b80 8
      4869.460201763 [121133-121144] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4080, height: 16 - 4080
      4869.460212653 [121133-121144] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7e0451705480)
      4869.460216233 [121133-121144] ../src/vabackend.c:1506            nvDeriveImage In nvDeriveImage
      4869.460220863 [121133-121144] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7e0451702d00)
      4869.460224653 [121133-121144] ../src/vabackend.c:1506            nvDeriveImage In nvDeriveImage
      4869.460259863 [121133-121144] ../src/direct/direct-export-buf.c: 149 direct_allocateBackingImage Allocating BackingImage: 0x7e0451551c60 128x128 = 24576 bytes
      4869.460520827 [121133-121144] ../src/direct/direct-export-buf.c: 158 direct_allocateBackingImage Allocate Buffer: 64 65 66
      4869.460525277 [121133-121144] ../src/direct/direct-export-buf.c: 168 direct_allocateBackingImage Importing memory to CUDA
      4869.461019103 [121133-121144] ../src/direct/direct-export-buf.c:  44                    debug [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: could not bind to DMA buffer

      4869.461055683 [121133-121144] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7e0451702d00)
      4869.461209145 [121133-121144] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 128x128, format 100 (0x7e0451706940)
      4869.461214265 [121133-121144] ../src/vabackend.c:1506            nvDeriveImage In nvDeriveImage
      4869.461227025 [121133-121144] ../src/direct/direct-export-buf.c: 149 direct_allocateBackingImage Allocating BackingImage: 0x7e0451702ea0 128x128 = 49152 bytes
      4869.461439208 [121133-121144] ../src/direct/direct-export-buf.c: 158 direct_allocateBackingImage Allocate Buffer: 64 65 66
      4869.461443648 [121133-121144] ../src/direct/direct-export-buf.c: 168 direct_allocateBackingImage Importing memory to CUDA
      4869.461833243 [121133-121144] ../src/direct/direct-export-buf.c:  44                    debug [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: could not bind to DMA buffer

      4869.461861623 [121133-121144] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7e0451706940)
      4869.461977095 [121133-121144] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 128x128, format 1000 (0x7e0451706ba0)
      4869.461982105 [121133-121144] ../src/vabackend.c:1506            nvDeriveImage In nvDeriveImage
      4869.461995245 [121133-121144] ../src/direct/direct-export-buf.c: 149 direct_allocateBackingImage Allocating BackingImage: 0x7e0451702bc0 128x128 = 49152 bytes
      4869.462153297 [121133-121144] ../src/direct/direct-export-buf.c: 158 direct_allocateBackingImage Allocate Buffer: 64 65 66
      4869.462157437 [121133-121144] ../src/direct/direct-export-buf.c: 168 direct_allocateBackingImage Importing memory to CUDA
      4869.462542262 [121133-121144] ../src/direct/direct-export-buf.c:  44                    debug [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: could not bind to DMA buffer

      4869.462570122 [121133-121144] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7e0451706ba0)
      4869.462688814 [121133-121144] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 128x128, format 4 (0x7e0451554160)
      4869.462693914 [121133-121144] ../src/vabackend.c:1506            nvDeriveImage In nvDeriveImage
      4869.462707194 [121133-121144] ../src/direct/direct-export-buf.c: 149 direct_allocateBackingImage Allocating BackingImage: 0x7e0451702ca0 128x128 = 49152 bytes
      4869.462873086 [121133-121144] ../src/direct/direct-export-buf.c: 158 direct_allocateBackingImage Allocate Buffer: 64 65 66
      4869.462877406 [121133-121144] ../src/direct/direct-export-buf.c: 168 direct_allocateBackingImage Importing memory to CUDA
      4869.463720437 [121133-121144] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7e0451554160)
      4869.463867139 [121133-121144] ../src/vabackend.c: 957        nvDestroySurfaces Destroying surface -1 (0x7e0451705480)
      4869.463872549 [121133-121144] ../src/vabackend.c: 640           nvCreateConfig got profile: -1 with 0 attributes
      4869.463875499 [121133-121144] ../src/vabackend.c: 645           nvCreateConfig Profile not supported: -1
      4869.496626334 [121133-121133] ../src/vabackend.c: 640           nvCreateConfig got profile: 6 with 0 attributes
      4869.496635654 [121133-121133] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      4869.496638544 [121133-121133] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) 0x5b9a20e98940 8
      4869.497965401 [121133-121133] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      4869.497973421 [121133-121133] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) (nil) 23450
      4869.497976101 [121133-121133] ../src/vabackend.c:1768 nvQuerySurfaceAttributes with 4 (8) 0x5b9a20e98940 8
      4869.499299108 [121133-121133] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      4869.499314858 [121133-121133] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x5b9a20e98ab0)
      4869.499318698 [121133-121133] ../src/vabackend.c:1506            nvDeriveImage In nvDeriveImage
      4869.499323728 [121133-121133] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x5b9a20a802e0)
      4869.499327738 [121133-121133] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x5b9a20a806e0)
      4869.499330998 [121133-121133] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x5b9a20a80ae0)
      4869.499334588 [121133-121133] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x5b9a20a80ee0)
      4869.499338858 [121133-121133] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x5b9a20a812e0)
      4869.499342398 [121133-121133] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x5b9a20a816e0)
      4869.499345858 [121133-121133] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x5b9a20a81ae0)
      4869.499349228 [121133-121133] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x5b9a20a81ee0)
      4869.499353058 [121133-121133] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x5b9a20a822e0)
      4869.499356598 [121133-121133] ../src/vabackend.c: 923        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x5b9a20a826e0)
[... 41 lines cut ...]
[... cut ...]
Exiting... (End of file)
AV: 00:00:06 / 00:00:06 (99%) A-V:  0.000      4871.593880360 [121133-121144] ../src/vabackend.c:2064              nvTerminate Terminating 0x7e045068a3d0
      4871.594019392 [121133-121144] ../src/vabackend.c:2078              nvTerminate Now have 0 (0 max) instances
      4871.631229264 [121133-121144] ../src/direct/direct-export-buf.c:  44                    debug [EGL] eglDestroySurface: EGL_BAD_SURFACE error: In internal function: Additional INFO may be available
```

mirh, 2024-02-01:
I can confirm reverting that fixes decoding.

elFarto, 2024-02-01:
Can you be a bit more precise? I've been using that code for months and hardware decoding still works for me.

mirh, 2024-02-01:
You have been using code that is 39 days old.. _for months_?
Anyway, I'm running 545.29.06 on X11 and I used [this video](https://drive.google.com/file/d/0BwxFVkl63-lEY3l3ODJReDg3RzQ/view).

thesword53, 2024-02-04:
> You have been using code that is 39 days old.. _for months_? Anyway, I'm running 545.29.06 on X11 and I used [this video](https://drive.google.com/file/d/0BwxFVkl63-lEY3l3ODJReDg3RzQ/view).

Same driver but on Wayland (KDE)

elFarto, 2024-05-08:
This issue should be fixed in master now.

thesword53, 2024-05-11:
I confirm the issue is fixed in git master but the version 0.0.12 is still affected.

thesword53, 2024-11-13:
Closing as the issue is fixed with the 0.0.13 version.

closed 2024-11-13 by thesword53