elFarto/nvidia-vaapi-driver #272: libva error: /usr/lib/dri/nvidia_drv_video.so init failed on Intel + Nvidia Wayland System

AniAggarwal, 2024-02-08:
Followed #160 who had a similar issue, but didn't work.
Below is all the info I could think of, lmk whatever else you think would help.
Thanks again for the help!

<details>
<summary> System info </summary>
Arch Linux

```shell
$ uname -a
Linux arrakis 6.6.16-1-lts #1 SMP PREEMPT_DYNAMIC Mon, 05 Feb 2024 21:20:21 +0000 x86_64 GNU/Linux
$ sudo lshw
arrakis
    description: Notebook
    product: XPS 15 9520 (0B19)
    vendor: Dell Inc.
    serial: JYZFJR3
    width: 64 bits
    capabilities: smbios-3.4.0 dmi-3.4.0 smp vsyscall32
    configuration: boot=normal chassis=notebook family=XPS sku=0B19 uuid=4c4c4544-0059-5a10-8046-cac04f4a5233
  *-core
       description: Motherboard
       product: 0MWGD4
       vendor: Dell Inc.
       physical id: 0
       version: A00
       serial: /JYZFJR3/CNCMK0027N0286/
     *-firmware
          description: BIOS
          vendor: Dell Inc.
          physical id: 1
          version: 1.20.0
          date: 12/19/2023
          size: 1MiB
          capacity: 32MiB
          capabilities: pci pnp upgrade shadowing cdboot bootselect edd int5printscreen int9keyboard int14serial int17printer acpi usb smartbattery biosbootspecification netboot uefi
     *-cpu
          description: CPU
          product: 12th Gen Intel(R) Core(TM) i7-12700H
          vendor: Intel Corp.
          physical id: 400
          bus info: cpu@0
          version: 6.154.3
          slot: U3E1
          size: 1731MHz
          capacity: 4600MHz
          width: 64 bits
          clock: 100MHz
          capabilities: lm fpu fpu_exception wp vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp x86-64 constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdseed adx smap clflushopt clwb intel_pt sha_ni xsaveopt xsavec xgetbv1 xsaves split_lock_detect user_shstk avx_vnni dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req hfi vnmi umip pku ospke waitpkg gfni vaes vpclmulqdq rdpid movdiri movdir64b fsrm md_clear serialize arch_lbr ibt flush_l1d arch_capabilities cpufreq
          configuration: cores=14 enabledcores=14 microcode=1074 threads=20
        *-cache:0
             description: L1 cache
             physical id: 705
             slot: L1 Cache
             size: 512KiB
             capacity: 512KiB
             capabilities: synchronous internal write-back instruction
             configuration: level=1
        *-cache:1
             description: L2 cache
             physical id: 706
             slot: L2 Cache
             size: 4MiB
             capacity: 4MiB
             capabilities: synchronous internal write-back unified
             configuration: level=2
        *-cache:2
             description: L3 cache
             physical id: 707
             slot: L3 Cache
[... 1321 lines cut ...]
mesa
mesa-utils
$ pacman -Qq | rg -i vdpau
libvdpau
vdpauinfo
```

```shell
$ cat /sys/module/nvidia_drm/parameters/modeset
Y
```

Some env vars in Hyprland:
```
env = XCURSOR_SIZE,24
env = LIBVA_DRIVER_NAME,nvidia
env = XDG_SESSION_TYPE,wayland
env = GBM_BACKEND,nvidia-drm
env = __GLX_VENDOR_LIBRARY_NAME,nvidia
env = WLR_NO_HARDWARE_CURSORS,1

# prioritize using intergrated GPU over dedicated Nvidia GPU
# doesn't seem to make a difference whether I use this or not
env = WLR_DRM_DEVICES,/dev/dri/card1:/dev/dri/card0
```

elFarto, 2024-02-11:
The issue you're having here is that Wayland is running on the Intel card. When libva asks it for the default device, it will return the Intel card. You would need to run either just Firefox on the NVIDIA card, or everything on the NVIDIA card.

AniAggarwal, 2024-02-11:
Huh, yeah that seemed to work, by doing `prime-run firefox` (or at least I can see my GPU being used while playing 4k video).
I don't have a great understanding of how this works, but shouldn't this driver allow me to run firefox on the intel card and have nvidia be the back end for vaapi?
Or does it not work like that?

I'd like to avoid running as much as possible on the Nvidia card for battery purposes. 

Also, maybe this is the wrong place to ask, but I can't seem to get hardware acceleration to run on brave, my primary browser.

byomi, 2024-03-31:
> The issue you're having here is that Wayland is running on the Intel card. When libva asks it for the default device, it will return the Intel card. You would need to run either just Firefox on the NVIDIA card, or everything on the NVIDIA card.

so u mean i need to remove these 
```
export XCURSOR_SIZE=24             
export LIBVA_DRIVER_NAME=nvidia
export XDG_SESSION_TYPE=wayland
export GBM_BACKEND=nvidia-drm
export __GLX_VENDOR_LIBRARY_NAME=nvidia
export WLR_NO_HARDWARE_CURSORS=1
```
and run hyprland on the intel but export them and then can run like firefox with prime-run ???? did u mean that @elFarto

byomi, 2024-03-31:
``Huh, yeah that seemed to work, by doing "prime-run firefox" (or at least I can see my GPU being used while playing 4k video). ``

@AniAggarwal  how did u managed it to work

AniAggarwal, 2024-03-31:
@byomi Here's my relevant Hyprland vars. I didn't change anything I just do prime-run firefox and it works. Doesn't work on brave or other chromium based browers, read somewhere that it hasn't been patched in or something?
```
# VARS
env = XCURSOR_SIZE,24
# Nvidia vars
env = LIBVA_DRIVER_NAME,nvidia
env = XDG_SESSION_TYPE,wayland
env = GBM_BACKEND,nvidia-drm
env = __GLX_VENDOR_LIBRARY_NAME,nvidia
env = WLR_NO_HARDWARE_CURSORS,1

# reddit dude said to add: https://www.reddit.com/r/hyprland/comments/188a8fr/people_with_nvidia_cards_please_share_your_secret/?rdt=35012
env = DRI_PRIME,pci-0000_01_00_0
env = __VK_LAYER_NV_optimus,NVIDIA_only 
```

byomi, 2024-03-31:
> @byomi Here's my relevant Hyprland vars. I didn't change anything I just do prime-run firefox and it works. Doesn't work on brave or other chromium based browers, read somewhere that it hasn't been [...]
i tried these alot but didnt work

but this down here is new but after thinking alot i conclude it wont work at all.
```
> # reddit dude said to add: https://www.reddit.com/r/hyprland/comments/188a8fr/people_with_nvidia_cards_please_share_your_secret/?rdt=35012
> env = DRI_PRIME,pci-0000_01_00_0
> env = __VK_LAYER_NV_optimus,NVIDIA_only  ```

here why i have gt 740m which uses nvidia-470xx-utils and dkms which is old and wayland need nvidia-495 as minimum so after some even work whatever i do it wont work correctly it will work with x11 apps under xwayland but wayland never will work but the problem is since its not supported with wayland the xwayland flickers alot and got black under hypland and buggy and fuzzy idk how if there a fix or no so i gave up .

but i do like actually hyprland is the best but i am looking to change to awesome wm or qtile since they use xorg would u recommend any of them??

AniAggarwal, 2024-04-29:
Not sure, I don't use them. I just turned off all nvidia related hyprland configs and am just using my igpu now. Everything works fine (except chromium browsers) and I am also able to run firefox with nvidia-vaapi driver with the following command:
`NVD_BACKEND=direct MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia prime-run firefox`.
I also am able to use my dgpu for ML, so I'm more or less satisfied except for the chromium browser issue.

closed 2024-04-29 by AniAggarwal

littleblack111, 2024-09-27:
still like that with the env, my DP is plugged into my GPU.
```
NVD_LOG=1 NVD_BACKEND=direct MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia prime-run vainfo --display drm
```:
```
Trying display: drm
    177317.435379065 [1922091-1922091] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
    177317.435384313 [1922091-1922091] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
    177317.435386073 [1922091-1922091] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
    177317.451010050 [1922091-1922091] ../nvidia-vaapi-driver-0.0.12/src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
    177317.451016700 [1922091-1922091] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2248       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```
but without --display drm seems to work:
```
Trying display: wayland
    177397.617758724 [1923376-1923376] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
    177397.617764154 [1923376-1923376] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
    177397.617766411 [1923376-1923376] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
    177397.632961159 [1923376-1923376] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
    177397.632975226 [1923376-1923376] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 560.35.03, major version: 560, minor version: 35
    177397.632977420 [1923376-1923376] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 1 2 6
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
    177397.804356632 [1923376-1923376] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2098              nvTerminate Terminating 0x5a17cf24efb0
    177397.804832765 [1923376-1923376] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances
```

but most electron apps(discord, vscode, figma etc) complain about gpu child process exiting and libva error: /usr/lib/dri/nvidia_drv_video.so init failed

AmmoniumX, 2024-10-03:
I'm having the opposite problem of the comment above me, it works when I set `--display drm` but not without it:
```
❯ NVD_LOG=1 vainfo
Trying display: wayland
      1593.888263084 [4040-4040] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
      1593.888278960 [4040-4040] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1593.888285705 [4040-4040] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
      1593.953825279 [4040-4040] ../nvidia-vaapi-driver-0.0.12/src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
      1593.953859338 [4040-4040] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2248       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit

❯ NVD_LOG=1 vainfo --display drm
Trying display: drm
      1598.778394119 [4070-4070] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      1598.778406872 [4070-4070] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1598.778433319 [4070-4070] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
      1598.830344417 [4070-4070] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
      1598.830419022 [4070-4070] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 560.35.03, major version: 560, minor version: 35
      1598.830437062 [4070-4070] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 1 0 fe
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      1598.971425687 [4070-4070] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2098              nvTerminate Terminating 0x558bf00a8a30
      1598.971590130 [4070-4070] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances
```

I have tried to mess with multiple different env vars such as `VAAPI_DEVICE` to try and make the default DRM device work correctly but none work. I have made a separate issue on https://github.com/elFarto/nvidia-vaapi-driver/issues/330

littleblack111, 2024-10-04:
ngl. i dunno why does it make a diff becuz i have env = GBM_BACKEND,nvidia-drm set

Shinigami-98, 2025-05-03:
LIBVA_DRIVER_NAME=nvidia vainfo --display drm --device /dev/dri/renderD129

This worked for me