elFarto/nvidia-vaapi-driver #274: An issue may solve the situation to use this library on Chrome/Chromium.

LeviMarvin, 2024-02-10:
Environment:
CPU: Intel Core i7 13700H
GPU (In about system): Mesa Intel(R) Graphics (On-Demand)
GPU1: Intel Alderlake_p (Gen12) (RPL-P) [Intel Iris Xe Graphics]
GPU2: NVIDIA GeForce RTX 4060 Laptop 8GB

Desktop: Debian Testing + KDE + X11

Run Chrome with the command:
```bash
NVD_LOG=1 VK_DRIVER_FILES=/usr/share/vulkan/icd.d/nvidia_icd.json LIBVA_DRIVER_NAME=nvidia google-chrome --use-angle=gl --use-gl=angle --ignore-gpu-blocklist --enable-features=VaapiVideoDecodeLinuxGL,VaapiVideoEncoder,VaapiOnNvidiaGPUs  --disable-features=UseChromeOSDirectVideoDecoder --disable-gpu-driver-bug-workarounds --enable-gpu-rasterization --enable-zero-copy
```

And I modified the function so that the library could run on multi-GPU mode:
```c++
static bool direct_initExporter(NVDriver *drv) {
    //this is only needed to see errors in firefox
    static const EGLAttrib debugAttribs[] = {EGL_DEBUG_MSG_WARN_KHR, EGL_TRUE, EGL_DEBUG_MSG_INFO_KHR, EGL_TRUE, EGL_NONE};
    const PFNEGLDEBUGMESSAGECONTROLKHRPROC eglDebugMessageControlKHR = (PFNEGLDEBUGMESSAGECONTROLKHRPROC) eglGetProcAddress("eglDebugMessageControlKHR");
    eglDebugMessageControlKHR(debug, debugAttribs);

    //make sure we have a drm fd
    if (drv->drmFd == -1) {
        LOG("We have NOT a drm fd.");
test:
        int nvdGpu = drv->cudaGpuId;
        if (nvdGpu == -1) {
            // The default GPU is the first one we find.
            nvdGpu = 0;
        }

        int fd;
        int nvIdx = 0;
        uint8_t drmIdx = 128;
        char node[20] = {0, };
        do {
            LOG("Searching for GPU: %d %d %d", nvIdx, nvdGpu, drmIdx)
            snprintf(node, 20, "/dev/dri/renderD%d", drmIdx++);
            fd = open(node, O_RDWR|O_CLOEXEC);
            if (fd == -1) {
                LOG("Unable to find NVIDIA GPU %d", nvdGpu);
                return false;
            }

            if (!isNvidiaDrmFd(fd, true) || !checkModesetParameterFromFd(fd)) {
                close(fd);
                continue;
            }

            if (nvIdx != nvdGpu) {
                close(fd);
                nvIdx++;
                continue;
            }
            break;
        } while (drmIdx < 128 + 16);

        drv->drmFd = fd;
        LOG("Found NVIDIA GPU %d at %s", nvdGpu, node);
    } else {
        LOG("We have a drm fd.");
        bool isNvDrmFd = isNvidiaDrmFd(drv->drmFd, true);
        bool resultCheckModesetParameterFromFd = checkModesetParameterFromFd(drv->drmFd);
        LOG("drv->drmFd: %d, isNvDrmFd: %d", drv->drmFd, isNvDrmFd);
        LOG("resultCheckModesetParameterFromFd: %d", resultCheckModesetParameterFromFd);

        if (!isNvDrmFd) {
            goto test;
        }

[... 112 lines cut ...]
    168268.525660000 [2006932-2007376] ../src/direct/direct-export-buf.c:  95      direct_initExporter drv->drmFd: 26, isNvDrmFd: 0
    168268.525660922 [2006932-2007376] ../src/direct/direct-export-buf.c:  96      direct_initExporter resultCheckModesetParameterFromFd: 1
    168268.525662270 [2006932-2007376] ../src/direct/direct-export-buf.c:  68      direct_initExporter Searching for GPU: 0 0 128
    168268.525710241 [2006932-2007376] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
    168268.525717375 [2006932-2007376] ../src/direct/direct-export-buf.c:  68      direct_initExporter Searching for GPU: 0 0 129
    168268.525723788 [2006932-2007376] ../src/direct/direct-export-buf.c:  90      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD129
    168268.525725342 [2006932-2007376] ../src/direct/nv-driver.c: 259            init_nvdriver Initing nvdriver...
    168268.525748307 [2006932-2007376] ../src/direct/nv-driver.c: 277            init_nvdriver NVIDIA kernel driver version: 545.23.08, major version: 545, minor version: 23
    168268.525751317 [2006932-2007376] ../src/direct/nv-driver.c: 284            init_nvdriver Got dev info: 100 1 2 6
    168268.577058173 [2006932-2007376] ../src/vabackend.c: 640           nvCreateConfig got profile: 7 with 1 attributes
    168268.577064032 [2006932-2007376] ../src/vabackend.c: 661           nvCreateConfig got config attrib: 0 0 1
    168269.070782101 [2006932-2007376] ../src/vabackend.c: 987          nvCreateContext creating context with 0 render targets, 0 surfaces, at 1920x1088
    168269.070789324 [2006932-2007376] ../src/vabackend.c:1013          nvCreateContext 0 surfaces have been passed to vaCreateContext, this might cause errors. Setting surface count to 32
    168269.079918456 [2006932-2007380] ../src/vabackend.c: 382          resolveSurfaces [RT] Resolve thread for 0xa2c039a5000 started
[2006932:2007376:0210/185127.505732:ERROR:vaapi_video_decoder.cc(1249)] failed Initialize()ing the frame pool
    168269.081882887 [2006932-2007376] ../src/vabackend.c:1087         nvDestroyContext Destroying context: 2
    168269.081886693 [2006932-2007376] ../src/vabackend.c: 291           destroyContext Signaling resolve thread to exit
    168269.081888225 [2006932-2007376] ../src/vabackend.c: 297           destroyContext Waiting for resolve thread to exit
    168269.081920533 [2006932-2007380] ../src/vabackend.c: 424          resolveSurfaces [RT] Resolve thread for 0xa2c039a5000 exiting
    168269.081943157 [2006932-2007376] ../src/vabackend.c: 299           destroyContext pthread_timedjoin_np finished with 0
    168269.087961603 [2006932-2007376] ../src/vabackend.c:2064              nvTerminate Terminating 0xa2c02f5ae00
    168269.088023564 [2006932-2007376] ../src/vabackend.c:2078              nvTerminate Now have 0 (0 max) instances
```

And I think it can be used for Chrome/Chromium if we fixed the error?

elFarto, 2024-02-11:
Unfortunately I don't think this will work. While it may appear to be initialising the correct device, Chrome doesn't know we're using a different device to the one it's using. When we go to export the DMA-BUF, Chrome will attempt to import it into the EGL/Vulkan context on the Intel card which will fail.

LeviMarvin, 2024-02-11:
We have solution to make sure the Chrome using NVIDIA Vulcan, and when I did it, NVIDIA GPU Utilization is float (means the NVIDIA GPU were used). But the video engine still keeps frozen 0%

shelterx, 2024-02-14:
> And there is an error from Chrome:
> 
> ```
> [2006932:2007376:0210/185127.505732:ERROR:vaapi_video_decoder.cc(1249)] failed Initialize()ing the frame pool
> ```

This is currently affecting Chrome with Nvidia and it's been active for a while. Depending on what flags you run I also got this now which I haven't seen before:
`[65661:65661:0214/084514.593507:WARNING:vaapi_wrapper.cc(1489)] : Skipping nVidia device named: nvidia-drm`

lafoletc, 2024-02-29:
> This is currently affecting Chrome with Nvidia and it's been active for a while. Depending on what flags you run I also got this now which I haven't seen before: `[65661:65661:0214/084514.593507:WARNING:vaapi_wrapper.cc(1489)] : Skipping nVidia device named: nvidia-drm`

To remove the Nvidia skipping, You need --enable-features=VaapiOnNvidiaGpus

lafoletc, 2024-02-29:
how have you removed the following error ?

> > And there is an error from Chrome:
> > ```
> > [2006932:2007376:0210/185127.505732:ERROR:vaapi_video_decoder.cc(1249)] failed Initialize()ing the frame pool
> > ```
>

ManuLinares, 2024-05-13:
Have any tests been conducted with the latest patch? I haven't achieved any success yet.

wugaosheng123, 2024-05-30:
Can this driver be used on top of chromium.

closed 2025-08-07 by elFarto