elFarto/nvidia-vaapi-driver #276: Fail to get NVidia driver version since V0.0.10 - working anyway up to V0.0.11 - master fails - NVidia V525.147.05

Jasmin68k, 2024-02-13:
Hello!

Using the driver on Debian Sid with NVidia drivers V525.147.05, NVidia driver version is not properly detected since V0.0.10 as reported by vainfo.

Hardware decoding in Firefox _works anyway_, **except for latest `master` branch**, where `nv-driver.c` forces a dummy version of 470.123.45, when no proper NVidia driver version is returned.

Since `nv-driver.c` sets different parameters depending on detected driver version, it might not work optimally though.

Let me know, if you need any further info.

```
master

vainfo --display drm
      3861.555803320 [11638-11638] ../src/vabackend.c:2163       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      3861.555822223 [11638-11638] ../src/vabackend.c:2172       __vaDriverInit_1_0 Now have 0 (100 max) instances
      3861.555826865 [11638-11638] ../src/vabackend.c:2198       __vaDriverInit_1_0 Selecting Direct backend
      3861.561292937 [11638-11638] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
      3861.561318965 [11638-11638] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 470.123.45, major version: 470, minor version: 123
      3861.561322774 [11638-11638] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 1 0 fe
vainfo: VA-API version: 1.20 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :   VAEntrypointVLD
      VAProfileMPEG2Main              :   VAEntrypointVLD
      VAProfileVC1Simple              :   VAEntrypointVLD
      VAProfileVC1Main                :   VAEntrypointVLD
      VAProfileVC1Advanced            :   VAEntrypointVLD
      VAProfileH264Main               :   VAEntrypointVLD
      VAProfileH264High               :   VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:   VAEntrypointVLD
      VAProfileHEVCMain               :   VAEntrypointVLD
      VAProfileVP9Profile0            :   VAEntrypointVLD
      VAProfileHEVCMain10             :   VAEntrypointVLD
      VAProfileHEVCMain12             :   VAEntrypointVLD
      VAProfileVP9Profile2            :   VAEntrypointVLD
      3861.609695301 [11638-11638] ../src/vabackend.c:2073              nvTerminate Terminating 0x561abc2e44d0
      3861.609723190 [11638-11638] ../src/vabackend.c:2087              nvTerminate Now have 0 (100 max) instances

0.0.11

vainfo --display drm
      1724.182846035 [9012-9012] ../src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      1724.182853024 [9012-9012] ../src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (100 max) instances
      1724.182854389 [9012-9012] ../src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
      1724.185516520 [9012-9012] ../src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
      1724.185533625 [9012-9012] ../src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: , major version: 0, minor version: 0
      1724.185535862 [9012-9012] ../src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 100 1 0 fe
vainfo: VA-API version: 1.20 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      1724.253564016 [9012-9012] ../src/vabackend.c:2050              nvTerminate Terminating 0x55faa863c4d0
      1724.253601577 [9012-9012] ../src/vabackend.c:2064              nvTerminate Now have 0 (100 max) instances

0.0.10

vainfo --display drm
[... 57 lines cut ...]
      1910.163438854 [9823-9823] ../src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (100 max) instances
      1910.163448955 [9823-9823] ../src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
      1910.169206016 [9823-9823] ../src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
      1910.169216726 [9823-9823] ../src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 100 1 0 fe
      1910.169285884 [9823-9823] ../src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.147.05
vainfo: VA-API version: 1.20 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      1910.215269854 [9823-9823] ../src/vabackend.c:1843              nvTerminate Terminating 0x55a4aca714d0
      1910.215295739 [9823-9823] ../src/vabackend.c:1857              nvTerminate Now have 0 (100 max) instances
```

thesword53, 2024-02-14:
> Hardware decoding in Firefox _works anyway_, **except for latest `master` branch**

Maybe related to https://github.com/elFarto/nvidia-vaapi-driver/issues/271

Jasmin68k, 2024-02-15:
From a quick glance at the code:

The `nv_get_versions` function in `nv-driver.c` changed somewhere between V0.0.9 and V0.0.10 and, at least on my system, the new version seems to return an empty string (see vainfo output in original post).

`nv-driver.c` sets different parameters based on driver version in the `nv_alloc_object` function (f.e. `if (driverMajorVersion < 525) {`...).

Up to about V0.0.10 these comparisons might not result in an optimal set of values for the 525 driver on my system due to the non-working `nv_get_version` function, but decoding still works.

In latest `master` however, the empty string is replaced with dummy version `470.123.45`, resulting in the `alloc_buffer` function handling `nvkms_params_size` differently `.nvkms_params_size = context->driverMajorVersion == 470 ? 0x20 : sizeof(nvkmsParams)`.

I assume the latter is why latest master does not work anymore on my system, and does work again, if I replace the dummy driver version with `525.147.05`.

I don't claim to know exactly, how the code is working, so I might be off here.

mirh, 2024-02-15:
That was added in 1d6a9d39bed38d1e2cfbe6528fafc91e1618af68 to fix #267

elFarto, 2024-09-09:
I've added an additional work around in master for reading the version if the ioctl fails to. I'm not sure if this is still an issue for you, but if you could test it and see if it resolves the issue.

Jasmin68k, 2024-09-10:
It's been a while, since I looked at this, since things have just been working for some time now.

It was with release V0.0.12 that I noticed, I don't need to patch in my driver version manually anymore as described above and `vainfo --display drm` showed me that the correct driver version was detected.

Whether that was some fix in V0.0.12, in the driver or somewhere else, I don't know.

I would love to give you more info, but `vainfo --display drm` does not give me verbose output anymore, just this:

```
vainfo --display drm
vainfo: VA-API version: 1.22 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
```

So I don't know, which driver version it actually detects or whether it uses your workaround or the fallback version.

I can tell you that V0.0.12 and `master` are still working, as in it shows GPU decode in `nvtop`.

Would you happen to know, why `vainfo` doesn't show all the info it used to show and how to fix this? Then I could tell you more.

I haven't ugpraded the system for a while, so tests where done with driver version 535.183.01. VA-API/libva as shown in `vainfo` might not be the latest either.

elFarto, 2024-09-10:
You need `NVD_LOG` set to `1` for this driver to log stuff.