elFarto/nvidia-vaapi-driver #278: Hardware rendering doesn't always initialize

raphaelventura, 2024-02-21:
Hi,

It happens quite often that my firefox is unable to initialize hw video acceleration properly.

Here are two sequences of error messages I get inside the `about:config` tab.

```
(#0) Error	Failed to create EGLSurface!: 0x3003
(#1) Error	Failed to create EGLSurface. 4 renderers, 3 active.
(#2) Error	Handling webrender error 3
(#3) Error	GFX: RenderThread detected a device reset in PostUpdate
(#4) Error	Failed to make render context current during destroying.
(#5) Error	Failed to make render context current during destroying.
(#6) Error	GFX: RenderThread detected a device reset in PostUpdate
(#7) Error	Failed to make render context current during destroying.
(#8) Error	Failed to make render context current during destroying.
```

```
(#0) Error	Failed to create EGLSurface!: 0x3003
(#1) Error	Failed to create EGLSurface. 4 renderers, 1 active.
(#2) Error	Handling webrender error 3
(#3) Error	Fallback WR to SW-WR
```

My HW config:
- CPU: quad core Intel Core i5-4570
- Kernel: 6.1.71-1-MANJARO x86_64

Environment variables:
```
VDPAU_DRIVER=nvidia
LIBVA_DRIVER_NAME=nvidia
NVD_BACKEND=direct
MOZ_DISABLE_RDD_SANDBOX=1
```

`vainfo` output:
```
Trying display: wayland
Trying display: x11
libva error: vaGetDriverNames() failed with unknown libva error
vainfo: VA-API version: 1.20 (libva 2.20.1)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
```

The following variables are set as stated in the README
- media.ffmpeg.vaapi.enabled
- media.rdd-ffmpeg.enabled
- media.av1.enabled
- gfx.x11-egl.force-enabled

sakithb, 2024-02-29:
Exact same issue here

GPU: Nvidia GTX 1060
Driver: 550
Kernel: 6.7.6

nathansmeal, 2024-03-08:
It looks like you are still using the EGL backend. Maybe the environment variable for the backend isn't working?

What is the output of `env NVD_BACKEND=direct vainfo`

peperunas, 2024-03-13:
Hello, I think I'm on the same boat.

This is the output of `VDPAU_DRIVER=nvidia LIBVA_DRIVER_NAME=nvidia NVD_BACKEND=direct DISPLAY=:0 vainfo --display drm --device /dev/dri/renderD128`:

```
libva info: VA-API version 1.20.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /run/opengl-driver/lib/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.20 (libva 2.20.1)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
```

NVIDIA driver: `545.29.02`

I guess we are missing `VAEntrypointEncSlice` and `VAEntrypointEncSliceLP`?

raphaelventura, 2024-03-15:
The environment variable for the backend seems to be working properly

```
~$ env NVD_BACKEND=direct vainfo
Trying display: wayland
Trying display: x11
libva error: vaGetDriverNames() failed with unknown libva error
vainfo: VA-API version: 1.20 (libva 2.20.1)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
```

nathansmeal, 2024-03-15:
> Hello, I think I'm on the same boat.
>
> This is the output of `VDPAU_DRIVER=nvidia LIBVA_DRIVER_NAME=nvidia NVD_BACKEND=direct DISPLAY=:0 vainfo --display drm --device /dev/dri/renderD128`:
>
> ```
> libva [...]

VAEntrypointEnc is for NVENC to encode video, this driver does not support NVENC yet.

See #116

nathansmeal, 2024-03-15:
> The environment variable for the backend seems to be working properly
>
> ```
> ~$ env NVD_BACKEND=direct vainfo
> Trying display: wayland
> Trying display: x11
> libva error: vaGetDriverNames() failed with [...]

Please set NVD_BACKEND=direct in your etc/environment file and reboot, as the egl backend does not work on your driver version.

raphaelventura, 2024-03-16:
> > The environment variable for the backend seems to be working properly
> > ```
> > ~$ env NVD_BACKEND=direct vainfo
> > Trying display: wayland
> > Trying display: x11
> > libva error: vaGetDriverNames() failed with unknown libva error
> > vainfo: VA-API version: 1.20 (libva 2.20.1)
> > vainfo: Driver version: VA-API NVDEC driver [direct backend]
> > vainfo: Supported profile and entrypoints
> >       VAProfileMPEG2Simple            : VAEntrypointVLD
> >       VAProfileMPEG2Main              : VAEntrypointVLD
> >       VAProfileVC1Simple              : VAEntrypointVLD
> >       VAProfileVC1Main                : VAEntrypointVLD
> >       VAProfileVC1Advanced            : VAEntrypointVLD
> >       VAProfileH264Main               : VAEntrypointVLD
> >       VAProfileH264High               : VAEntrypointVLD
> >       VAProfileH264ConstrainedBaseline: VAEntrypointVLD
> > ```
> 
> Please set NVD_BACKEND=direct in your etc/environment file and reboot, as the egl backend does not work on your driver version.
> 
> Also please set gfx.x11-egl.force-enabled=false in your Firefox settings

Oh now I see that in my first message, `vainfo` showed the egl backend, which is weird since the env variable was already set in my `etc/environment`. Maybe I'd unset it by mistake after boot and before printing the output.

As for the `gfx.x11-egl.force-enabled=false`, I'd understood from [this reply](https://github.com/elFarto/nvidia-vaapi-driver/issues/265#issuecomment-1879740924) that I should set it to true anyway.

nathansmeal, 2024-03-16:
Yeah my bad on the Firefox, keep that set to true.

Do you still have issues even with the direct backend?

raphaelventura, 2024-03-18:
Yes, still the same error messages in the `about:support` tab: 
- Failed to create EGLSurface!: 0x3003
- Fallback WR to SW-WR