elFarto/nvidia-vaapi-driver #280: HEVC yuv420p10le HDR: out of memory

leifhelm, 2024-03-23:
When playing back HEVC 8-bit content with `mpv --hwdec=vaapi` (`NVD_BACKEND=direct`) on an GTX 1060 I have no problems.
However when playing back HDR10 content I get the following log with (`NVD_LOG=1`):
<details><summary>log</summary>

```
      1405.405771150 [21727-21738] ../src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      1405.405783564 [21727-21738] ../src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1405.405788463 [21727-21738] ../src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
      1405.405869443 [21727-21738] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
      1405.405892391 [21727-21738] ../src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD129
      1405.405899181 [21727-21738] ../src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
      1405.405920483 [21727-21738] ../src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 550.67, major version: 550, minor version: 67
      1405.405928441 [21727-21738] ../src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 100 1 0 fe
      1405.468799970 [21727-21738] ../src/vabackend.c:1398      nvQueryImageFormats In nvQueryImageFormats
      1405.495043540 [21727-21738] ../src/vabackend.c: 637           nvCreateConfig got profile: 0 with 0 attributes
      1405.495057643 [21727-21738] ../src/vabackend.c:1754 nvQuerySurfaceAttributes with 1 (8) (nil) 0
      1405.495061813 [21727-21738] ../src/vabackend.c:1754 nvQuerySurfaceAttributes with 1 (8) 0x7fd9153f3980 8
      1405.496599447 [21727-21738] ../src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4080, height: 16 - 4080
      1405.496615676 [21727-21738] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7fd9153f5940)
      1405.496622349 [21727-21738] ../src/vabackend.c:1492            nvDeriveImage In nvDeriveImage
      1405.496629051 [21727-21738] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7fd9150ec5c0)
      1405.496633232 [21727-21738] ../src/vabackend.c:1492            nvDeriveImage In nvDeriveImage
      1405.496672170 [21727-21738] ../src/vabackend.c:2024    nvExportSurfaceHandle Exporting surface: -1 (0x7fd9150ec5c0)
      1405.496677121 [21727-21738] ../src/direct/direct-export-buf.c: 188 direct_allocateBackingImage Allocating BackingImages: 0x7fd9150ef160 128x128
      1405.496680984 [21727-21738] ../src/direct/nv-driver.c: 447              alloc_image Calculated GOB size: 64x128 (0x4)
      1405.496684323 [21727-21738] ../src/direct/nv-driver.c: 456              alloc_image Aligned image size: 128x128 = 16384
      1405.496795581 [21727-21738] ../src/direct/nv-driver.c: 532              alloc_image created image: 128x128 3000000004fe014 128 4000
      1405.496801098 [21727-21738] ../src/direct/nv-driver.c: 447              alloc_image Calculated GOB size: 64x64 (0x3)
      1405.496804435 [21727-21738] ../src/direct/nv-driver.c: 456              alloc_image Aligned image size: 128x64 = 8192
      1405.496940925 [21727-21738] ../src/direct/nv-driver.c: 532              alloc_image created image: 64x64 3000000004fe013 128 2000
      1405.496948506 [21727-21738] ../src/direct/direct-export-buf.c: 194 direct_allocateBackingImage Importing images
      1405.496953973 [21727-21738] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 128x128 = 4000
      1405.497089088 [21727-21738] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 64x64 = 2000
      1405.497290744 [21727-21738] ../src/vabackend.c:2037    nvExportSurfaceHandle Exporting with 128 128 0 128 3000000004fe014 0 128 3000000004fe013
      1405.497648368 [21727-21738] ../src/vabackend.c: 954        nvDestroySurfaces Destroying surface -1 (0x7fd9150ec5c0)
      1405.497654291 [21727-21738] ../src/direct/direct-export-buf.c: 233      destroyBackingImage Destroying BackingImage: 0x7fd9150ef160
      1405.497800944 [21727-21738] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 128x128, format 100 (0x7fd9150f09a0)
      1405.497807423 [21727-21738] ../src/vabackend.c:1492            nvDeriveImage In nvDeriveImage
      1405.497825844 [21727-21738] ../src/vabackend.c:2024    nvExportSurfaceHandle Exporting surface: -1 (0x7fd9150f09a0)
      1405.497830022 [21727-21738] ../src/direct/direct-export-buf.c: 188 direct_allocateBackingImage Allocating BackingImages: 0x7fd9150f0b60 128x128
      1405.497833640 [21727-21738] ../src/direct/nv-driver.c: 447              alloc_image Calculated GOB size: 64x128 (0x4)
      1405.497836956 [21727-21738] ../src/direct/nv-driver.c: 456              alloc_image Aligned image size: 256x128 = 32768
      1405.497928318 [21727-21738] ../src/direct/nv-driver.c: 532              alloc_image created image: 128x128 3000000004fe014 256 8000
      1405.497933779 [21727-21738] ../src/direct/nv-driver.c: 447              alloc_image Calculated GOB size: 64x64 (0x3)
      1405.497937092 [21727-21738] ../src/direct/nv-driver.c: 456              alloc_image Aligned image size: 256x64 = 16384
      1405.498012760 [21727-21738] ../src/direct/nv-driver.c: 532              alloc_image created image: 64x64 3000000004fe013 256 4000
      1405.498017838 [21727-21738] ../src/direct/direct-export-buf.c: 194 direct_allocateBackingImage Importing images
      1405.498021154 [21727-21738] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 128x128 = 8000
      1405.498101933 [21727-21738] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 64x64 = 4000
      1405.498203566 [21727-21738] ../src/vabackend.c:2037    nvExportSurfaceHandle Exporting with 128 128 0 256 3000000004fe014 0 256 3000000004fe013
      1405.498473668 [21727-21738] ../src/vabackend.c: 954        nvDestroySurfaces Destroying surface -1 (0x7fd9150f09a0)
      1405.498479347 [21727-21738] ../src/direct/direct-export-buf.c: 233      destroyBackingImage Destroying BackingImage: 0x7fd9150f0b60
      1405.498591436 [21727-21738] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 128x128, format 1000 (0x7fd9150ec420)
      1405.498597271 [21727-21738] ../src/vabackend.c:1492            nvDeriveImage In nvDeriveImage
      1405.498613658 [21727-21738] ../src/vabackend.c:2024    nvExportSurfaceHandle Exporting surface: -1 (0x7fd9150ec420)
      1405.498618119 [21727-21738] ../src/direct/direct-export-buf.c: 188 direct_allocateBackingImage Allocating BackingImages: 0x7fd9150f1fc0 128x128
      1405.498622385 [21727-21738] ../src/direct/nv-driver.c: 447              alloc_image Calculated GOB size: 64x128 (0x4)
      1405.498626124 [21727-21738] ../src/direct/nv-driver.c: 456              alloc_image Aligned image size: 256x128 = 32768
      1405.498715910 [21727-21738] ../src/direct/nv-driver.c: 532              alloc_image created image: 128x128 3000000004fe014 256 8000
      1405.498721229 [21727-21738] ../src/direct/nv-driver.c: 447              alloc_image Calculated GOB size: 64x64 (0x3)
      1405.498724912 [21727-21738] ../src/direct/nv-driver.c: 456              alloc_image Aligned image size: 256x64 = 16384
      1405.498789836 [21727-21738] ../src/direct/nv-driver.c: 532              alloc_image created image: 64x64 3000000004fe013 256 4000
      1405.498794751 [21727-21738] ../src/direct/direct-export-buf.c: 194 direct_allocateBackingImage Importing images
      1405.498798083 [21727-21738] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 128x128 = 8000
      1405.498876505 [21727-21738] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 64x64 = 4000
      1405.498991462 [21727-21738] ../src/vabackend.c:2037    nvExportSurfaceHandle Exporting with 128 128 0 256 3000000004fe014 0 256 3000000004fe013
      1405.499264563 [21727-21738] ../src/vabackend.c: 954        nvDestroySurfaces Destroying surface -1 (0x7fd9150ec420)
      1405.499269830 [21727-21738] ../src/direct/direct-export-buf.c: 233      destroyBackingImage Destroying BackingImage: 0x7fd9150f1fc0
      1405.499390221 [21727-21738] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 128x128, format 4 (0x7fd9150f0860)
[... cut ...]
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      2056.656698816 [22587-22587] ../src/vabackend.c:2050              nvTerminate Terminating 0x16da4e0
      2056.656771906 [22587-22587] ../src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances
```

</details>

thesword53, 2024-03-24:
Is your GTX 1060 a 3GB or a 6GB variant ?
What version of the driver do you use ?
I couldn't reproduce the issue with my GTX 1060 mobile (6GB).

leifhelm, 2024-03-24:
My Card is a 1060 6GB (desktop). Driver version 550.67 with CUDA 12.4 according to `nvidia-smi`. I used these test files: https://www.demolandia.net/4k-video-test/hdr10/page-1.html?highlight=1

thesword53, 2024-04-14:
I can reproduce the issue with my RTX 2080 SUPER and 550.67 drivers. My GTX 1060 is running with 535 drivers due to a bug with 535 drivers causing kernel crashes in laptops with integrated graphics https://forums.developer.nvidia.com/t/series-550-freezes-laptop/284772.

elFarto, 2024-05-06:
Found the issue. It's related to use selecting Adaptive deinterlacing. This seems to consume more memory. Changing `cudaVideoDeinterlaceMode_Adaptive` to `cudaVideoDeinterlaceMode_Weave` seems to solve that issue. 

However, when I do that I end up with a blue image instead of the video, so something else might be broken too.

elFarto, 2024-05-08:
This issue should be fixed in master now.

leifhelm, 2024-05-09:
@elFarto Thank you. Your fix works for me.

closed 2024-05-09 by leifhelm