elFarto/nvidia-vaapi-driver #281: VAProfileH264High10 (decode) support?

oscarbg, 2024-03-28:
Hi,
I don't even know if NV supports it in HW, but seems Chinese GPUs yes..
https://www.phoronix.com/news/Moore-Threads-VA-API-FFmpeg
and running vainfo can't see it:
```
vainfo: VA-API version: 1.20 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
```

also FFMpeg supports decoding and encoding this profile from phoronix:
https://github.com/FFmpeg/FFmpeg/commit/1cd5a383d49f94b05699b234b0d3c98f2969b810
https://github.com/FFmpeg/FFmpeg/commit/65e537b83379d2d6f87343f9bd59968b21d949a3

thanks..

philipl, 2024-03-29:
None of the main desktop gpu manufacturers support Hi10p and they never will - there was no commercial content and then hevc came along to become the standard.

oscarbg, 2024-03-29:
thanks for info..

closed 2024-03-29 by oscarbg