elFarto/nvidia-vaapi-driver #282: Not working anymore on Kepler 470.239.06

lucasmr, 2024-03-29:
GPU: 660Ti
Driver: 470.239.06
Kernel: 6.6.21 Gentoo
Using nvidia-vaapi-driver 0.0.11 and Firefox 124.0.1

```
$ vainfo
Trying display: x11
libva info: VA-API version 1.20.0
libva error: vaGetDriverNames() failed with unknown libva error
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/va/drivers/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
       417.742729135 [7994-7994] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
       417.742735644 [7994-7994] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
       417.742738852 [7994-7994] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2172       __vaDriverInit_1_0 Selecting EGL backend
       417.746473220 [7994-7994] ../nvidia-vaapi-driver-0.0.11/src/export-buf.c: 132       findGPUIndexFromFd Defaulting to CUDA GPU ID 0. Use NVD_GPU to select a specific CUDA GPU
       417.746486766 [7994-7994] ../nvidia-vaapi-driver-0.0.11/src/export-buf.c: 149       findGPUIndexFromFd Looking for GPU index: 0
       417.747546758 [7994-7994] ../nvidia-vaapi-driver-0.0.11/src/export-buf.c: 161       findGPUIndexFromFd Found 1 EGL devices
       417.747559833 [7994-7994] ../nvidia-vaapi-driver-0.0.11/src/export-buf.c: 199       findGPUIndexFromFd No DRM device file for EGLDevice 0
       417.747563183 [7994-7994] ../nvidia-vaapi-driver-0.0.11/src/export-buf.c: 202       findGPUIndexFromFd No match found, falling back to default device
       417.747566109 [7994-7994] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2200       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib64/va/drivers/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit
```

```$ vdpauinfo
display: :0   screen: 0
API version: 1
Information string: NVIDIA VDPAU Driver Shared Library  470.239.06  Sat Feb  3 06:02:58 UTC 2024

Video surface:

name   width height types
-------------------------------------------
420     4096  4096  NV12 YV12 
422     4096  4096  UYVY YUYV 
444     4096  4096  NV24 YV24 
420_16  4096  4096  P010 P016 
422_16  4096  4096  UYVY YUYV 
444_16  4096  4096  Y_U_V_444_16 

Decoder capabilities:

name                        level macbs width height
----------------------------------------------------
MPEG1                           0 65536  4032  4048
MPEG2_SIMPLE                    3 65536  4032  4048
MPEG2_MAIN                      3 65536  4032  4048
H264_BASELINE                  51 65536  4032  4080
H264_MAIN                      51 65536  4032  4080
H264_HIGH                      51 65536  4032  4080
VC1_SIMPLE                      1  8190  2048  2048
VC1_MAIN                        2  8190  2048  2048
VC1_ADVANCED                    4  8190  2048  2048
MPEG4_PART2_SP                  3  8192  2048  2048
MPEG4_PART2_ASP                 5  8192  2048  2048
DIVX4_QMOBILE                   0  8192  2048  2048
DIVX4_MOBILE                    0  8192  2048  2048
DIVX4_HOME_THEATER              0  8192  2048  2048
DIVX4_HD_1080P                  0  8192  2048  2048
DIVX5_QMOBILE                   0  8192  2048  2048
DIVX5_MOBILE                    0  8192  2048  2048
DIVX5_HOME_THEATER              0  8192  2048  2048
DIVX5_HD_1080P                  0  8192  2048  2048
H264_CONSTRAINED_BASELINE      51 65536  4032  4080
H264_EXTENDED                  51 65536  4032  4080
H264_PROGRESSIVE_HIGH          51 65536  4032  4080
H264_CONSTRAINED_HIGH          51 65536  4032  4080
H264_HIGH_444_PREDICTIVE       51 65536  4032  4080
VP9_PROFILE_0                  --- not supported ---
VP9_PROFILE_1                  --- not supported ---
VP9_PROFILE_2                  --- not supported ---
VP9_PROFILE_3                  --- not supported ---
HEVC_MAIN                      --- not supported ---
HEVC_MAIN_10                   --- not supported ---
HEVC_MAIN_STILL                --- not supported ---
HEVC_MAIN_12                   --- not supported ---
HEVC_MAIN_444                  --- not supported ---
HEVC_MAIN_444_10               --- not supported ---
HEVC_MAIN_444_12               --- not supported ---
AV1_MAIN                       --- not supported ---
AV1_HIGH                       --- not supported ---
AV1_PROFESSIONAL               --- not supported ---

Output surface:

name              width height nat types
----------------------------------------------------
B8G8R8A8         16384 16384    y  Y8U8V8A8 V8U8Y8A8 A4I4 I4A4 A8I8 I8A8 
R10G10B10A2      16384 16384    y  Y8U8V8A8 V8U8Y8A8 A4I4 I4A4 A8I8 I8A8 

Bitmap surface:

name              width height
------------------------------
B8G8R8A8         16384 16384
R8G8B8A8         16384 16384
R10G10B10A2      16384 16384
B10G10R10A2      16384 16384
A8               16384 16384

Video mixer:

feature name                    sup
------------------------------------
DEINTERLACE_TEMPORAL             y
DEINTERLACE_TEMPORAL_SPATIAL     y
INVERSE_TELECINE                 y
NOISE_REDUCTION                  y
SHARPNESS                        y
LUMA_KEY                         y
HIGH QUALITY SCALING - L1        y
HIGH QUALITY SCALING - L2        -
HIGH QUALITY SCALING - L3        -
HIGH QUALITY SCALING - L4        -
HIGH QUALITY SCALING - L5        -
HIGH QUALITY SCALING - L6        -
HIGH QUALITY SCALING - L7        -
HIGH QUALITY SCALING - L8        -
HIGH QUALITY SCALING - L9        -

parameter name                  sup      min      max
-----------------------------------------------------
VIDEO_SURFACE_WIDTH              y         1     4096
VIDEO_SURFACE_HEIGHT             y         1     4096
CHROMA_TYPE                      y  
LAYERS                           y         0        4

attribute name                  sup      min      max
-----------------------------------------------------
BACKGROUND_COLOR                 y  
CSC_MATRIX                       y  
NOISE_REDUCTION_LEVEL            y      0.00     1.00
SHARPNESS_LEVEL                  y     -1.00     1.00
LUMA_KEY_MIN_LUMA                y  
LUMA_KEY_MAX_LUMA                y
```

I noticed since last system upgrade it stopped working, and there was a NVIDIA driver update so it might be related.

mirh, 2024-04-10:
470 was already [broken](267) months ago, you need the last master
Idk then if the last driver couldn't have made an extra difference too

elFarto, 2024-05-06:
Can you try the latest version, v0.0.12?

lucasmr, 2024-05-11:
Thanks, it's working now.

closed 2024-05-11 by lucasmr