elFarto/nvidia-vaapi-driver #283: HW acceleration not working in firefox 124.0.2

Vitaly76ASDgnaerkldbnak, 2024-04-06:
HW acceleration not working in firefox 124.0.2, since FF self upgrade, I didn't notice from what version it broke, but in version 110 it still works normal. I check this on nvidia drivers V525.147.05 and V545.29.06. Option widget.dmabuf.force-enabled is true, but in 124.0.2 it is not helping. BTW, HW acceleration in MPV works fine.

mirh, 2024-04-10:
Is there no difference in `about:support`?

Vitaly76ASDgnaerkldbnak, 2024-04-10:
There are too many info in about:support, what exactly should I pay attention to?

mirh, 2024-04-12:
The ones specifically about acceleration?
More or less everything in the graphics section, but decision log has the most important stuff.

Vitaly76ASDgnaerkldbnak, 2024-04-14:
In section media, in ff 110 there are H264 HW and VP8 HW, but in FF 125 there are only SWs. It seems obvious, but why FF 125 is not detecting HW acceleration?

mirh, 2024-04-14:
Ok, and that's the symptoms. But it's the section right before to be telling you the causes.

LocalRaccoonEnthusiast, 2024-04-22:
[image]
This section should be telling a lot if you send it here. Mine isn't what the result is supposed to be and I'm still trying to fix it, but just so you know what section to pay attention to.

Vitaly76ASDgnaerkldbnak, 2024-04-22:
ff 110
[image]
ff 125
[image]

Vitaly76ASDgnaerkldbnak, 2024-04-22:
Maybe this can help, I'm totally frustrated in this.

mirh, 2024-04-23:
Obviously those two errors are either the usual [`widget.dmabuf.force-enabled`](https://github.com/elFarto/nvidia-vaapi-driver/issues/182) or [`media.hardware-video-decoding.force-enabled`](https://wiki.archlinux.org/title/firefox#Hardware_video_acceleration) thing.

Vitaly76ASDgnaerkldbnak, 2024-04-24:
Even after enable [media.hardware-video-decoding.force-enabled](https://wiki.archlinux.org/title/firefox#Hardware_video_acceleration) in FF 125 there is no video acceleration. In FF 110 without this was. So it seems, it's something else...

mirh, 2024-04-25:
Did you remember `MOZ_DISABLE_RDD_SANDBOX=1` and the dmabuf thing?

Vitaly76ASDgnaerkldbnak, 2024-04-25:
Alas, but all this options are set in both versions of FF.

Vitaly76ASDgnaerkldbnak, 2024-04-25:
In /etc/environment
``
NVD_BACKEND=direct
MOZ_DISABLE_RDD_SANDBOX=1
LIBVA_DRIVER_NAME=nvidia
``

mirh, 2024-04-26:
Try firefox nightly
https://fedoraproject.org/wiki/How_to_debug_Firefox_problems#Testing_Mozilla_binaries

Vitaly76ASDgnaerkldbnak, 2024-04-27:
I checked in FF 126.0b6 and it's works (stand along version). I think problem is in FF of kubuntu distribution, so it's no problem with nvidia-vaapi-driver. So the issue can be close, big thanks for helping me anyway.

mirh, 2024-04-27:
Then it seems like you skipped *widget.dmabuf.force-enabled*...

Vitaly76ASDgnaerkldbnak, 2024-04-28:
No, it is set true in FF 125. For me it's clear that FF in  ubuntu distributive in some way compiled without some essential thing that is not letting  HW acceleration to function, maybe it's connected to dmabuf, maybe to something else.  But when I download standalone version of FF 110 and later, than HW acceleration is here.

mirh, 2024-05-06:
Wayland support has been built-in since at least some years [by now](https://bugzilla.mozilla.org/show_bug.cgi?id=1725245), if any it wasn't until [recent](https://www.phoronix.com/news/Ubuntu-23.10-Firefox-Wayland) that the snap enabled it by default (but that's kinda a different story). 

Speaking of ubuntu.. I just realized the version wasn't even reported here.

Vitaly76ASDgnaerkldbnak, 2024-05-06:
I'm using FF from PPA https://ppa.launchpadcontent.net/mozillateam/ppa/ubuntu in Kubuntu 22.04.3 LTS, any versions of FF from 124.x to 125.x. And I'm in X11 not Wayland.

mirh, 2024-05-06:
#210 had it working though?

sekaiacg, 2024-05-18:
In addition to the recommended settings, I turned on hardware acceleration using the following additional settings on archlinux.
[image]

https://wiki.archlinux.org/title/firefox#Hardware_video_acceleration

elFarto, 2025-08-07:
Assuming issue is resolved.