elFarto/nvidia-vaapi-driver #284: Failed to build on RHEL 8.9

IterableTrucks, 2024-04-14:
failed to build nvidia-vaapi-driver with meson:
> [16/16] Linking target nvidia_drv_video.so
FAILED: nvidia_drv_video.so 
cc  -o nvidia_drv_video.so nvidia_drv_video.so.p/src_av1.c.o nvidia_drv_video.so.p/src_backend-common.c.o nvidia_drv_video.so.p/src_export-buf.c.o nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o nvidia_drv_video.so.p/src_direct_nv-driver.c.o nvidia_drv_video.so.p/src_h264.c.o nvidia_drv_video.so.p/src_hevc.c.o nvidia_drv_video.so.p/src_jpeg.c.o nvidia_drv_video.so.p/src_mpeg2.c.o nvidia_drv_video.so.p/src_mpeg4.c.o nvidia_drv_video.so.p/src_vabackend.c.o nvidia_drv_video.so.p/src_vc1.c.o nvidia_drv_video.so.p/src_vp8.c.o nvidia_drv_video.so.p/src_list.c.o nvidia_drv_video.so.p/src_vp9.c.o -Wl,--as-needed -Wl,--no-undefined -shared -fPIC -Wl,-soname,nvidia_drv_video.so -Wl,--start-group -lm -ldl /usr/lib64/libEGL.so -pthread /usr/lib64/libgstcodecparsers-1.0.so /usr/lib64/libgstbase-1.0.so /usr/lib64/libgstreamer-1.0.so /usr/lib64/libgobject-2.0.so /usr/lib64/libglib-2.0.so -Wl,--end-group
nvidia_drv_video.so.p/src_vabackend.c.o: In function `logger':
/tmp/nvidia-vaapi-driver/build/../src/vabackend.c:166: undefined reference to `gettid'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Could not rebuild /tmp/nvidia-vaapi-driver/build

The kernel version is 4.18.0-513.18.1.el8_9.x86_64 and gcc version is 8.5.0 20210514 (Red Hat 8.5.0-20)

nerijus, 2024-05-04:
https://github.com/elFarto/nvidia-vaapi-driver/pull/286 should fix it.

elFarto, 2024-05-06:
This should be fixed in v0.0.12.

elFarto, 2025-08-07:
Assuming issue is resolved.

closed 2025-08-07 by elFarto