elFarto/nvidia-vaapi-driver #285: libva error: vaGetDriverNames() failed with unknown libva error with 2 GPUs & NVIDIA drivers

rchennafi, 2024-04-27:
I was recently trying to use `usevia`  https://github.com/the-via/releases - which uses VA-API to render things through electron, but it did not work, upon further investigation there appeared to be some sort of issue with my install of VA-API. The browser `usevia.app` version of this app also has problems but I don't see what the particular error actually is there.,

The error that seems to pop up is `libva error: vaGetDriverNames() failed with unknown libva error` 

I went through several issues here and setup my nvidia card as a direct backend, ensured other drivers were removed, and set the NVD_BACKEND=direct and LIBVA_DRIVER_NAME=nvidia.

When the usevia app is on in system or on the browser, I see in nvidia-settings that my GPU usage is going up but the Video Engine Utilization is going unchanged.

Maybe this issue is caused by the presence of two GPUs on my system? One is isolated from the machine and setup in passthrough to a local windows VM.

System: `Linux 6.8.7-arch1-1`

<details><summary>lspci | grep NVIDIA </summary>

```
03:00.0 VGA compatible controller: NVIDIA Corporation GA102 [GeForce RTX 3080] (rev a1)
03:00.1 Audio device: NVIDIA Corporation GA102 High Definition Audio Controller (rev a1)
09:00.0 VGA compatible controller: NVIDIA Corporation GP106 [GeForce GTX 1060 3GB] (rev a1)
09:00.1 Audio device: NVIDIA Corporation GP106 High Definition Audio Controller (rev a1)
```

</details> 

<details>
<summary>sudo lspci -knn | grep nvidia</summary>

```bash
Kernel modules: nouveau, nvidia_drm, nvidia
Kernel driver in use: nvidia
Kernel modules: nouveau, nvidia_drm, nvidia
```

</details>

<details><summary>nvidia-smi</summary>

```
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.76                 Driver Version: 550.76         CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce GTX 1060 3GB    Off |   00000000:09:00.0  On |                  N/A |
|  0%   52C    P5              8W /  150W |     254MiB /   3072MiB |      1%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      1010      G   /usr/lib/Xorg                                 161MiB |
|    0   N/A  N/A      3171      G   /usr/lib/insync/insync                          2MiB |
|    0   N/A  N/A      6984      G   ...00,262144 --variations-seed-version         86MiB |
+-----------------------------------------------------------------------------------------+
```

</details>

This command seems to fail when trying the x11 display.
<details><summary>NVD_LOG=1 vainfo</summary>

```
Trying display: wayland
Trying display: x11
libva error: vaGetDriverNames() failed with unknown libva error
      1634.538727240 [24568-24568] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      1634.538732940 [24568-24568] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1634.538735295 [24568-24568] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
      1634.544265330 [24568-24568] ../nvidia-vaapi-driver-0.0.11/src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
      1634.544270500 [24568-24568] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
      1634.544284997 [24568-24568] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 550.76, major version: 550, minor version: 76
      1634.544289876 [24568-24568] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 900 1 0 fe
vainfo: VA-API version: 1.21 (libva 2.21.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      1634.621263968 [24568-24568] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2050              nvTerminate Terminating 0x61cba0922360
      1634.621338188 [24568-24568] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances
```

</details>

It seems it works fine if I just use the drm display, though I do not fully understand what this setting means.

<details><summary>NVD_LOG=1 vainfo --display=drm</summary>

````
Trying display: drm
      1852.216741160 [26010-26010] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      1852.216746400 [26010-26010] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1852.216748664 [26010-26010] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
      1852.222217695 [26010-26010] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
      1852.222236831 [26010-26010] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 550.76, major version: 550, minor version: 76
      1852.222241240 [26010-26010] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 900 1 0 fe
vainfo: VA-API version: 1.21 (libva 2.21.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      1852.288709435 [26010-26010] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2050              nvTerminate Terminating 0x57191f4cea30
      1852.288759850 [26010-26010] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances
```

</details>

Happy to provide more details. Thanks in advance.

elFarto, 2024-05-11:
Is it possible for you to run whatever app is producing the `failed with unknown libva error` with `NVD_LOG=1` set?

philipl, 2024-05-11:
I get this with mpv. It seems harmless in that context but here's what I see (not much it seems)

```
$ NVD_LOG=1 mpv /storage/video/test-patterns/juddertest_60.mp4 --profile=nvidia-vaapi --pause
[auto_hwdec] Applying profile: nvidia
 (+) Video --vid=1 (*) (h264 1920x1072 60.000fps)
[vaapi] libva: vaGetDriverNames() failed with unknown libva error
    153672.905497062 [522207-522224] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
    153672.905505190 [522207-522224] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 0 (0 max) instances
    153672.905506954 [522207-522224] ../src/vabackend.c:2222       __vaDriverInit_1_0 Selecting Direct backend
    153672.905514018 [522207-522224] ../src/direct/direct-export-buf.c:  68      direct_initExporter Searching for GPU: 0 0 128
    153672.905569045 [522207-522224] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
    153672.905576246 [522207-522224] ../src/direct/direct-export-buf.c:  68      direct_initExporter Searching for GPU: 0 0 129
    153672.905581442 [522207-522224] ../src/direct/direct-export-buf.c:  90      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD129
    153672.905583630 [522207-522224] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
    153672.905595619 [522207-522224] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 550.78, major version: 550, minor version: 78
    153672.905598362 [522207-522224] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 1 2 6
```

philipl, 2024-05-11:
Or just vainfo:

```
$ NVD_LOG=1 vainfo 
libva info: VA-API version 1.20.0
libva error: vaGetDriverNames() failed with unknown libva error
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
    153782.803536185 [522753-522753] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
    153782.803545194 [522753-522753] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 0 (0 max) instances
    153782.803547590 [522753-522753] ../src/vabackend.c:2222       __vaDriverInit_1_0 Selecting Direct backend
    153782.809043867 [522753-522753] ../src/direct/direct-export-buf.c:  68      direct_initExporter Searching for GPU: 0 0 128
    153782.809098594 [522753-522753] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
    153782.809104074 [522753-522753] ../src/direct/direct-export-buf.c:  68      direct_initExporter Searching for GPU: 0 0 129
    153782.809108815 [522753-522753] ../src/direct/direct-export-buf.c:  90      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD129
    153782.809110525 [522753-522753] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
    153782.809122503 [522753-522753] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 550.78, major version: 550, minor version: 78
    153782.809125456 [522753-522753] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 1 2 6
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.20 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
    153782.879260909 [522753-522753] ../src/vabackend.c:2097              nvTerminate Terminating 0x5c446236e060
    153782.879299807 [522753-522753] ../src/vabackend.c:2111              nvTerminate Now have 0 (0 max) instances
```

mirh, 2024-07-15:
> The error that seems to pop up is libva error: vaGetDriverNames() failed with unknown libva error

That seems the usual error when you aren't compiling libva with `-Dwith_legacy=nvctrl`

rchennafi, 2024-07-16:
@elFarto 
> Is it possible for you to run whatever app is producing the `failed with unknown libva error` with `NVD_LOG=1` set?

Sorry about the delay, kept forgetting to answer this one. The app here is UseVIA. 

https://github.com/the-via/releases/releases/tag/v3.0.0

```
libva error: vaGetDriverNames() failed with unknown libva error
   4233189.194632299 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
   4233189.194641867 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
   4233189.194644763 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
   4233189.209202604 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
   4233189.209209357 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
   4233189.209224495 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 550.76, major version: 550, minor version: 76
   4233189.209228322 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 900 1 0 fe
   4233189.341736983 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 569    nvGetConfigAttributes Got here with profile: 6 == 4
   4233189.341748395 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 6 with 1 attributes
   4233189.341752453 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 658           nvCreateConfig got config attrib: 0 0 1
   4233189.341757161 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) (nil) -1431655766
   4233189.341760237 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) 0x2e9801777180 8
   4233189.343499991 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
   4233189.343511303 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 6 with 0 attributes
   4233189.343517685 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 569    nvGetConfigAttributes Got here with profile: 7 == 4
   4233189.343520269 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 7 with 1 attributes
   4233189.343522353 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 658           nvCreateConfig got config attrib: 0 0 1
   4233189.343525038 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) (nil) -1431655766
   4233189.343527724 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) 0x2e9801777180 8
   4233189.345257059 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
   4233189.345263751 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 7 with 0 attributes
   4233189.345269322 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 569    nvGetConfigAttributes Got here with profile: 13 == 4
   4233189.345271897 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 13 with 1 attributes
   4233189.345273870 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 658           nvCreateConfig got config attrib: 0 0 1
   4233189.345276425 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) (nil) -1431655766
   4233189.345278409 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) 0x2e9801777180 8
   4233189.347290065 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
   4233189.347295836 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 13 with 0 attributes
   4233189.347300996 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 569    nvGetConfigAttributes Got here with profile: 17 == 8
   4233189.347303481 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 17 with 1 attributes
   4233189.347306136 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 658           nvCreateConfig got config attrib: 0 0 1
   4233189.347308420 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 8 (8) (nil) -1431655766
   4233189.347310885 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 8 (8) 0x2e9801777180 8
   4233189.348971030 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 144 - 8192, height: 144 - 8192
[... cut ...]
   4233189.352400906 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 144 - 8192, height: 144 - 8192
   4233189.352406536 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 18 with 0 attributes
   4233189.352420443 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2050              nvTerminate Terminating 0x2e9801791400
   4233189.352506905 [1439902-1439902] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances

```

elFarto, 2024-09-09:
The output looks ok, while there is an error there, the driver is initialised and is returning what formats it supports.

However, I'm not sure the driver is going to work with Electron. Chromium has seemingly only just got support for the last bit needed to make this work properly, I'm not sure when those changes will make it into Electron. This driver only officially supports Firefox.