elFarto/nvidia-vaapi-driver #289: No hardware decode on firefox

Latrolage, 2024-05-06:
Videos just glitch out: (skip to end of video)

https://github.com/elFarto/nvidia-vaapi-driver/assets/67372293/e0032b9a-47fa-47b4-9fc5-65eb5702fb7a

<details>

<summary> NVD_LOG=1 logs </summary>

```
NVD_LOG=1 NVD_BACKEND=direct MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia firefox-nightly -Profile ~/.mozilla/firefox/profile
[GFX1-]: More than 1 GPU from same vendor detected via PCI, cannot deduce device

       572.587116630 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
       572.587130857 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
       572.587133111 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
       572.592982084 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
       572.593031407 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 550.78, major version: 550, minor version: 78
       572.593035254 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 271            init_nvdriver Got dev info: a00 1 2 6
       572.648077001 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1398      nvQueryImageFormats In nvQueryImageFormats
       572.682213661 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 6 with 0 attributes
       572.682227928 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) (nil) 0
       572.682230413 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) 0x78d896a5a100 8
       572.683730067 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       572.683739254 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 7 with 0 attributes
       572.683743813 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) (nil) 0
       572.683747510 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) 0x78d896a5a100 8
       572.685111076 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       572.685119532 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 13 with 0 attributes
       572.685124221 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) (nil) 0
       572.685127437 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) 0x78d896a5ac40 8
       572.686841005 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       572.686848199 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 14 with 0 attributes
       572.686851746 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 9 (8) (nil) 0
       572.686854952 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 9 (8) 0x78d896a5a100 8
       572.688501774 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       572.688509459 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 19 with 0 attributes
       572.688512755 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 10 (8) (nil) 0
       572.688515199 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 10 (8) 0x78d896a5a640 8
       572.689879898 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
       572.689887282 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 32 with 0 attributes
       572.689890218 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 700           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
       572.689893354 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 11 (8) (nil) 0
       572.689895658 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 11 (8) 0x78d896a5ad00 8
       572.691247703 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
       572.691254847 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 21 with 0 attributes
       572.691258273 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 10 (10) (nil) 0
       572.691267881 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 10 (10) 0x78d896a5a100 8
       572.692616009 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
       572.692863066 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2050              nvTerminate Terminating 0x78d8def27e20
       572.693010044 [8530-8542] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances
       573.543948002 [8530-8570] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
       573.543959093 [8530-8570] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
       573.543964774 [8530-8570] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
       573.543988289 [8530-8570] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
       573.544030949 [8530-8570] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 550.78, major version: 550, minor version: 78
       573.544035869 [8530-8570] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 271            init_nvdriver Got dev info: a00 1 2 6
       573.590649465 [8530-8570] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1398      nvQueryImageFormats In nvQueryImageFormats
       573.590921509 [8530-8570] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2050              nvTerminate Terminating 0x78d8def265c0
       573.591083195 [8530-8570] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances
       573.617138993 [8530-8574] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
       573.617148871 [8530-8574] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
       573.617152107 [8530-8574] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
       573.617172987 [8530-8574] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
       573.617198365 [8530-8574] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 550.78, major version: 550, minor version: 78
       573.617201912 [8530-8574] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 271            init_nvdriver Got dev info: a00 1 2 6
       573.666642490 [8530-8574] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1398      nvQueryImageFormats In nvQueryImageFormats
       573.666915426 [8530-8574] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2050              nvTerminate Terminating 0x78d8def267a0
[... cut ...]
       577.303733152 [8530-8670] ../nvidia-vaapi-driver-0.0.11/src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 640x360 = 3c000
       577.303962165 [8530-8670] ../nvidia-vaapi-driver-0.0.11/src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 320x180 = 28000
       577.304252473 [8530-8670] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2037    nvExportSurfaceHandle Exporting with 640 360 0 640 300000000606014 0 640 300000000606014
       577.304259507 [8530-8670] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1339            nvSyncSurface Surface 8 not resolved, waiting
```

</details>

Latrolage, 2024-05-06:
It repeats lines like this:
```
 nvExportSurfaceHandle Exporting with 640 360 0 640 300000000606014 0 640 300000000606014
       578.403716943 [8530-8667] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1339            nvSyncSurface Surface 9 not resolved, waiting
       578.403756178 [8530-8672] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 415          resolveSurfaces Surface 9 exported
       578.499323468 [8530-8670] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1300             nvEndPicture Decoded frame successfully to idx: 8 (0x78d8972197a0)
       578.499349267 [8530-8672] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 391          resolveSurfaces Reading from queue: 5 6
       578.499367702 [8530-8672] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 407          resolveSurfaces Mapping surface 8
       578.499352403 [8530-8670] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2024    nvExportSurfaceHandle Exporting surface: 8 (0x78d8972197a0)
       578.499392068 [8530-8670] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2037    nvExportSurfaceHandle Exporting with 640 360 0 640 300000000606014 0 640 300000000606014
       578.499397858 [8530-8670] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1339            nvSyncSurface Surface 8 not resolved, waiting
       578.499439878 [8530-8672] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 411          resolveSurfaces Mapped surface 8 to 0x205428000 (1024)
       578.499483440 [8530-8672] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 415          resolveSurfaces Surface 8 exported
       578.604452375 [8530-8667] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1300             nvEndPicture Decoded frame successfully to idx: 6 (0x78d897219f20)
       578.604478805 [8530-8672] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 391          resolveSurfaces Reading from queue: 6 7
       578.604495115 [8530-8672] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 407          resolveSurfaces Mapping surface 6
       578.604482882 [8530-8667] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2024    nvExportSurfaceHandle Exporting surface: 6 (0x78d897219f20)
       578.604509102 [8530-8667] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2037    nvExportSurfaceHandle Exporting with 640 360 0 640 300000000606014 0 640 300000000606014
       578.604514993 [8530-8667] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1339            nvSyncSurface Surface 6 not resolved, waiting
       578.604558706 [8530-8672] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 411          resolveSurfaces Mapped surface 6 to 0x205428000 (1024)
       578.604598922 [8530-8672] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 415          resolveSurfaces Surface 6 exported
       578.703632561 [8530-8670] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1300             nvEndPicture Decoded frame successfully to idx: 2 (0x78d897219840)
       578.703659362 [8530-8672] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 391          resolveSurfaces Reading from queue: 7 8
       578.703676164 [8530-8672] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 407          resolveSurfaces Mapping surface 2
       578.703665143 [8530-8670] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2024    nvExportSurfaceHandle Exporting surface: 2 (0x78d897219840)
       578.703689770 [8530-8670] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2037    nvExportSurfaceHandle Exporting with 640 360 0 640 300000000606014 0 640 300000000606014
```
over and over

elFarto, 2024-05-06:
Can you try the latest version (v0.0.12)?

Latrolage, 2024-05-06:
NVD_LOG=1 NVD_BACKEND=direct MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia firefox-nightly -Profile ~/.mozilla/firefox/profile
[GFX1-]: More than 1 GPU from same vendor detected via PCI, cannot deduce device

      7326.077572582 [39236-39430] ../src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      7326.077583312 [39236-39430] ../src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
      7326.077586698 [39236-39430] ../src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
      7326.083631602 [39236-39430] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
      7326.083669062 [39236-39430] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 550.78, major version: 550, minor version: 78
      7326.083674863 [39236-39430] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: a00 1 2 6
      7326.149195173 [39236-39430] ../src/vabackend.c:1446      nvQueryImageFormats In nvQueryImageFormats
      7326.190682887 [39236-39430] ../src/vabackend.c: 674           nvCreateConfig got profile: 6 with 0 attributes
      7326.190700781 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      7326.190703325 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) 0x7658c0a5a280 8
      7326.192692277 [39236-39430] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      7326.192708759 [39236-39430] ../src/vabackend.c: 674           nvCreateConfig got profile: 7 with 0 attributes
      7326.192714850 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      7326.192718046 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) 0x7658c0a5af40 8
      7326.194512912 [39236-39430] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      7326.194526417 [39236-39430] ../src/vabackend.c: 674           nvCreateConfig got profile: 13 with 0 attributes
      7326.194531116 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      7326.194534763 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) 0x7658c0a5a280 8
      7326.196051795 [39236-39430] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      7326.196060171 [39236-39430] ../src/vabackend.c: 674           nvCreateConfig got profile: 14 with 0 attributes
      7326.196063517 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 9 (8) (nil) 0
      7326.196065801 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 9 (8) 0x7658c0a5a280 8
      7326.198052239 [39236-39430] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      7326.198061045 [39236-39430] ../src/vabackend.c: 674           nvCreateConfig got profile: 19 with 0 attributes
      7326.198065293 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 10 (8) (nil) 0
      7326.198071976 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 10 (8) 0x7658c0a5a700 8
      7326.199985145 [39236-39430] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      7326.199997829 [39236-39430] ../src/vabackend.c: 674           nvCreateConfig got profile: 32 with 0 attributes
      7326.200008810 [39236-39430] ../src/vabackend.c: 737           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
      7326.200013458 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 11 (8) (nil) 0
      7326.200016254 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 11 (8) 0x7658c0a5ab80 8
      7326.201609429 [39236-39430] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      7326.201619578 [39236-39430] ../src/vabackend.c: 674           nvCreateConfig got profile: 21 with 0 attributes
      7326.201623636 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 10 (10) (nil) 0
      7326.201626251 [39236-39430] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 10 (10) 0x7658c0a5a280 8
      7326.203265303 [39236-39430] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      7326.203563676 [39236-39430] ../src/vabackend.c:2098              nvTerminate Terminating 0x7658f8c263e0
      7326.203650008 [39236-39430] ../src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances
      7326.230995234 [39236-39436] ../src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      7326.231015903 [39236-39436] ../src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
      7326.231020351 [39236-39436] ../src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
      7326.231046100 [39236-39436] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
      7326.231071598 [39236-39436] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 550.78, major version: 550, minor version: 78
[... cut ...]
      7349.736557678 [39236-39497] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7658ae6522e0)
      7349.736559571 [39236-39497] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7658ae6513e0)
      7349.736561104 [39236-39497] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7658ae652380)
      7349.736569089 [39236-39497] ../src/vabackend.c:2098              nvTerminate Terminating 0x7658f8c267a0
      7349.736784055 [39236-39497] ../src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances
```

Latrolage, 2024-05-11:
I haven't fixed the issue but I can't test out the issue conveniently anymore, I went from 2x nvidia GPU to 1x amd+ 1x nvidia and hardware decode on the nvidia GPU is working now so i'm going to close this issue unless someone else has the same issue.

closed 2024-05-11 by Latrolage