elFarto/nvidia-vaapi-driver #292: playback with mpv failing with nvidia 550.78 drivers

philipl, 2024-05-06:
Hi. I know that mpv isn't actually a real use-case, and accelerated playback in firefox continues to work fine, but I'm seeing that playback no longer works in mpv, and I worry it's a sign of a more serious problem.

It appears to be failing to import dmabufs on the EGL side for the useful formats (NV12, P010, etc) and then it fails to decode (I think because the output surface is incompatible).

Maybe something has regressed on the driver side? It could be in mpv but I can't quickly test old versions here.

```
$ NVD_LOG=1 mpv --profile=nvidia-vaapi --vd-lavc-check-hw-profile=no juddertest_60.mp4 --pause
[auto_hwdec] Applying profile: nvidia
 (+) Video --vid=1 (*) (h264 1920x1072 60.000fps)
[vaapi] libva: vaGetDriverNames() failed with unknown libva error
     29788.152716920 [66915-66932] ../src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     29788.152722691 [66915-66932] ../src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
     29788.152725646 [66915-66932] ../src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
     29788.152732081 [66915-66932] ../src/direct/direct-export-buf.c:  68      direct_initExporter Searching for GPU: 0 0 128
     29788.152772338 [66915-66932] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
     29788.152778426 [66915-66932] ../src/direct/direct-export-buf.c:  68      direct_initExporter Searching for GPU: 0 0 129
     29788.152782770 [66915-66932] ../src/direct/direct-export-buf.c:  90      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD129
     29788.152785185 [66915-66932] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
     29788.152794601 [66915-66932] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 550.78, major version: 550, minor version: 78
     29788.152797161 [66915-66932] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 1 2 6
     29788.188938420 [66915-66932] ../src/vabackend.c:1446      nvQueryImageFormats In nvQueryImageFormats
     29788.206188344 [66915-66932] ../src/vabackend.c: 674           nvCreateConfig got profile: 0 with 0 attributes
     29788.206200332 [66915-66932] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 1 (8) (nil) 0
     29788.206202841 [66915-66932] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 1 (8) 0x7c9e091d8b00 8
     29788.206884195 [66915-66932] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4080, height: 16 - 4080
     29788.206898301 [66915-66932] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7c9e091d8d80)
     29788.206903146 [66915-66932] ../src/vabackend.c:1540            nvDeriveImage In nvDeriveImage
     29788.206907836 [66915-66932] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7c9e09838140)
     29788.206909698 [66915-66932] ../src/vabackend.c:1540            nvDeriveImage In nvDeriveImage
     29788.206931600 [66915-66932] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x7c9e0952f2a0 128x128 = 24576 bytes
     29788.206999669 [66915-66932] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 64 65 66
     29788.207008805 [66915-66932] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
     29788.207223970 [66915-66932] ../src/direct/direct-export-buf.c:  46                    debug [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: could not bind to DMA buffer

     29788.207245578 [66915-66932] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7c9e09838140)
     29788.207310177 [66915-66932] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 100 (0x7c9e09530820)
     29788.207312712 [66915-66932] ../src/vabackend.c:1540            nvDeriveImage In nvDeriveImage
     29788.207322459 [66915-66932] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x7c9e098382c0 128x128 = 49152 bytes
     29788.207372478 [66915-66932] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 64 65 66
     29788.207373926 [66915-66932] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
     29788.207523897 [66915-66932] ../src/direct/direct-export-buf.c:  46                    debug [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: could not bind to DMA buffer

     29788.207537607 [66915-66932] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7c9e09530820)
     29788.207591371 [66915-66932] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 1000 (0x7c9e09530a20)
     29788.207593599 [66915-66932] ../src/vabackend.c:1540            nvDeriveImage In nvDeriveImage
     29788.207599709 [66915-66932] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x7c9e0952fbe0 128x128 = 49152 bytes
     29788.207643748 [66915-66932] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 64 65 66
     29788.207645139 [66915-66932] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
     29788.207776682 [66915-66932] ../src/direct/direct-export-buf.c:  46                    debug [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: could not bind to DMA buffer

     29788.207788160 [66915-66932] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7c9e09530a20)
     29788.207838659 [66915-66932] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 4 (0x7c9e095305b0)
     29788.207840993 [66915-66932] ../src/vabackend.c:1540            nvDeriveImage In nvDeriveImage
     29788.207846779 [66915-66932] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x7c9e0952fda0 128x128 = 49152 bytes
     29788.207889480 [66915-66932] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 64 65 66
     29788.207890848 [66915-66932] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
     29788.208166071 [66915-66932] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7c9e095305b0)
     29788.208234756 [66915-66932] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7c9e091d8d80)
     29788.208238512 [66915-66932] ../src/vabackend.c: 674           nvCreateConfig got profile: -1 with 0 attributes
     29788.208239971 [66915-66932] ../src/vabackend.c: 679           nvCreateConfig Profile not supported: -1
[auto_hwdec] Applying profile: hwdec.vaapi
     29788.226458885 [66915-66915] ../src/vabackend.c: 674           nvCreateConfig got profile: 7 with 0 attributes
     29788.226469290 [66915-66915] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) (nil) 0
     29788.226471241 [66915-66915] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) 0x597eed529e80 8
     29788.227158356 [66915-66915] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
     29788.227168071 [66915-66915] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) (nil) -1
     29788.227170106 [66915-66915] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) 0x597eed529e80 8
     29788.227911943 [66915-66915] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
     29788.227930081 [66915-66915] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 1920x1072, format 1 (0x597eed54c570)
[... 15 lines cut ...]
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
     29788.251847760 [66915-66915] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 1920x1072, format 1 (0x597eede39c20)
     29788.252131466 [66915-66915] ../src/vabackend.c:1345             nvEndPicture cuvidDecodePicture failed: 1
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
[... cut ...]
     29788.499949800 [66915-66915] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 1 (0x597eede36be0)
     29788.499961558 [66915-66915] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 0 (0x597eed54c570)
VO: [gpu] 1920x1072 yuv420p
(Paused) V: 00:00:00 / 00:01:04 (0%) Cache: 63s/2MB
Exiting... (Quit)
     29789.486477689 [66915-66932] ../src/vabackend.c:2098              nvTerminate Terminating 0x7c9e08571e40
     29789.486932039 [66915-66932] ../src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances
```

elFarto, 2024-05-06:
I spotted this issue (a similar issue?) earlier while testing earlier. The move to a single buffer export seems to have interacted badly with the GOB calculation code (which ran per-plane), and mpv/ffmpeg does some small scale tests (128x128) to make sure it all works, but fails due to an invalid log2GobsPerBlockY value.

I really don't understand the `log2GobsPerBlockY` value, or how to compute it correctly, or if it's even compatible with single buffer export. I've pushed a change to master than hard codes it to 4, which works for most things except videos with ~64px high planes.

However, I've never seen it fail to decode, so try master and see if that fixes it, if not I'll have to dig a bit more into it.

philipl, 2024-05-06:
Yeah, your fix does stop the failures to import but the failures to decode are still there, sorry to say.

```
$ NVD_LOG=1 mpv --profile=nvidia-vaapi juddertest_60.mp4 --pause
[auto_hwdec] Applying profile: nvidia
 (+) Video --vid=1 (*) (h264 1920x1072 60.000fps)
[vaapi] libva: vaGetDriverNames() failed with unknown libva error
     34146.562078372 [69960-69977] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     34146.562083703 [69960-69977] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 0 (0 max) instances
     34146.562084788 [69960-69977] ../src/vabackend.c:2222       __vaDriverInit_1_0 Selecting Direct backend
     34146.562089785 [69960-69977] ../src/direct/direct-export-buf.c:  68      direct_initExporter Searching for GPU: 0 0 128
     34146.562133430 [69960-69977] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
     34146.562137743 [69960-69977] ../src/direct/direct-export-buf.c:  68      direct_initExporter Searching for GPU: 0 0 129
     34146.562141055 [69960-69977] ../src/direct/direct-export-buf.c:  90      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD129
     34146.562142610 [69960-69977] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
     34146.562149838 [69960-69977] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 550.78, major version: 550, minor version: 78
     34146.562151802 [69960-69977] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 1 2 6
     34146.607554351 [69960-69977] ../src/vabackend.c:1445      nvQueryImageFormats In nvQueryImageFormats
     34146.623883194 [69960-69977] ../src/vabackend.c: 674           nvCreateConfig got profile: 0 with 0 attributes
     34146.623894495 [69960-69977] ../src/vabackend.c:1801 nvQuerySurfaceAttributes with 1 (8) (nil) 0
     34146.623897119 [69960-69977] ../src/vabackend.c:1801 nvQuerySurfaceAttributes with 1 (8) 0x79dc611d9340 8
     34146.624602628 [69960-69977] ../src/vabackend.c:1868 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4080, height: 16 - 4080
     34146.624616304 [69960-69977] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x79dc611d9540)
     34146.624620397 [69960-69977] ../src/vabackend.c:1539            nvDeriveImage In nvDeriveImage
     34146.624624720 [69960-69977] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x79dc6152f4c0)
     34146.624626340 [69960-69977] ../src/vabackend.c:1539            nvDeriveImage In nvDeriveImage
     34146.624648872 [69960-69977] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x79dc6152ff20 128x128 = 32768 bytes
     34146.624716249 [69960-69977] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 64 65 66
     34146.624717709 [69960-69977] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
     34146.625088507 [69960-69977] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x79dc6152f4c0)
     34146.625173013 [69960-69977] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 100 (0x79dc6152f4c0)
     34146.625178440 [69960-69977] ../src/vabackend.c:1539            nvDeriveImage In nvDeriveImage
     34146.625187926 [69960-69977] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x79dc61531620 128x128 = 65536 bytes
     34146.625253794 [69960-69977] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 64 65 66
     34146.625255280 [69960-69977] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
     34146.625541598 [69960-69977] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x79dc6152f4c0)
     34146.625609942 [69960-69977] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 1000 (0x79dc615332e0)
     34146.625613789 [69960-69977] ../src/vabackend.c:1539            nvDeriveImage In nvDeriveImage
     34146.625621802 [69960-69977] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x79dc6152f4e0 128x128 = 65536 bytes
     34146.625676212 [69960-69977] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 64 65 66
     34146.625678070 [69960-69977] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
     34146.625892240 [69960-69977] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x79dc615332e0)
     34146.625948227 [69960-69977] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 4 (0x79dc6152f600)
     34146.625951004 [69960-69977] ../src/vabackend.c:1539            nvDeriveImage In nvDeriveImage
     34146.625957259 [69960-69977] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x79dc61532f20 128x128 = 49152 bytes
     34146.626004504 [69960-69977] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 64 65 66
     34146.626006082 [69960-69977] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
[... cut ...]
     34146.919460597 [69960-69960] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 3 (0x6266aad03ce0)
     34146.919494576 [69960-69960] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 2 (0x6266aad03620)
     34146.919502235 [69960-69960] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 1 (0x6266aad00de0)
     34146.919509923 [69960-69960] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 0 (0x6266aa4e6700)
VO: [gpu] 1920x1072 yuv420p
(Paused) V: 00:00:00 / 00:01:04 (0%) Cache: 63s/2MB
```

elFarto, 2024-05-07:
I can play files with mpv (v0.36.0) on my machine, both h264 and hevc including 10-bit videos with that NVIDIA driver version, so it's not an inherent problem with the driver.

Any chance you've got some stray mpv config item set?

philipl, 2024-05-08:
It's an ffmpeg change. I did a bisect and it came down to:

https://github.com/FFmpeg/FFmpeg/commit/41e3d36a39979a5c6ca36198b03be740e14ef7b0

which hides a lot of implications.

elFarto, 2024-05-08:
Ok, looking at the log output after looking at the patch, I see the issue. FFMPEG has switched to creating surfaces dynamically, rather than all upfront. This actually wouldn't be a problem if it created 0 surfaces before initialising it as we'd catch that and allocate a fixed amount, but it does allocate 1, which trips us up.

This is problematic, as NVDEC just doesn't work that way. I guess we're going to have to allocate some minimum amount and hope for the best...? I'd be hesitant in allocating too many as those 4k HDR frames are already quite large.

We may need to rethink the decode process. Maybe we could allocate a smaller pool of frames and preemptively copy them to the CUDA images we create, but I think we'd just run into the issue we had early on with not knowing which frames are needed when.

philipl, 2024-05-08:
Is it possible to treat the n=1 case the same as n=0? No real world decode scenario can actually work with just one decode surface, so we know it's asking for the same dynamic behaviour as n=0.

I separately wonder if it's possible to reinitialise the decoder when the number of surfaces changes. I know there is a reconfigure operation in nvdec now but I don't know if it lets you change the number of surfaces. But even if not, can a full reinit be done without killing playback pacing?

philipl, 2024-05-08:
As for vdpau, it might be possible to use some of the undocumented kernel ioctls to get the underlying surface resource and work with it like the direct backend does today, but that sounds scary to me...

elFarto, 2024-05-08:
I did investigate the reconfigure option early on, unfortunately it helpfully destroys/reallocates all the existing surfaces for you so it's not usable without visual artefacts.

For VDPAU 'all' that's needed is a few handles that get allocated to the library and the surface (we might also need the fd that it's using), once we have those we could reuse the export code we already have (in theory at least). However my previous attempts to find where it stored those handles didn't get anywhere. I think we're need to fully decompile the library rather than just trying to step through it in the debugger.

I don't think it's as scary as you think, alot of the direct backend's code is based on how VDPAU created surfaces, so I already know they're very similar.

PR #295 (2024-05-10): vabackend: Treat surfaceCount == 1 the same as surfaceCount == 0
When Chrome(ium) wants dynamic surface allocation it starts with an initial count of 0, and that is handled today. On the other hand, recent versions of ffmpeg have switched to preferring dynamic allocation but they start with a surfaceCount of 1. If we treat this as a legitimate static request, we cannot do any decoding in practice, as you always need more than 1 surface.

So, treat a surfaceCount of 1 the same as 0, and use our same dynamic allocation work-around.

Fixes #292
closed 2024-05-10 by elFarto