elFarto/nvidia-vaapi-driver #297: Doesn't appear to work on Ubuntu 22.04/Nvidia Driver 535.171.04/RTX 2060/Firefox 126

C-Lunn, 2024-05-25:
Hi,

Steps followed: 
1. Edit `/etc/default/grub` to set `GRUB_CMDLINE_LINUX="nvidia-drm.modeset=1"` and run `grub-update` then restart.
2. Clone, configure, and compile, allowing `sudo` access when prompted.
3. Observe the `nvidia_drv_video.so` file has been successfully placed in the `/usr/lib/x86_64-linux-gnu/dri` directory.
4. Observe the output of `ffmpeg -hwaccels` to be:
```
Hardware acceleration methods:
vdpau
cuda
vaapi
qsv
drm
opencl
```
5. Edit /etc/environment to have:
```
MOZ_DISABLE_RDD_SANDBOX=1
LIBVA_DRIVER_NAME=nvidia
NVD_BACKEND=direct
```
6. Restart session, observe those env variables are set.
7. Go to about:config in firefox, and set the following variables:
```
media.ffmpeg.vaapi.enabled: TRUE
media.rdd-ffmpeg.enabled: already TRUE
gfx.x11-egl.force-enabled: TRUE
widget.dmabuf.force-enabled: TRUE
```
8. Restart whole PC just to be sure

Observation: It doesn't use the GPU. Without even looking at `nvidia-smi` I'm getting 75+% CPU usage on any given YouTube video, but well, `nvidia-smi` does not report a Firefox process.

The output of `./test.sh`: https://gist.github.com/C-Lunn/2f3b1f62b956e3100b5a0bb5f5b44a2a
`about: support`: https://gist.github.com/C-Lunn/e1711ca8daad9631f46b21afc158e7b7

Any ideas? Thanks!

TsunamiMommy, 2024-05-28:
Are you by chance on a laptop? The gpu driver strings in your `about:support` indicate that you have your integrated gpu active. The nvidia card isn't active so that is why you don't see hardware decoding. You can post the output of `vainfo` here but it will likely tell you that you don't have your nvidia card active, but rather your integrated gpu active instead.

C-Lunn, 2024-06-03:
> Are you by chance on a laptop? The gpu driver strings in your `about:support` indicate that you have your integrated gpu active. The nvidia card isn't active so that is why you don't see hardware [...]

I was on a desktop with a Ryzen 3600X which I don't believe even has an iGPU. Life circumstances have meant I'm currently without that particular setup. I'll close it for now, but if I get in a similar situation at some point in the future I'll re-open. Cheers.

closed 2024-06-03 by C-Lunn

carlinux, 2024-06-17:
Since you're on ubuntu it may be that you're using the snap version of FF.. (it think it doesn't work with the snap version)  and wayland.. it doesn't work well on wayland either but at least it should show up in nvidia-smi