elFarto/nvidia-vaapi-driver #300: NVK support, is it even possible?

Dungeonseeker, 2024-06-04:
Been testing NVK for a few days now and its pretty good, not as good as the open or proprietary drivers yet but its certainly at a point where it runs a desktop environment with no issues.

It would be really nice to get hardware video acceleration support working as right now, if I switch to the NVIDIA GPU on my (hybrid) laptop it uses 40% cpu just watching a YT video and burns though a full battery in about an hour and a half. Sure for me I can just remember to swap back to hybrid mode and use VAAPI with intel-media-driver but for those without that option, having no HW accel would probably be a pretty big barrier, heck even without a battery to worry about it makes my fan run constantly and the CPU gets pretty toasty.

I'm also aware that this driver relies pretty heavily on stuff from the FOSS drivers and uses some of the nvidia driver binaries as well so instead of asking you to do it, I thought it would be a good idea to ask is it even possible at all?

thesword53, 2024-06-04:
This driver is based on NVIDIA NVDEC API which isn't supported by the Nouveau driver. To support VAAPI with NVK it would be better to implement a generic (not limited to NVIDIA) VAAPI driver over Vulkan such as https://gitlab.freedesktop.org/airlied/mesa/-/commits/zink-video-wip

Dungeonseeker, 2024-06-04:
> This driver is based on NVIDIA NVDEC API which isn't supported by the Nouveau driver. To support VAAPI with NVK it would be better to implement a generic (not limited to NVIDIA) VAAPI driver over Vulkan such as https://gitlab.freedesktop.org/airlied/mesa/-/commits/zink-video-wip

Figured something like that was the case, thanks anyway.

closed 2024-06-04 by Dungeonseeker

lufog, 2026-01-31:
Is it possible, in theory, to run this driver with patched nouveau ([nouveau: Expose the nvdec engine](https://gitlab.freedesktop.org/gfxstrand/linux/-/commit/98d56182eb223d5988219137d0bae66b26435a32))? Unfortunately, as far as I understand, work on adding hardware acceleration for Nvidia cards in Mesa ([NVK Vulkan Video support](https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/31867) / [zink: add initial video decoding support (h264)](https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/31517)) has been stalled indefinitely...