elFarto/nvidia-vaapi-driver #301: Confusion about "Video Engine Utilization"

akorb, 2024-06-04:
First, thanks for your amazing work, I appreciate that you spend your time on this.

Now, I installed [libva-nvidia-driver](https://archlinux.org/packages/?name=libva-nvidia-driver) on my Manjaro system and enabled VA-API (media.ffmpeg.vaapi.enabled). That's all I did. I'm on driver 550.78, X11, and an NVIDIA GeForce RTX 4070 SUPER.

Thereby, I receive this result in `about:support`, which is great:
[image]

However, the **GPU Utilization** and **Video Engine Utilization** metrics in the NVIDIA settings behave differently depending on whether I set `MOZ_DISABLE_RDD_SANDBOX=1`.

[image]

For the following values, I used the YouTube video [Norway 4K](https://youtu.be/KLuTLF3x9sA?si=I39iow75Y2Nu-BEs) as reference (2160p, AV1).

## `MOZ_DISABLE_RDD_SANDBOX=0 firefox`

Keeping the RDD sandbox enabled results in reoccuring console outputs like this

> libva error: /usr/lib/dri/nvidia_drv_video.so init failed

GPU Utilization: 10 - 25%
Video Engine Utilization: 0%

##  `MOZ_DISABLE_RDD_SANDBOX=1 firefox`

No console output.
GPU Utilization: ~2%
Video Engine Utilization: 5-10%

---

What confuses me is that firefox claims either way that HW Decoding support is enabled, and that the GPU is always somehow used (either as GPU or Video Engine) when streaming a video.

My first wild theory is that with the RDD sandbox enabled the VA-API driver uses the direct backend, which consequently uses CUDA, which is not represented in **Video Engine Utilization** but **GPU Utilization**.

My second theory is that the table HW support table in `about:support` as attached at the top is unreliable. Maybe firefox detects a VA-API driver and therefore thinks HW decoding is available, even though this VA-API driver cannot actually launch the hardware decoding because the sandbox is still enabled.

I'm asking because overall, if possible, I'd prefer to leave the RDD sandbox enabled, since it has security benefits. And if both options are valid  `MOZ_DISABLE_RDD_SANDBOX=*`, I'd not disable the sandbox.

akorb, 2024-06-07:
I think it's really that firefox shows the table wrong. When I enable the logs (`NVD_LOG=1`), it shows:
```
init ERROR: Potential Firefox sandbox detected, failing to init!
init If running in Firefox, set env var MOZ_DISABLE_RDD_SANDBOX=1 to disable sandbox.
```

KaeLL, 2024-08-16:
I use `nvtop` to check usage stats.

elFarto, 2024-09-09:
It's not possible to run with the sandbox enabled. CUDA is just not designed to run in such an environment, and we'd have to poke so many holes in the sandbox to make it work, it wouldn't be worth it.

As to you first question, I believe this is just a quirk with Firefox although I confess I'm not quite sure what's happening. For this driver to tell you what codecs are supported it must init the GPU, which can't be done under the sandbox. This likely means that the process that checks which codecs are available isn't running under the RDD sandbox (as it's not decoding video that's not really a risk).

closed 2024-09-09 by elFarto