elFarto/nvidia-vaapi-driver #304: Is HEVC decoding from Firefox Supported?

meriley, 2024-06-11:
I am having trouble playing HEVC files in Firefox. Based on this project I feels like HEVC should be supported?

I am on Fedora 39, with Nvidia Driver 555.42.02. I have built the driver from the Master branch (was testing MPV, was the same on 0.0.12).

```
NVD_LOG=1 vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.20.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
     16971.744036279 [43845-43845] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     16971.744039689 [43845-43845] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 0 (0 max) instances
     16971.744041559 [43845-43845] ../src/vabackend.c:2222       __vaDriverInit_1_0 Selecting Direct backend
     16971.749821102 [43845-43845] ../src/direct/direct-export-buf.c:  68      direct_initExporter Searching for GPU: 0 0 128
     16971.749835092 [43845-43845] ../src/direct/direct-export-buf.c:  90      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
     16971.749838022 [43845-43845] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
     16971.749850532 [43845-43845] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 555.42.02, major version: 555, minor version: 42
     16971.749853852 [43845-43845] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 2d00 1 2 6
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.20 (libva 2.20.1)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
     16971.826287808 [43845-43845] ../src/vabackend.c:2097              nvTerminate Terminating 0x5605b0837400
     16971.826360819 [43845-43845] ../src/vabackend.c:2111              nvTerminate Now have 0 (0 max) instances
```

When attempting to play HEVC I see errors

```
[Child 43806: MediaSupervisor #3]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type video/hevc
[Child 43806: MediaSupervisor #3]: D/PlatformDecoderModule Sandbox Utility Generic decoder rejects requested type video/hevc
[Child 43806, MediaDecoderStateMachine #1] WARNING: Decoder=7f491cc7c200 state=DECODING_METADATA Decode metadata failed, shutting down decoder: file /builddir/build/BUILD/firefox-126.0/dom/media/MediaDecoderStateMachine.cpp:372
[Child 43806, MediaDecoderStateMachine #1] WARNING: Decoder=7f491cc7c200 Decode error: NS_ERROR_DOM_MEDIA_METADATA_ERR (0x806e0006): file /builddir/build/BUILD/firefox-126.0/dom/media/MediaDecoderStateMachineBase.cpp:167
[Child 43806, MediaDecoderStateMachine #1] WARNING: Decoder=7f491f9f4800 state=DECODING_METADATA Decode metadata failed, shutting down decoder: file /builddir/build/BUILD/firefox-126.0/dom/media/MediaDecoderStateMachine.cpp:372
[Child 43806, MediaDecoderStateMachine #1] WARNING: Decoder=7f491f9f4800 Decode error: NS_ERROR_DOM_MEDIA_METADATA_ERR (0x806e0006): file /builddir/build/BUILD/firefox-126.0/dom/media/MediaDecoderStateMachineBase.cpp:167
```

akorb, 2024-06-11:
Might also fail with your graphics card. Not all graphics card support HEVC decoding. You can check it here: https://developer.nvidia.com/video-encode-and-decode-gpu-support-matrix-new

meriley, 2024-06-11:
I have a GTX3080ti, I'm full green on that chart.

thesword53, 2024-06-12:
Firefox doesn't support HEVC decoding at all on Linux even using software decoding.

mirh, 2024-07-15:
That's a 2017 answer that [doesn't](https://bugzilla.mozilla.org/show_bug.cgi?id=1889731) really cover the latest progress. 

https://bugzilla.mozilla.org/show_bug.cgi?id=1894818
This is the bug you want to track.

elFarto, 2024-09-09:
I've updated the README with a note on HEVC.

closed 2024-09-09 by elFarto

mirh, 2024-10-25:
There is no HEVC support at all on linux (regardless of system codecs, even though those will be required) it is just planned.