elFarto/nvidia-vaapi-driver #305: widget.dmabuf.force-enabled = true no longer required.

TsunamiMommy, 2024-06-12:
Hi there. I would like to report that in recent firefox versions and with the nvidia 550 and up drivers `widget.dmabuf.force-enabled = true` is no longer required for hardware video decoding and enabling it will cause lots of graphical corruption and glitching on 3D web applications. This is with all other required parameters enabled. I just thought that I would let you know. You can test this to comfirm that this is the case for your hardware. I currently have an RTX 4080 running Cinnamon on X11.

TsunamiMommy, 2024-06-23:
It is an issue exclusive to Nvidia and EGL with widget.dmabuff enabled. I couldn't track down an official reason as to why it happens but the corruption and glitching is mostly related to bad buffer updates with certain web apps and the firefox built in PDF reader. Essentially to trigger the bug I had Desmos, several PDF pages open, and youtube at the same time. It would take a bit of alt-tabbing but what would eventually happen was that the PDF pages were displayed in an incorrect order and even the pages themselves wouldn't be from the same PDF. Desmos would stop updating the graphical contents of the app (so moving graphs around was impossible as it wouldn't update the image to represent any transformations or movements on the graphs). I did extensive testing and it was `widget.dmabuff.force=true` that was causing the issue. It turns out that it wasn't required to enable hardware video decoding with vaapi on my system. So the obvious recommendation was to simply disable it. I'm not sure about wayland because my curreent setup can't use hardware acceleration due to firefox being incompatible with explicit sync atm. My current setup is Cinnamon on Mint 21.3, Nvidia 555.42.02, and 6.6.30-rt30-x64v3-xanmod1. Hopefully that answers your question.

elFarto, 2024-06-23:
Looks like it's still needed on my setup (550 driver), but there does seem to be some activity on the [issue](https://bugzilla.mozilla.org/show_bug.cgi?id=1788573) indicating that they might be reverting the block.

ManuLinares, 2024-07-03:
I've disabled widget.dmabuff.force, seems to work fine. I've had some issues with pdfs on Firefox, didn't know could be related.

Archlinux
linux 6.9.7.arch1-1
libva-nvidia-driver-git 0.0.12.r11.gf3205bd-1
nvidia-dkms 555.58-2

elFarto, 2024-09-09:
It does seem that this flag is no longer required now.

elFarto, 2025-08-07:
I've removed the note in the README about this being required.