elFarto/nvidia-vaapi-driver #306: Videos occasionally skip several seconds

phonetic112, 2024-06-13:
On YouTube, videos occasionally skip forward several seconds. I've only observed this with VP9 so far. Using an RTX 4070.

Clutchnp, 2024-06-18:
@phonetic112 Can confirm , I am currently also testing with avc1 which is not giving me problems for now

Clutchnp, 2024-06-19:
Update: This dosen't seem like a hardware acceleration issue on my end , if youtube videos are not accelerated on firefox this still happens,whereas on chomium based browsers, this dosen't happen because default codec youtube sends on those browsers is av1. Whereas Firefox receives vp9 by default.

phonetic112, 2024-06-19:
you're right, I can repro with software decoding as well. I'll close this then

closed 2024-06-19 by phonetic112