elFarto/nvidia-vaapi-driver #308: nvidia_drv_video.so init failed

KaeLL, 2024-06-28:
```
Operating System: Fedora Linux 40
KDE Plasma Version: 6.1.1
KDE Frameworks Version: 6.3.0
Qt Version: 6.7.1
Kernel Version: 6.9.5-200.fc40.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 5800X3D 8-Core Processor
Memory: 31.3 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 4090/PCIe/SSE2
Manufacturer: Micro-Star International Co., Ltd.
Product Name: MS-7C91
System Version: 2.0
```
```
Firefox 127.0.2
```
[image]
[image]
[image]
[image]
^ Tried enabling it and starting firefox with MOZ_X11_EGL=1, nothing changed
[image]

```
$ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo
Trying display: wayland
libva info: VA-API version 1.21.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri-nonfree/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
13763.246588057 [44973-44973] ../src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
13763.246592077 [44973-44973] ../src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
13763.246595557 [44973-44973] ../src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
13763.253190935 [44973-44973] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
13763.253213935 [44973-44973] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 550.90.07, major version: 550, minor version: 90
13763.253219355 [44973-44973] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 2b00 1 2 6
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.21 (libva 2.21.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
VAProfileMPEG2Simple            : VAEntrypointVLD
VAProfileMPEG2Main              : VAEntrypointVLD
VAProfileVC1Simple              : VAEntrypointVLD
VAProfileVC1Main                : VAEntrypointVLD
VAProfileVC1Advanced            : VAEntrypointVLD
VAProfileH264Main               : VAEntrypointVLD
VAProfileH264High               : VAEntrypointVLD
VAProfileH264ConstrainedBaseline: VAEntrypointVLD
VAProfileHEVCMain               : VAEntrypointVLD
VAProfileVP8Version0_3          : VAEntrypointVLD
VAProfileVP9Profile0            : VAEntrypointVLD
VAProfileAV1Profile0            : VAEntrypointVLD
VAProfileHEVCMain10             : VAEntrypointVLD
VAProfileHEVCMain12             : VAEntrypointVLD
VAProfileVP9Profile2            : VAEntrypointVLD
VAProfileHEVCMain444            : VAEntrypointVLD
VAProfileHEVCMain444_10         : VAEntrypointVLD
VAProfileHEVCMain444_12         : VAEntrypointVLD
13763.340359154 [44973-44973] ../src/vabackend.c:2098              nvTerminate Terminating 0x55f8e3edf570
13763.340420164 [44973-44973] ../src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances
```

```
$ MOZ_GFX_DEBUG=1 firefox
GLX_TEST: childgltest start
GLX_TEST: get_pci_status start
GLX_TEST: get_pci_status finished
GLX_TEST: wayland_egltest start
GLX_TEST: get_egl_status start
GLX_TEST: get_egl_gl_status start
GLX_TEST: get_egl_gl_status finished
GLX_TEST: get_egl_status finished with return: 1
GLX_TEST: wayland_egltest finished
PCI_VENDOR_ID
0x10de
PCI_DEVICE_ID
0x2684
VENDOR
NVIDIA Corporation
RENDERER
NVIDIA GeForce RTX 4090/PCIe/SSE2
VERSION
4.6.0 NVIDIA 550.90.07
TFP
TRUE
DRM_RENDERDEVICE
/dev/dri/renderD128
MESA_VENDOR_ID
0x10de
MESA_DEVICE_ID
0x2684
TEST_TYPE
EGL
GLX_TEST: childgltest finished
vaapitest start, device /dev/dri/renderD128
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib64/dri-nonfree/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vaInitialize finished
Profile: H264Main
Profile: H264High
Profile: H264ConstrainedBaseline
Profile: VP8Version0_3
Profile: VP9Profile0
Profile: AV1Profile0
Profile: VP9Profile2
vaapitest finished
VAAPI_SUPPORTED
TRUE
VAAPI_HWCODECS
240
```
Then I open youtube and wether I start a video or just hover over it, I get an endless stream of the following:
```
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib64/dri-nonfree/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva error: /usr/lib64/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
```

jackdaw213, 2024-07-01:
Can you try setting MOZ_DISABLE_RDD_SANDBOX env variable to either 0 or 1 as described in [my other comment](https://github.com/elFarto/nvidia-vaapi-driver/issues/259#issuecomment-2199252508) ?

KaeLL, 2024-07-01:
I'm stupid, I missed that in the README.md
Sorry about that. Seems it's working now.

<details>
<summary>$ MOZ_DISABLE_RDD_SANDBOX=0 NVD_LOG=1 MOZ_GFX_DEBUG=1 firefox</summary>

```
GLX_TEST: childgltest start
GLX_TEST: get_pci_status start
GLX_TEST: get_pci_status finished
GLX_TEST: wayland_egltest start
GLX_TEST: get_egl_status start
GLX_TEST: get_egl_gl_status start
GLX_TEST: get_egl_gl_status finished
GLX_TEST: get_egl_status finished with return: 1
GLX_TEST: wayland_egltest finished
PCI_VENDOR_ID
0x10de
PCI_DEVICE_ID
0x2684
VENDOR
NVIDIA Corporation
RENDERER
NVIDIA GeForce RTX 4090/PCIe/SSE2
VERSION
4.6.0 NVIDIA 550.90.07
TFP
TRUE
DRM_RENDERDEVICE
/dev/dri/renderD128
MESA_VENDOR_ID
0x10de
MESA_DEVICE_ID
0x2684
TEST_TYPE
EGL
GLX_TEST: childgltest finished
vaapitest start, device /dev/dri/renderD128
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib64/dri-nonfree/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vaInitialize finished
Profile: H264Main
Profile: H264High
Profile: H264ConstrainedBaseline
Profile: VP8Version0_3
Profile: VP9Profile0
Profile: AV1Profile0
Profile: VP9Profile2
vaapitest finished
VAAPI_SUPPORTED
TRUE
VAAPI_HWCODECS
240
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib64/dri-nonfree/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
3494.574828208 [11521-11533] ../src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
3494.574832678 [11521-11533] ../src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
3494.574835418 [11521-11533] ../src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
3494.581671534 [11521-11533] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
3494.581692553 [11521-11533] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 550.90.07, major version: 550, minor version: 90
3494.581697913 [11521-11533] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 2b00 1 2 6
libva info: va_openDriver() returns 0
3494.637656968 [11521-11533] ../src/vabackend.c:1446      nvQueryImageFormats In nvQueryImageFormats
[... 610 lines cut ...]
3532.780789395 [11521-11723] ../src/vabackend.c: 458          resolveSurfaces [RT] Resolve thread for 0x7f70aa711000 exiting
3532.780823205 [11521-11695] ../src/vabackend.c: 333           destroyContext pthread_timedjoin_np finished with 0
3532.822234756 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 3 (0x7f70aa6e4a20)
3532.822375145 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 9 (0x7f70a9e3de20)
3532.822685242 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 0 (0x7f70aa6e48e0)
3532.822998548 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 7 (0x7f70a9e3dec0)
3532.823306025 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 2 (0x7f70aa6e5060)
3532.823609742 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 8 (0x7f70a9e3dba0)
3532.823773920 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 1 (0x7f70aa6e45c0)
3532.824097226 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 10 (0x7f70a9e3dd80)
3532.824390203 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 6 (0x7f70a9e3df60)
3532.824515222 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 5 (0x7f70aa6e4de0)
3532.824827748 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 4 (0x7f70aa6e40c0)
3532.825197864 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f70a9e3da60)
3532.825202344 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f70a9e3cb60)
3532.825204544 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f70a9e3cfc0)
3532.825207584 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f70a9e3c840)
3532.825210124 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f70a9e3cf20)
3532.825212394 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f70a9e3c3e0)
3532.825214864 [11521-11695] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f70a9e3cd40)
3532.825220144 [11521-11695] ../src/vabackend.c:2098              nvTerminate Terminating 0x7f70e0f35e20
3532.825294333 [11521-11695] ../src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances
```

</details>

nerijus, 2024-07-01:
Please close the issue then.

closed 2024-07-02 by KaeLL