elFarto/nvidia-vaapi-driver #312: The driver doesn't work with NVIDIA 560 drivers

thesword53, 2024-07-23:
```
 (+) Video --vid=1 (*) (h264 1280x720 30.151fps)
 (+) Audio --aid=1 --alang=eng (*) (aac 2ch 44100Hz)
       429.646350406 [2028-2038] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
       429.646363003 [2028-2038] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 0 (0 max) instances
       429.646365442 [2028-2038] ../src/vabackend.c:2222       __vaDriverInit_1_0 Selecting Direct backend
       429.646380561 [2028-2038] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
       429.646404318 [2028-2038] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 560.28.03, major version: 560, minor version: 28
       429.646408725 [2028-2038] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 0 fe 1
       429.710210911 [2028-2038] ../src/vabackend.c:1445      nvQueryImageFormats In nvQueryImageFormats
       429.745447511 [2028-2038] ../src/vabackend.c: 674           nvCreateConfig got profile: 0 with 0 attributes
       429.745463770 [2028-2038] ../src/vabackend.c:1801 nvQuerySurfaceAttributes with 1 (8) (nil) 0
       429.745470403 [2028-2038] ../src/vabackend.c:1801 nvQuerySurfaceAttributes with 1 (8) 0x7a566d2a87c0 8
       429.746751013 [2028-2038] ../src/vabackend.c:1868 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4080, height: 16 - 4080
       429.746767859 [2028-2038] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7a566d2a8580)
       429.746775235 [2028-2038] ../src/vabackend.c:1539            nvDeriveImage In nvDeriveImage
       429.746783831 [2028-2038] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7a566d2a7df0)
       429.746787640 [2028-2038] ../src/vabackend.c:1539            nvDeriveImage In nvDeriveImage
       429.746817466 [2028-2038] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x7a566d2a8e90 128x128 = 32768 bytes
       429.746945074 [2028-2038] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 65 66 67
       429.746949925 [2028-2038] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
       429.747595780 [2028-2038] ../src/direct/direct-export-buf.c:  46                    debug [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: requested buffer attributes are not supported

       429.747613731 [2028-2038] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7a566d2a7df0)
       429.747752505 [2028-2038] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 100 (0x7a566d2a5fb0)
       429.747759146 [2028-2038] ../src/vabackend.c:1539            nvDeriveImage In nvDeriveImage
       429.747771677 [2028-2038] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x7a566d2a7f40 128x128 = 65536 bytes
       429.747895264 [2028-2038] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 65 66 67
       429.747900483 [2028-2038] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
       429.748108070 [2028-2038] ../src/direct/direct-export-buf.c:  46                    debug [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: requested buffer attributes are not supported

       429.748125210 [2028-2038] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7a566d2a5fb0)
       429.748250667 [2028-2038] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 1000 (0x7a566d2a9210)
       429.748257443 [2028-2038] ../src/vabackend.c:1539            nvDeriveImage In nvDeriveImage
       429.748272985 [2028-2038] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x7a566d2a48e0 128x128 = 65536 bytes
       429.748389469 [2028-2038] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 65 66 67
       429.748394120 [2028-2038] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
       429.748602392 [2028-2038] ../src/direct/direct-export-buf.c:  46                    debug [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: requested buffer attributes are not supported

       429.748625613 [2028-2038] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7a566d2a9210)
       429.748751639 [2028-2038] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 4 (0x7a566d2a4980)
       429.748758014 [2028-2038] ../src/vabackend.c:1539            nvDeriveImage In nvDeriveImage
       429.748771743 [2028-2038] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x7a566d2a7da0 128x128 = 49152 bytes
       429.748875033 [2028-2038] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 65 66 67
       429.748879707 [2028-2038] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
       429.749113182 [2028-2038] ../src/direct/direct-export-buf.c:  46                    debug [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: requested buffer attributes are not supported

       429.749128469 [2028-2038] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7a566d2a4980)
       429.749272106 [2028-2038] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7a566d2a8580)
       429.749278381 [2028-2038] ../src/vabackend.c: 674           nvCreateConfig got profile: 1 with 0 attributes
       429.749284339 [2028-2038] ../src/vabackend.c:1801 nvQuerySurfaceAttributes with 1 (8) (nil) 0
       429.749288333 [2028-2038] ../src/vabackend.c:1801 nvQuerySurfaceAttributes with 1 (8) 0x7a566d2a87c0 8
       429.751379750 [2028-2038] ../src/vabackend.c:1868 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4080, height: 16 - 4080
       429.751395147 [2028-2038] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7a566d2a7380)
       429.751401003 [2028-2038] ../src/vabackend.c:1539            nvDeriveImage In nvDeriveImage
       429.751407748 [2028-2038] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7a566d2a98f0)
       429.751411906 [2028-2038] ../src/vabackend.c:1539            nvDeriveImage In nvDeriveImage
       429.751426083 [2028-2038] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x7a566d2a6920 128x128 = 32768 bytes
       429.751542426 [2028-2038] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 65 66 67
       429.751547382 [2028-2038] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
       429.751749392 [2028-2038] ../src/direct/direct-export-buf.c:  46                    debug [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: requested buffer attributes are not supported

       429.751765584 [2028-2038] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7a566d2a98f0)
       429.751885532 [2028-2038] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 100 (0x7a566cf9c780)
       429.751891621 [2028-2038] ../src/vabackend.c:1539            nvDeriveImage In nvDeriveImage
       429.751905266 [2028-2038] ../src/direct/direct-export-buf.c: 151 direct_allocateBackingImage Allocating BackingImage: 0x7a566d2a7da0 128x128 = 65536 bytes
       429.752010078 [2028-2038] ../src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 65 66 67
       429.752014833 [2028-2038] ../src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA
       429.752234388 [2028-2038] ../src/direct/direct-export-buf.c:  46                    debug [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: requested buffer attributes are not supported

[... 391 lines cut ...]
       429.790453426 [2028-2038] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 128x128, format 1000 (0x7a566d2a8500)
[... cut ...]
       429.791460865 [2028-2038] ../src/vabackend.c:2097              nvTerminate Terminating 0x7a566c670cc0
       429.791516276 [2028-2038] ../src/vabackend.c:2111              nvTerminate Now have 0 (0 max) instances
AO: [pipewire] 44100Hz stereo 2ch floatp
VO: [gpu] 1280x720 yuv420p
AV: 00:00:01 / 00:00:06 (26%) A-V:  0.000
Exiting... (Quit)
       432.035976501 [2028-2038] ../src/direct/direct-export-buf.c:  46                    debug [EGL] eglDestroySurface: EGL_BAD_SURFACE error: In internal function: Additional INFO may be available
```

TheBigK02, 2024-08-05:
can confirm this behavior. I downgraded to 555 which works.

elFarto, 2024-08-10:
The driver is working for me with 560.31.02 installed. Can you recheck with this version?

wxlyyy, 2024-08-10:
```
➜  ~ vainfo
Trying display: wayland
libva info: VA-API version 1.22.0
libva info: Trying to open /run/opengl-driver/lib/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva error: /run/opengl-driver/lib/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit
➜  ~ nvidia-smi
Sat Aug 10 23:39:42 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 560.31.02              Driver Version: 560.31.02      CUDA Version: 12.6     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce GTX 1650 Ti     Off |   00000000:01:00.0  On |                  N/A |
| N/A   49C    P5              9W /   50W |     526MiB /   4096MiB |     20%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      2539      G   ...ofiles/per-user/super6/bin/Hyprland        294MiB |
|    0   N/A  N/A      2701      G   Xwayland                                        2MiB |
|    0   N/A  N/A      2946      G   .../profiles/per-user/super6/bin/kitty        131MiB |
+-----------------------------------------------------------------------------------------+
```

elFarto, 2024-08-10:
Can you run that with NVD_LOG set, e.x: `NVD_LOG=1 vainfo`

wxlyyy, 2024-08-10:
```
➜  ~ NVD_LOG=1 vainfo
Trying display: wayland
libva info: VA-API version 1.22.0
libva info: Trying to open /run/opengl-driver/lib/dri/nvidia_drv_video.so
      2027.998552533 [29455-29455] ../src/vabackend.c: 168                     init CUDA ERROR 'unknown error' (999)

libva info: Found init function __vaDriverInit_1_0
      2027.998588726 [29455-29455] ../src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
      2027.998591878 [29455-29455] ../src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2027.998594775 [29455-29455] ../src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
      2028.010155485 [29455-29455] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
      2028.010186533 [29455-29455] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 560.31.02, major version: 560, minor version: 31
      2028.010191875 [29455-29455] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 1 2 6
      2028.011733970 [29455-29455] ../src/direct/direct-export-buf.c:  27       findGPUIndexFromFd CUDA ERROR 'initialization error' (3)

      2028.011742062 [29455-29455] ../src/vabackend.c:2253       __vaDriverInit_1_0 CUDA ERROR 'initialization error' (3)

libva error: /run/opengl-driver/lib/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit
```

elFarto, 2024-08-10:
Did you install the open source version or the proprietary version of the driver. As far as I know it won't work with the open source version, I can't test it as my card isn't supported.

wxlyyy, 2024-08-11:
I found the reason, because on NixOS, for versions above 560, the open source version is used by default

thesword53, 2024-08-16:
> The driver is working for me with 560.31.02 installed. Can you recheck with this version?

I tested again with 560.31.02 drivers and it's now working. Perhaps only the first beta version (560.28.03) was problematic.

closed 2024-08-16 by thesword53

TheBigK02, 2024-08-20:
hmm.. this version is not provided for me on the official debian version of the nvidia driver i guess. Then I stay at 555 for now till a newer or same version as 560.31.02 is released, correct?

`apt-cache policy nvidia-driver
nvidia-driver:
  Installiert:           555.42.06-1
  Installationskandidat: 555.42.06-1
  Versionstabelle:
     560.28.03-1 500
        500 https://developer.download.nvidia.com/compute/cuda/repos/debian12/x86_64  Packages
 *** 555.42.06-1 1001
        500 https://developer.download.nvidia.com/compute/cuda/repos/debian12/x86_64  Packages
        100 /var/lib/dpkg/status
     555.42.02-1 1001
        500 https://developer.download.nvidia.com/compute/cuda/repos/debian12/x86_64  Packages
     550.90.07-1 500
        500 https://developer.download.nvidia.com/compute/cuda/repos/debian12/x86_64  Packages
     550.54.15-1 500
        500 https://developer.download.nvidia.com/compute/cuda/repos/debian12/x86_64  Packages
     550.54.14-1 500
        500 https://developer.download.nvidia.com/compute/cuda/repos/debian12/x86_64  Packages
     545.23.08-1 500
        500 https://developer.download.nvidia.com/compute/cuda/repos/debian12/x86_64  Packages
     545.23.06-1 500
        500 https://developer.download.nvidia.com/compute/cuda/repos/debian12/x86_64  Packages
     535.183.01-1~deb12u1 500
        500 http://ftp.hosteurope.de/mirror/ftp.debian.org/debian bookworm/non-free amd64 Packages`