elFarto/nvidia-vaapi-driver #313: The problem with hevc decoding in chrome

OverDDose, 2024-07-24:
In chrome, when decoding h265 video in 1920x1080 resolution, there is a problem with colors, however, when reducing the resolution to 1280x720, everything returns to normal. Video logs:
[1920x1080_logs.txt](https://github.com/user-attachments/files/16363375/1920x1080_logs.txt)
[1280x720_logs.txt](https://github.com/user-attachments/files/16363376/1280x720_logs.txt)

Screenshots:
1920x1080
[image]
1280x720
[image]

Thorium browser v124 was used, with flags `--use-gl=angle --use-angle=vulkan --enable-features=VaapiVideoEncoder,VaapiOnNvidiaGPUs,VaapiIgnoreDriverChecks,Vulkan,DefaultANGLEVulkan,VulkanFromANGLE --ignore-gpu-blocklist --disable-gpu-driver-bug-workaround`

GPU: NVIDIA RTX 3060
DRIVER: 535 metapackage
OS: ubuntu 24.04
nvidia-vaapi-driver version:  0.0.12

insilications, 2024-07-27:
Similar problem here, with a GTX 3070. When trying to play a 1920x1080 h264 high video, I get the same green tinted artifacts.  When switching to 1280x720 h264 main for the same file, it plays perfectly.

OverDDose, 2024-07-27:
UPD: the problem was noticed when playing videos with a resolution other than 1280x720 (for example, 1920x1080, 640x360, etc.)

ManuLinares, 2024-08-01:
Chrome is currently unsupported, and will not function.

OverDDose, 2024-08-02:
> Chrome is currently unsupported, and will not function.
> 
> https://github.com/elFarto/nvidia-vaapi-driver?tab=readme-ov-file#chrome

Yes, I read this, but everything started up and even worked on chromium. The only problem is the glitch on most resolutions. Suddenly, there is a possibility that someone in firefox has a similar problem. Plus, there is already an open issue with work on chrome: [Chrome Support](https://github.com/elFarto/nvidia-vaapi-driver/issues/5)

ManuLinares, 2024-08-02:
I recommend posting on that open issue as it fits perfectly with your problem. Also close this one, my reasoning being that duplicate issues makes it harder for anyone to help/contribute a patch or a fix.

Better to have Chrome support issues all in one place.

closed 2024-08-02 by OverDDose