elFarto/nvidia-vaapi-driver #314: Was working, formatted, no longer working.

ddouglas87, 2024-07-29:
OpenSUSE Tumbleweed, X11 Cinnamon desktop.

First I tried `zypper install nvidia-vaapi-driver`.  Set `/etc/environment` to

```
MOZ_DISABLE_RDD_SANDBOX=1
LIBVA_DRIVER_NAME=nvidia
NVD_BACKEND=direct
```

```
bunny@happy:~> ffmpeg -hwaccels 2>/dev/null
Hardware acceleration methods:
vdpau
cuda
vaapi
qsv
drm
vulkan
```

In Firefox `about:config` I set:

|Config|Setting|
|-------|------|
|media.ffmpeg.vaapi.enabled|true|
|media.rdd-ffmpeg.enabled|true|
|gfx.x11-egl.force-enabled|true|
|widget.dmabuf.force-enabled|true|

In Firefox `about:support`:
[image]

I'm using the plugin h264ify, which I used in the past successfully and have validated it is working.  (Stats for nerds -> AVC1.)

I pull up a youtube video and while playing in `watch nvidia-smi` I see Type G for firefox instead of the G+C I'm supposed to be seeing.

My nvidia drivers are 550.100-25.1.

I downloaded the repo with `git clone`, installed the prerequisites for my system (`zypper install gstreamer-plugins-bad-devel cmake-full Mesa-libEGL-devel ffnvcodec-devel libdrm-devel libva-devel`), did the meson build and install instructions, verified the nvidia .so file was updated.  Rebooted again.  Same issue with Firefox.  It shows H264 supported, and the video is H264 but it's decoding with software.

I tried changing the `NVD_BACKEND` type.  I tried changing `MOZ_DISABLE_RDD_SANDBOX`.  I'm not sure what settings I should mess with to get this working.

Here's the output from the `test.sh` file: [testsh_output.txt](https://github.com/user-attachments/files/16408247/testsh_output.txt)

I had this working fine in the past, so I know it's possible on my hardware.  I'm quite confused.

ManuLinares, 2024-08-01:
You haven't shared any logs or your system specs. I don't know what kind of help do you expect.

share a pastebin with your specs: '`inxi -Fxxxza --no-host`'
run firefox, open a video, close firefox: '`env MOZ_LOG="PlatformDecoderModule:5" NVD_LOG=1 firefox`' and share the stdout in a pastebin

ddouglas87, 2024-08-02:
Here you go: 
[specs.txt](https://github.com/user-attachments/files/16471091/specs.txt)
[firefox.txt](https://github.com/user-attachments/files/16471097/firefox.txt)

I wasn't sure exactly what to log.  Apologies.  Hopefully it helps.  At first glance it looks like everything should be working, but it's not.

ManuLinares, 2024-08-02:
You should set this in about:config

```
media.ffmpeg.vaapi.enabled	true
media.av1.enabled		false
gfx.x11-egl.force-enabled 	false
widget.dmabuf.force-enabled 	false
```

Then try to run like this on a terminal:

```
unset LIBVA_DRIVER_NAME
MOZ_DISABLE_RDD_SANDBOX=1 NVD_BACKEND=direct MOZ_LOG="PlatformDecoderModule:5" NVD_LOG=1 firefox
```

Is firefox a native package or a flatpak, snap, etc?
Could you provide the output of `vainfo --display drm --device /dev/dri/renderD128` (assuming this is your device /dev/dri/renderD128, idk)

---

If the above doesn't work, try EGL backend.
```
gfx.x11-egl.force-enabled 	true
widget.dmabuf.force-enabled 	true
```

Then try to run like this on a terminal:
```
unset LIBVA_DRIVER_NAME
MOZ_DISABLE_RDD_SANDBOX=1 NVD_BACKEND=egl MOZ_LOG="PlatformDecoderModule:5" NVD_LOG=1 firefox
```

ddouglas87, 2024-08-02:
Thanks for the attempted help, but I just solved it.  I noticed something in the logs you asked me to send in, so I investigated. From firefox.txt:

>HEVC NONE

That's odd.  Why would it say NONE?  Maybe it's a lack of codec.  I did `opi codecs` to install any missing codecs.  After rebooting hardware acceleration in Firefox works just fine!

For FFmpeg hardware decoding it requires "non-free" codecs to be installed.  It's possible to have FFmpeg come with your OS by default without codecs.  Without codecs FFmpeg still runs, but only software decoding.  Codecs are required for hardware decoding.  (Or at least it's that way on my setup using OpenSUSE.)  I recommend adding this note to the README.md file, as this is a necessary prerequisite.

Cheers!

closed 2024-09-09 by elFarto