elFarto/nvidia-vaapi-driver #315: [Stutter] Video Stuttering When Switching Between Multiple Tabs

ManuLinares, 2024-07-31:
Description:

When playing several videos in different tabs, switching back to a tab after a few seconds causes the video to stutter before resuming playback.

System information:

- Os Archlinux
- linux 6.10.2.arch1-1
- libva-nvidia-driver-git 0.0.12.r11.gf3205bd-1
- nvidia-dkms 555.58.02-1
- firefox 128.0.3-1
- NVIDIA GeForce RTX 3070 using direct backend

Steps to Reproduce:

1. Open 6 videos in different tabs and ensure they are playing.
2. Switch to the first tab and wait for approximately 10 seconds.
3. Switch to the sixth tab and observe the video stuttering for 1-2 seconds before it resumes playback.

elFarto, 2024-09-09:
I can see the same thing here (although the delay seems <1s), but I'm not sure it's an issue with this driver. It's more likely an issue with Firefox's VA-API code, specifically it will stop decoding video on tabs that aren't visible, when that tab becomes visible it has to restart the driver, including catching up to the correct frame, which can take a moment.

Not sure there's much I can do about this one.

thesword53, 2024-09-19:
This driver depends on CUDA and CUDA context creation is pretty slow.

closed 2024-10-19 by elFarto

ManuLinares, 2025-01-25:
I just wanted to say that with nvidia driver 570.86.10 this problem went away.

I'm on RTX3070, nvidia 570.86.10. GSP disabled with `nvidia.NVreg_EnableGpuFirmware=0` cmdline.

For future reference