elFarto/nvidia-vaapi-driver #318: Firefox reverting back to CPU decoding 10s into 8K Yt video

KaeLL, 2024-08-16:
Title. Logging below. I added some references to make it easier to follow when it happens.

<details>
<summary>$ NVD_LOG=1 firefox https://www.youtube.com/watch?v=QqI9eKQnFL4</summary>

```yaml
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib64/dri-nonfree/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
8925.163968069 [24806-24815] ../src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
8925.163971569 [24806-24815] ../src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
8925.163973829 [24806-24815] ../src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
8925.170627336 [24806-24815] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
8925.170658616 [24806-24815] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 555.58.02, major version: 555, minor version: 58
8925.170664246 [24806-24815] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 2b00 1 2 6
libva info: va_openDriver() returns 0
8925.277101384 [24806-24815] ../src/vabackend.c:1446      nvQueryImageFormats In nvQueryImageFormats
8925.391499465 [24806-24815] ../src/vabackend.c: 674           nvCreateConfig got profile: 6 with 0 attributes
8925.391509745 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) (nil) 0
8925.391512495 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) 0x7f628f5d5700 8
8925.396247233 [24806-24815] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
8925.396258593 [24806-24815] ../src/vabackend.c: 674           nvCreateConfig got profile: 7 with 0 attributes
8925.396261593 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) (nil) 0
8925.396263163 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) 0x7f628f5d5640 8
8925.404333625 [24806-24815] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
8925.404341295 [24806-24815] ../src/vabackend.c: 674           nvCreateConfig got profile: 13 with 0 attributes
8925.404344315 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) (nil) 0
8925.404346185 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) 0x7f628f5d5640 8
8925.412900631 [24806-24815] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
8925.412908201 [24806-24815] ../src/vabackend.c: 674           nvCreateConfig got profile: 14 with 0 attributes
8925.412911091 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 9 (8) (nil) 0
8925.412912981 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 9 (8) 0x7f628f5d5ac0 8
8925.420821885 [24806-24815] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
8925.420830465 [24806-24815] ../src/vabackend.c: 674           nvCreateConfig got profile: 19 with 0 attributes
8925.420833105 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 10 (8) (nil) 0
8925.420836095 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 10 (8) 0x7f628f5d5ac0 8
8925.428169065 [24806-24815] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
8925.428177075 [24806-24815] ../src/vabackend.c: 674           nvCreateConfig got profile: 32 with 0 attributes
8925.428179324 [24806-24815] ../src/vabackend.c: 737           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
8925.428182374 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 11 (8) (nil) 0
8925.428184114 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 11 (8) 0x7f628f5d5d00 8
8925.437066337 [24806-24815] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
8925.437075857 [24806-24815] ../src/vabackend.c: 674           nvCreateConfig got profile: 21 with 0 attributes
8925.437078697 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 10 (10) (nil) 0
8925.437080637 [24806-24815] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 10 (10) 0x7f628f5d5640 8
8925.444244579 [24806-24815] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
8925.444637855 [24806-24815] ../src/vabackend.c:2098              nvTerminate Terminating 0x7f62a3534200
8925.446446075 [24806-24815] ../src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances
8925.563820944 [24806-24829] ../src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
8925.563829204 [24806-24829] ../src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
8925.563832694 [24806-24829] ../src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
8925.563848713 [24806-24829] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
8925.563861263 [24806-24829] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 555.58.02, major version: 555, minor version: 58
8925.563863953 [24806-24829] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 2b00 1 2 6
8925.678451512 [24806-24829] ../src/vabackend.c:1446      nvQueryImageFormats In nvQueryImageFormats
8925.679166754 [24806-24829] ../src/vabackend.c:2098              nvTerminate Terminating 0x7f62a3534200
8925.681187702 [24806-24829] ../src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances
8925.762234947 [24806-24833] ../src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
8925.762243547 [24806-24833] ../src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
8925.762245517 [24806-24833] ../src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
8925.762270997 [24806-24833] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
8925.762285707 [24806-24833] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 555.58.02, major version: 555, minor version: 58
8925.762288767 [24806-24833] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 2b00 1 2 6
8925.884288235 [24806-24833] ../src/vabackend.c:1446      nvQueryImageFormats In nvQueryImageFormats
8925.884429173 [24806-24847] ../src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
8925.884433913 [24806-24847] ../src/vabackend.c:2197       __vaDriverInit_1_0 Now have 1 (0 max) instances
8925.884436313 [24806-24847] ../src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
8925.884450973 [24806-24847] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
[... 381 lines cut ...]
8966.760658789 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 0 (0x7f626ebc8f20)
8966.761635958 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 2 (0x7f626ebc8ca0)
8966.762612058 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 10 (0x7f626ebc80c0)
8966.763587147 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 11 (0x7f626ebc87a0)
8966.764577216 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 3 (0x7f626ebc8fc0)
8966.766724693 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface 4 (0x7f626ebc88e0)
8966.767715692 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f626ebc8de0)
8966.767719472 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f626ebc8660)
8966.767720752 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f626ebc8d40)
8966.767722272 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f626ebc8520)
8966.767724102 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f626ebc8700)
[... cut ...]
8966.767734822 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f6287bf2d80)
8966.767736222 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f6287bf2a60)
8966.767737902 [24806-24979] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f6287bf2920)
8966.767740392 [24806-24979] ../src/vabackend.c:2098              nvTerminate Terminating 0x7f62a3534200
8966.772009405 [24806-24979] ../src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances
```

</details>

martynhare, 2024-09-03:
Firefox itself deliberately ditches VA-API decoding and reverts irrespective of which driver you use if the browser believes the 8K decode will be too slow. Mozilla currently considers this fallback a fix for the fact that their current implementation isn't yet efficient enough. (Source: https://bugzilla.mozilla.org/show_bug.cgi?id=1750388)

To test if this is the cause, you could either make a custom build with the change stripped out, or try using Firefox 96 (where it's definitely not implemented yet) to see what happens.

I hope this helps!

KaeLL, 2024-09-05:
Honestly, I didn't expect them to ignore such an issue in a nonchalant, almost neglectful way.
Thanks for the info, anyways.

closed 2024-09-05 by KaeLL