elFarto/nvidia-vaapi-driver #319: nvidia direct broken with driver 560

groopcd, 2024-08-19:
starting with driver 560 (i jumped from 550 to 560) firefox now gives up with the following error:

libva info: VA-API version 1.22.0
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      4200.268969579 [37465-37733] ../src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      4200.268974010 [37465-37733] ../src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
      4200.268976536 [37465-37733] ../src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
      4200.274431014 [37465-37733] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
      4200.274454812 [37465-37733] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 560.28.03, major version: 560, minor version: 28
      4200.274461271 [37465-37733] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 2 6 1
libva info: va_openDriver() returns 0
      4200.398661110 [37465-37733] ../src/vabackend.c:1446      nvQueryImageFormats In nvQueryImageFormats
      4200.520148678 [37465-37733] ../src/vabackend.c: 674           nvCreateConfig got profile: 6 with 0 attributes
      4200.520159574 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      4200.520161234 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) 0x7f365e3dc7c0 8
      4200.526104988 [37465-37733] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      4200.526118296 [37465-37733] ../src/vabackend.c: 674           nvCreateConfig got profile: 7 with 0 attributes
      4200.526121369 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      4200.526122661 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) 0x7f365e3dcc40 8
      4200.531018868 [37465-37733] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      4200.531028646 [37465-37733] ../src/vabackend.c: 674           nvCreateConfig got profile: 13 with 0 attributes
      4200.531031071 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      4200.531032292 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) 0x7f365e3dc7c0 8
      4200.536877503 [37465-37733] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      4200.536893200 [37465-37733] ../src/vabackend.c: 674           nvCreateConfig got profile: 14 with 0 attributes
      4200.536897107 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 9 (8) (nil) 0
      4200.536899055 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 9 (8) 0x7f365e3dc7c0 8
      4200.543022752 [37465-37733] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      4200.543036070 [37465-37733] ../src/vabackend.c: 674           nvCreateConfig got profile: 19 with 0 attributes
      4200.543039439 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 10 (8) (nil) 0
      4200.543041158 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 10 (8) 0x7f365e3dcc40 8
      4200.548237708 [37465-37733] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      4200.548253653 [37465-37733] ../src/vabackend.c: 674           nvCreateConfig got profile: 32 with 0 attributes
      4200.548256666 [37465-37733] ../src/vabackend.c: 737           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
      4200.548260529 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 11 (8) (nil) 0
      4200.548262309 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 11 (8) 0x7f365e3dcc40 8
      4200.555111579 [37465-37733] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      4200.555127495 [37465-37733] ../src/vabackend.c: 674           nvCreateConfig got profile: 21 with 0 attributes
      4200.555131495 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 10 (10) (nil) 0
      4200.555133312 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 10 (10) 0x7f365e3dc7c0 8
      4200.560849558 [37465-37733] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      4200.612462135 [37465-37747] ../src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      4200.612470501 [37465-37747] ../src/vabackend.c:2197       __vaDriverInit_1_0 Now have 1 (0 max) instances
      4200.612472802 [37465-37747] ../src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
      4200.612493106 [37465-37747] ../src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
      4200.612508395 [37465-37747] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 560.28.03, major version: 560, minor version: 28
      4200.612512189 [37465-37747] ../src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 2 6 1
      4200.728130044 [37465-37747] ../src/vabackend.c:1446      nvQueryImageFormats In nvQueryImageFormats
      4200.757503707 [37465-37733] ../src/vabackend.c: 674           nvCreateConfig got profile: 6 with 0 attributes
      4200.757516756 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      4200.757519551 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) 0x7f364c5fc580 8
      4200.762570441 [37465-37733] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      4200.762582756 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) (nil) 32566
      4200.762585553 [37465-37733] ../src/vabackend.c:1802 nvQuerySurfaceAttributes with 4 (8) 0x7f364c5fc280 8
      4200.809217918 [37465-37733] ../src/vabackend.c:1869 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      4200.809242499 [37465-37733] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 720x720, format 1 (0x7f3648ecf980)
      4200.809246016 [37465-37733] ../src/vabackend.c:1540            nvDeriveImage In nvDeriveImage
      4200.809249428 [37465-37733] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 720x720, format 1 (0x7f3648ecff20)
      4200.809251804 [37465-37733] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 720x720, format 1 (0x7f3648ecfa20)
      4200.809254038 [37465-37733] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 720x720, format 1 (0x7f3648ecf020)
      4200.809256358 [37465-37733] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 720x720, format 1 (0x7f3648ecfac0)
      4200.809258639 [37465-37733] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 720x720, format 1 (0x7f3648ecfc00)
      4200.809260869 [37465-37733] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 720x720, format 1 (0x7f3648ecf0c0)
      4200.809263491 [37465-37733] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 720x720, format 1 (0x7f3648ed0240)
      4200.809265942 [37465-37733] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 720x720, format 1 (0x7f3648ecf200)
      4200.809268177 [37465-37733] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 720x720, format 1 (0x7f3648ed02e0)
      4200.809270434 [37465-37733] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 720x720, format 1 (0x7f3648ed0380)
      4200.809272802 [37465-37733] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 720x720, format 1 (0x7f3648ecf2a0)
      4200.809275051 [37465-37733] ../src/vabackend.c: 957        nvCreateSurfaces2 Creating surface 720x720, format 1 (0x7f3648ecf340)
[... 198 lines cut ...]
      4202.495770192 [37465-37773] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f3641a0b2e0)
      4202.495771761 [37465-37773] ../src/vabackend.c: 991        nvDestroySurfaces Destroying surface -1 (0x7f3641a0ade0)
[... cut ...]
      4202.896338238 [37465-37830] ../src/vabackend.c:1446      nvQueryImageFormats In nvQueryImageFormats
      4202.896694347 [37465-37830] ../src/vabackend.c:2098              nvTerminate Terminating 0x7f367043a980
      4202.897455697 [37465-37830] ../src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances

using:

NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 NVD_BACKEND=direct __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json firefox

and all the proper options in the readme file.

any workaround yet?

TheBigK02, 2024-08-23:
560.35.03 works now which was released on debian today. So all good.

Thanks for looking into this.

rkoot, 2024-08-27:
I had to install ffmpeg (more specifically, the various libav* packages) from the packman repo to get it going.

groopcd, 2024-08-27:
> I had to install ffmpeg (more specifically, the various libav* packages) from the packman repo to get it going.

mind you this was working with driver 550, i just changed to 560 and direct broke, no further changes

elFarto, 2024-08-27:
The eglCreateImageKHR mentioned in the log output suggest this is an EGL issue, as that method is only available in the MESA library. Make sure that the proprietary version of the driver was installed, and that `eglinfo` is showing NVIDIA as the default for the various options.

rkoot, 2024-08-28:
> > I had to install ffmpeg (more specifically, the various libav* packages) from the packman repo to get it going.
> 
> mind you this was working with driver 550, i just changed to 560 and direct broke, no further changes

I'm running the same stack as you do, OpenSuse Tumbleweed (up to date), firefox 129.0, nvidia (version 560.35.3) open kernel module (so not the proprietary one) and it works just fine for me. I have to mention that all 560 drivers prior to 560.35.3 did not work with the open kernel module.
I just used the instructions from the README. I did need to switch my ffmpeg install from the oss repo to the packman repo to get it all working.

groopcd, 2024-08-28:
> I had to install ffmpeg (more specifically, the various libav* packages) from the packman repo to get it going.

mind you this was working with driver 550, i just changed to 560 and direct broke, no further changes

> I'm running the same stack as you do, OpenSuse Tumbleweed (up to date), firefox 129.0, nvidia (version 560.35.3) open kernel module (so not the proprietary one) and it works just fine for me. I have [...]

sweet, could you post here a rpm -qa|egrep -i nvidia and rpm -qa|egrep -i ffmpeg so i can compare the packages?

also, how did you switch to the packman repo for ffmpeg (i have it enabled but im new to tumbleweed).

rkoot, 2024-08-29:
> sweet, could you post here a rpm -qa|egrep -i nvidia and rpm -qa|egrep -i ffmpeg so i can compare the packages?

rpm -qa | egrep 'ffmpeg|nvidia'
ffmpeg-7-libavfilter-devel-7.0.2-1699.3.pm.3.x86_64
nvidia-texture-tools-2.1.2-2.13.x86_64
ffmpegthumbs-24.08.0-1.1.x86_64
nvidia-compute-utils-G06-560.35.03-0.x86_64
ffmpeg-7-libavcodec-devel-7.0.2-1699.3.pm.3.x86_64
kernel-firmware-nvidia-gspx-G06-560.35.03-0.x86_64
kernel-firmware-nvidia-gspx-G06-555.42.06-0.x86_64
ffmpeg-7-libswresample-devel-7.0.2-1699.3.pm.3.x86_64
nvidia-open-560.35.03-1.x86_64
ffmpeg-7-libswscale-devel-7.0.2-1699.3.pm.3.x86_64
nvidia-drivers-G06-560.35.03-0.x86_64
nvidia-compute-G06-32bit-560.35.03-0.x86_64
kernel-firmware-nvidia-20240809-1.1.noarch
nvidia-video-G06-560.35.03-0.x86_64
ffmpeg-7-libavdevice-devel-7.0.2-1699.3.pm.3.x86_64
ffmpeg-7-libpostproc-devel-7.0.2-1699.3.pm.3.x86_64
nvidia-open-560-560.35.03-1.x86_64
nvidia-gl-G06-560.35.03-0.x86_64
ffmpeg-7-libavutil-devel-7.0.2-1699.3.pm.3.x86_64
ffmpeg-7-libavformat-devel-7.0.2-1699.3.pm.3.x86_64
kernel-firmware-nvidia-gspx-G06-560.28.03-0.x86_64
nvidia-utils-G06-560.35.03-0.x86_64
nvidia-compute-G06-560.35.03-0.x86_64
ffmpeg-7-7.0.2-1699.3.pm.3.x86_64
nvidia-gl-G06-32bit-560.35.03-0.x86_64
nvidia-video-G06-32bit-560.35.03-0.x86_64
nvidia-open-driver-G06-kmp-default-560.35.03_k4.12.14_lp150.12.82-0.x86_64

My firefox settings (only those settings that changed from default settings in firefox):
media.ffmpeg.vaapi.enabled = true
gfx.x11-egl.force-enabled = true
widget.dmabuf.force-enabled = true

contents of ~/.bashrc 
export NVD_BACKEND=direct
export LIBVA_DRIVER_NAME=nvidia 
export MOZ_X11_EGL=1 
export MOZ_DISABLE_RDD_SANDBOX=1 

> 
> also, how did you switch to the packman repo for ffmpeg (i have it enabled but im new to tumbleweed).

Case 1: you want to switch all relevant packages from the oss repo to the packman repo.
- First check with 'zypper lr -u' what number the packman repo has.
- The perform the switch with 'zypper dup -r <number of the packman repo> --allow-vendor-change

Case 2 (more fine grained control, but also a lot more finicky): you only want to switch certain packages from the oss repo to the packman repo.
- Start yast and go to the software management module.
- search for ie 'ffmpeg' and select the 'Versions' tab.
- here you can toggle between the various available versions, ie the oss and the packman ones.
- yast will ask you what to do with the various dependant packages, ie switch or not. 

I hope this sorts you out.

groopcd, 2025-06-18:
> I'm also running into this, except on NixOS and with driver version
> 575.57.08. This *was* definitely working until about a week or two ago,
> and may have coincided with a driver update, but I always notice breakage a
> little late ("hmm, why is my CPU fan spinning so loudly?").
>
> I'm also getting the egl message mentioned earlier:
>
> [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: requested buffer attributes are not supported
>
> As for eglinfo, there *are* non-nvidia devices available, but it seems
> that for anything relevant the nvidia driver should be used:
>
> GBM platform:
> EGL API version: 1.5
> EGL vendor string: NVIDIA
> EGL version string: 1.5
> EGL client APIs: OpenGL_ES OpenGL
> OpenGL core profile vendor: NVIDIA Corporation
> OpenGL core profile renderer: NVIDIA GeForce RTX 4060 Ti/PCIe/SSE2
> OpenGL core profile version: 4.6.0 NVIDIA 575.57.08
> OpenGL core profile shading language version: 4.60 NVIDIA
> OpenGL compatibility profile vendor: NVIDIA Corporation
> OpenGL compatibility profile renderer: NVIDIA GeForce RTX 4060 Ti/PCIe/SSE2
> OpenGL compatibility profile version: 4.6.0 NVIDIA 575.57.08
> OpenGL compatibility profile shading language version: 4.60 NVIDIA
> [... 114 lines cut ...]
> OpenGL compatibility profile shading language version: 4.50
> OpenGL ES profile vendor: Mesa
> OpenGL ES profile renderer: llvmpipe (LLVM 19.1.7, 256 bits)
> OpenGL ES profile version: OpenGL ES 3.2 Mesa 25.0.7
> OpenGL ES profile shading language version: OpenGL ES GLSL ES 3.20

EGL has been broken for some time now, use DIRECT instead.

regards
| Paulo Dias
| ***@***.***

*Tempora mutantur, nos et mutamur in illis.*

TLATER, 2025-06-19:
I am using the direct backend, and have been since I started using this years ago:

```console
~> echo $env.NVD_BACKEND
direct
```

Also using all other suggested configuration, precisely as instructed: https://github.com/TLATER/dotfiles/blob/master/nixos-modules/nvidia/vaapi.nix#L72

groopcd, 2025-06-19:
what do you get when you run vainfo with egl and direct?
| Paulo Dias
| ***@***.***

*Tempora mutantur, nos et mutamur in illis.*

TLATER, 2025-09-27:
Finally figured it out, pretty sure this is a duplicate of #370 (and should be closed, just update, my bad, I *thought* I was on the latest version).