elFarto/nvidia-vaapi-driver #321: Only works with libavcodec-freeworld (Fedora)

S7venLights, 2024-08-22:
Hi I had HW acceleration working on my Librewolf flatpak (according to [this config](https://github.com/elFarto/nvidia-vaapi-driver/issues/23#issuecomment-2160190481)) and according to [this Fedora guide](https://fedoraproject.org/wiki/Firefox_Hardware_acceleration#Configure_VA-API_Video_decoding_on_NVIDIA) which advises installing libavcodec-freeworld unlike your Github.

But I think after Nvidia releases non-free driver 555 (fixing explicit sync) I started experiencing crashes on Firefox based browsers.

I removed some unnecessary layered packages but still had crashes then I  removed libavcodec-freeworld and it seems the crashes stopped. But HW acc also stopped working in FF based browsers.

So my question is, is having libavcodec-freeworld the only way to have HW Acc working in FF browsers on Fedora?

If not, what am I missing?

I run:
Operating System: Fedora Linux 40 [Kinoite](https://kinoite.fedoraproject.org)
KDE Plasma Version: 6.1.4
KDE Frameworks Version: 6.5.0
Qt Version: 6.7.2
Kernel Version: 6.10.5-200.fc40.x86_64 (64-bit)
Graphics Platform: Wayland
Graphics Processor: NVIDIA GeForce GTX 1060/PCIe/SSE2

I have installed all the packages you required and FF abou:config tweaks
my rpm-ostree status:

 Version: 40.20240821.0 (2024-08-21T01:01:38Z)
               BaseCommit: 319076423a90a3de2cd75e9e439359218f3a5bd056dc9bbca6e5d98b8001c679
             GPGSignature: Valid signature by 115DF9AEF857853EE8445D0A0727707EA15B79CC
          LayeredPackages: cockpit-composer cockpit-machines cockpit-networkmanager cockpit-ostree cockpit-pcp
                           cockpit-podman cockpit-selinux cockpit-storaged cockpit-system egl-wayland gcc
                           glibc-devel gnome-disk-utility gnome-tweak-tool gstreamer1-plugins-bad-free-extras
                           gstreamer1-plugins-bad-freeworld gstreamer1-plugins-ugly gstreamer1-vaapi
                           libguestfs-tools libva-devel libva-nvidia-driver libva-utils
                           libvirt-daemon-config-network libvirt-daemon-kvm meson mozilla-ublock-origin
                           nv-codec-headers nvidia-vaapi-driver pcp python3-distutils-extra python3-libguestfs
                           qemu-kvm rpmfusion-free-release rpmfusion-nonfree-release tlp vdpauinfo virt-install
                           virt-manager virt-top virt-viewer waydroid xorg-x11-drv-nvidia
                           xorg-x11-drv-nvidia-cuda xorg-x11-drv-nvidia-power yt-dlp
            LocalPackages: virtio-win-0.1.225-2.noarch

My Librewolf environment variables: 
[image]

S7venLights, 2024-08-22:
Oh I just had Librewolf crash on me, with the above mentioned config. 
Not sure if it's related but here's the CLI readout:

``
Crash Annotation GraphicsCriticalError: |[0][GFX1-]: TOpAddBlobImage failed (t=7790.87) |[136][GFX1-]: TOpAddBlobImage failed (t=7790.9) |[137][GFX1-]: TOpAddBlobImag
e failed (t=7790.9) |[138][GFX1-]: TOpAddBlobImage failed (t=7790.9) |[139][GFX1-]: TOpAddBlobImage failed (t=7790.9) |[125][GFX1-]: TOpAddBlobImage failed (t=7790.89
) |[126][GFX1-]: TOpAddBlobImage failed (t=7790.89) |[127][GFX1-]: TOpAddBlobImage failed (t=7790.89) |[128][GFX1-]: TOpAddBlobImage failed (t=7790.89) |[129][GFX1-]:
 TOpAddBlobImage failed (t=7790.89) |[130][GFX1-]: TOpAddBlobImage failed (t=7790.89) |[131][GFX1-]: TOpAddBlobImage failed (t=7790.89) |[132][GFX1-]: TOpAddBlobImage
 failed (t=7790.9) |[133][GFX1-]: TOpAddBlobImage failed (t=7790.9) |[134][GFX1-]: TOpAddBlobImage failed (t=7790.9) |[135][GFX1-]: TOpAddBlobImage failed (t=7790.9) 
[GFX1-]: TOpAddBlobImage failed 
Exiting due to channel error. 
Exiting due to channel error. 
Crash Annotation GraphicsCriticalError: |[C0][GFX1-]: CompositorBridgeChild receives IPC close with reason=AbnormalShutdown (t=574.17) Exiting due to channel error. 
Exiting due to channel error. 
Exiting due to channel error. 
Exiting due to channel error. 
Exiting due to channel error. 
Exiting due to channel error. 
Crash Annotation GraphicsCriticalError: |[C0][GFX1-]: CompositorBridgeChild receives IPC close with reason=AbnormalShutdown (t=7802.98) [GFX1-]: CompositorBridgeChild
 receives IPC close with reason=AbnormalShutdown 
Exiting due to channel error.
``

Happened while scrolling up and down https://piped.adminforge.de/playlist?list=PLG49S3nxzAnkL2ulFS3132mOVKuzzBxA8

S7venLights, 2024-08-22:
Just now, when loading a tab, I had a freeze for a bit then it recovered, and CLI said:

```
Crash Annotation GraphicsCriticalError: |[0][GFX1-]: Detect DeviceReset DeviceResetReason::RESET DeviceResetDetectPlace::WR_POST_UPDATE in Parent process (t=436.547) [GFX1-]: Detect DeviceReset DeviceResetReason::RESET DeviceResetDetectPlace::WR_POST_UPDATE in Parent process Crash Annotation GraphicsCriticalError: |[0][GFX1-]: Detect DeviceReset DeviceResetReason::RESET DeviceResetDetectPlace::WR_POST_UPDATE in Parent process (t=436.547) |[1][GFX1-]: Failed to make render context current during destroying. (t=436.561) [GFX1-]: Failed to make render context current during destroying. 

Unflushed glGetGraphicsResetStatus: 0x92bb
```

S7venLights, 2024-08-22:
Tor Browser is crashing too (without info in CLI), I'm going to try switch to Nouveau drivers for now

elFarto, 2024-09-09:
I don't think any of the errors you're getting are directly related to this driver. The link you posted doesn't contain any video elements (it doesn't initialise the driver on my machine).

closed 2026-09-26 by elFarto