elFarto/nvidia-vaapi-driver #322: vainfo fails with nv_alloc_object failed: -1 0 22

Pesc0, 2024-08-29:
Hi, with driver version 550.90.07 i get this:
```
~$ NVD_LOG=1 vainfo --display drm --device /dev/dri/renderD130
Trying display: drm
      3286.873294574 [5842-5842] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      3286.873302332 [5842-5842] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
      3286.873304117 [5842-5842] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
      3286.878124981 [5842-5842] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
      3286.878144011 [5842-5842] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 470.123.45, major version: 470, minor version: 123
      3286.878146783 [5842-5842] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 0 fe 1
      3286.878150029 [5842-5842] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c:  74          nv_alloc_object nv_alloc_object failed: -1 0 22
      3286.878151361 [5842-5842] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 297            init_nvdriver nv_alloc_object NV01_ROOT_CLIENT failed
      3286.878152521 [5842-5842] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 353            init_nvdriver Got error initing
      3286.878156436 [5842-5842] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 118            nv_rm_control nv_rm_control failed: -1 0 25
      3286.878159421 [5842-5842] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2248       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

Not sure when it broke, i'm not even sure if this is the driver's fault and has nothing to do with this software, in which case feel free to close this immediately. If you need any other info please let me know.

rkoot, 2024-08-30:
It looks like you're running the 470.123.45 kernel module. That won't work with 550.90.07 userland. If you haven't rebooted yet, try that. And if that fails, try to reinstall the nvidia kernel module of the correct version.
Userland version and kernel module version should match.

Pesc0, 2024-08-30:
Hi, thanks for helping out.
I'm pretty sure my setup is correct, and that number is coming from this https://github.com/elFarto/nvidia-vaapi-driver/issues/276 instead.

I also found this similar issue https://github.com/elFarto/nvidia-vaapi-driver/issues/211 that may be related.

elFarto, 2024-09-09:
I've added a fix to master to attempt to get the version number from somewhere else if it's not available through the normal mechanism. The fact the normal mechanism isn't working is a bit concerning.

Pesc0, 2024-09-10:
Thank you.
In your opinion would detecting the right version fix this error as well?

elFarto, 2024-09-10:
Probably.

moetayuko, 2024-09-22:
> I've added a fix to master to attempt to get the version number from somewhere else if it's not available through the normal mechanism. The fact the normal mechanism isn't working is a bit concerning.

Hello, I experienced the same issue and it's fixed in aur/libva-nvidia-driver-git which contains the new version fetcher. Can you make a new release so more people can receive the fix? Thanks.

Pesc0, 2025-05-05:
Looks like this has been fixed in 0.0.13
Closing, thank you :)

closed 2025-05-05 by Pesc0