elFarto/nvidia-vaapi-driver #323: something wrong with vaapi

xeleven11, 2024-09-03:
Ubuntu22.04 xfce. pascal 10series. driver535.183.01 librewolf129.0.1-1

I only see H264 and VP9 hardware decoding in about:support. DMABUF & HARDWARE_VIDEO_DECODING still blocklisted by gfxInfo. REMOTE_CANVAS is blocked as well. video engine utilization goes up in NVIDIA-settings but there are some errors from terminal. Intel libva was built from source but only vaapi is updated to latest version https://github.com/intel/libva/issues/762#issuecomment-2324096594

there is something wrong with nvidia-settings too as I cannot reset to default. here is error log from terminal
```
(librewolf-default:19725): GLib-GObject-CRITICAL **: 07:17:45.305: g_type_add_interface_static: assertion 'g_type_parent (interface_type) == G_TYPE_INTERFACE' failed
[Parent 19725, Main Thread] WARNING: g_type_add_interface_static: assertion 'g_type_parent (interface_type) == G_TYPE_INTERFACE' failed: 'glib warning', file /work/librewolf-129.0.1-1/toolkit/xre/nsSigHandlers.cpp:187
[RDD 20184, MediaSupervisor #7] WARNING: Call to mmap failed: Bad file descriptor: file /work/librewolf-129.0.1-1/ipc/chromium/src/base/shared_memory_posix.cc:515
###!!! [Parent][DispatchAsyncMessage] Error: SHMEM_CREATED_MESSAGE Payload error: message could not be deserialized
[Child 26002, MediaDecoderStateMachine #3] WARNING: Decoder=7e96a4594a00 Decode error: NS_ERROR_OUT_OF_MEMORY (0x8007000e) - DecodeNextSample: file /work/librewolf-129.0.1-1/dom/media/MediaDecoderStateMachineBase.cpp:167
```

EiPiFun, 2024-09-08:
I think it is necessary to add something to the doc now.
Due to license, Firefox on Linux does not support HEVC till now.

Some Pascal GPU does not support VP8, and no AV1 certainly.

I start a small pull request: https://github.com/elFarto/nvidia-vaapi-driver/pull/324

BTW, I am the uploader of nvidia-vaapi-driver for Debian and l also have an old Pascal laptop.

The nvidia-vaapi-driver is out-dated in Debian 12 and Ubuntu 22.04. I suggest you to compile it instead of using apt.

elFarto, 2024-09-10:
I don't think there's a problem here, it sounds like it's working as expected. The lack of HEVC is a Firefox issue, the README has been updated to reflect that.

closed 2024-09-10 by elFarto