elFarto/nvidia-vaapi-driver #325: Firefox - Maxwell hardware acceleration

bonswouar, 2024-09-06:
Environnement:
- GTX 970 (Maxwell)
- Debian testing
- `nvidia-driver:535`
- `nvidia-vaapi-driver:0.0.12-0.1`

I've enabled the kernel option, everything seems properly installed and working (`vainfo`, `glxinfo` and `nvidia-settings` all work as expected with nvidia output)
I've also configured the required firefox options.
But I still get no hw acceleration in `about:support`, specifically: 
```
Blocklisted; failure code FEATURE_HARDWARE_VIDEO_DECODING_NO_LINUX_NVIDIA
```
Tested with `librewolf:129.0.1-1` and `firefox-esr:115.15.0` , using `NVD_BACKEND=direct MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia`
I've tried to add  `NVD_LOG=1` but unfortunately I still see no output from cmd

Any idea what could be wrong?

P.S. I've noticed `open-gpu-kernel-modules` headers are used, although afaik my gpu isn't supported by those open modules, I hope there isn't a compatibility issue with Maxwell cards...

EiPiFun, 2024-09-08:
I think it is necessary to add something to the doc now.
Due to license, Firefox on Linux does not support HEVC till now.

I start a small pull request: https://github.com/elFarto/nvidia-vaapi-driver/pull/324

bonswouar, 2024-09-08:
I see, I should have thought about that with hevc, my bad!
So I guess having only H264 and VP8 supported is the expected behavior for this gpu?

_For some reason most videos using software decoding are laggy in ff, although my cpu isn't that bad, and no issue on Windows, but it's unrelated to this project.._

EiPiFun, 2024-09-08:
H264 and VP8 are expected.
Maxwell should support VP9 but not sure about yours.

The package in Debian contains VP9 support I confirm.

mpv could use this translation layer for HEVC.

bonswouar, 2024-09-08:
Well actually according to [nvidia support matrix](https://developer.nvidia.com/video-encode-and-decode-gpu-support-matrix-new) my GPU only supports H264, VP8, VC-1 and MPEG1/2
You can close this issue then I guess :(

_It's out of scope of this project, but if you have any suggestion to improve my video playback in web browsers on Linux I'm all ears.
I'd like to understand why videos always play smoothly on windows while they're stuttering every couple of seconds on linux.. But I guess it's not a gpu/driver problem in the end (**as I basically never watch any supported codec - basically only AV1**)_

EiPiFun, 2024-09-08:
I am not the reviewer of this project, I just come to help especially you use Debian.

For Firefox, go to about:config, media.webm.enabled=false, media.av1.enabled=false

webm means VP8 and VP9. Youtube switched to VP9 and even AV1, VP8 is not really useful.

AV1 software decoding is expensive for old CPUs.

EiPiFun, 2024-09-08:
Try to disable AV1 only, cause VP9 supports up to 2160P/4K (My GPU does not support AV1, I tried to know it). AV1 supports 8K, that is what it is designed.

VP9 is less expensive.

bonswouar, 2024-09-08:
> Try to disable AV1 only, cause VP9 supports up to 2160P/4K (My GPU does not support AV1, I tried to know it). AV1 supports 8K, that is what it is designed.
> 
> VP9 is less expensive.

I mean, the only thing I want to use is AV1, I wouldn't want to disable it :sweat_smile: 
And disabling it wouldn't explain why AV1 decoding does run smoothly on windows...

For context again, here are the resolutions available for yt if I set `media.av1.enabled=false`:
[image]

While when set to `true`:
[image]
This is a (kinda recent) known behavior of yt I believe

> Two more things:
> export MOZ_X11_EGL=1
> export MOZ_ENABLE_WAYLAND=1
> Firefox video performance is bad under Xwayland sometimes.

That could have been a good guess! But unfortunately I don't use wayland, but Cinnamon with lightdm (X11)

EiPiFun, 2024-09-08:
This is not my case.

Try to enable webm/VP9, the last suggestion.

Or use Chromium for software decoding, Firefox for hardware acceleration.

[image]

bonswouar, 2024-09-08:
Sorry I've edited my message just before you answered I believe :)

bonswouar, 2024-09-08:
> Or use Chromium for software decoding, Firefox for hardware acceleration.

FYI I've just tried with Chromium, pretty much the same behavior, except for some reason AV1 goes up to 4K while VP9 only 1440p.
And both are stuttering still..

bonswouar, 2024-09-08:
Well anyway, not really related to this project

And maybe VP9 is actually a bit less choppy, I'm gonna keep AV1 disabled just in case.
That sucks so few codecs are actually supported (if only hevc incensing wasn't that bad..), that might be time to change my gpu soon ;)

Thank you for your help!!

closed 2024-09-08 by bonswouar

bonswouar, 2024-09-09:
@EiPiFun I actually just did try again on Windows, and that's unexpected:
[image]

Despite Nvidia not having VP9 hw decoding for my gpu, it seems it is supported! Any idea how/why?

EiPiFun, 2024-09-09:
I guess that is something related to driver, beyond our control.

Maxwell has a limited support for VP9, 8 bits only, while VP9 has 10 bits and 12 bits version.
It supports and it does not. Thus I guess the driver may play an pivotal role.

Actually, I also notice some choppy sometimes, I need to refresh the web page.