elFarto/nvidia-vaapi-driver #326: HW acceleration not working on Ubuntu 22.04 and Firefox

RobertoAlda, 2024-09-16:
Hi, i can't make nvidia gpu hw decoding work in Firefox,
plaese help me.

My system is:

```
Distributor ID: Ubuntu
Description:    Ubuntu 24.04.1 LTS
Release:        24.04
Codename:       noble
```
My video card is Nvidia RTX4000 Ada SFF with 20Gb DDR6,
with driver version 550.107.02

I have succesfully built nvidia-vaapi-driver project and file /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so is created.

I have set these enviroment variables in /etc/environment file:
```
export NVD_BACKEND=direct
export MOZ_DISABLE_RDD_SANDBOX=1
export LIBVA_DRIVER_NAME=nvidia

```
ffmpeg has vaapi  in ffmpeg -hwaccels output.

After i've tried with no success the official Firefox, i've installed Firefox Nightly 132.0a1 (2024-09-16) (64-bit)
Firefox flags are:
```
 media.ffmpeg.vaapi.enabled	  = true
 media.rdd-ffmpeg.enabled	  = true
 media.av1.enabled	                  = true  (my video card has AV! hw decoding chip onboard)
 gfx.x11-egl.force-enabled	  = false (i am using wayland)
 widget.dmabuf.force-enabled = true
   ```

I start it with these flags:

```
LIBVA_MESSAGING_LEVEL=1 \
MOZ_ENABLE_WAYLAND=1 \
NVD_LOG=1  \
 firefox 
```
and when i open a H264 file or HEVC or AV1 it will not use hardware decoding.

This is the Output:

```
Gtk-Message: 17:02:17.021: Not loading module "atk-bridge": The functionality is provided by GTK natively. Please try to not load it.
libEGL warning: egl: failed to create dri2 screen

```

and this is the about:support page:

[image]

[image]

and nvidia-smi output is:
```

+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.107.02             Driver Version: 550.107.02     CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA RTX 4000 SFF Ada ...    Off |   00000000:01:00.0  On |                  Off |
| 30%   51C    P2             27W /   70W |    1130MiB /  20475MiB |     10%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      2417    C+G   ...libexec/gnome-remote-desktop-daemon        609MiB |
|    0   N/A  N/A      2480      G   /usr/bin/gnome-shell                          502MiB |
|    0   N/A  N/A      4497      G   /usr/bin/Xwayland                               3MiB |
+-----------------------------------------------------------------------------------------+
```

Plaes tell me how to make it work, if necessary i can use another browser like chromium, brave,... but i need it running on Ubuntu 22.04 or Ubuntu 24.04,

thank you.

RobertoAlda, 2024-09-17:
the nvidia-drm.modeset=1 kernel module parameter seems ok.

 cat /etc/modprobe.d/nvidia-graphics-drivers-kms.conf shows this:

```
# This file was generated by nvidia-driver-550
# Set value to 0 to disable modesetting
options nvidia-drm modeset=1
```
and 

sudo sudo dmesg | grep nvidia:
```

[    2.963662] nvidia: loading out-of-tree module taints kernel.
[    2.963668] nvidia: module license 'NVIDIA' taints kernel.
[    2.963671] nvidia: module license taints kernel.
[    3.026005] nvidia-nvlink: Nvlink Core is being initialized, major device number 235
[    3.026953] nvidia 0000:01:00.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
[    3.105589] nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  550.107.02  Wed Jul 24 23:24:27 UTC 2024
[    3.132074] [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
[    3.951690] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:01:00.0 on minor 1
[    4.340382] nvidia_uvm: module uses symbols nvUvmInterfaceDisableAccessCntr from proprietary module nvidia, inheriting taint.
[    4.372141] nvidia-uvm: Loaded the UVM driver, major device number 511.
```

Maybe am i missing something?

thesword53, 2024-09-19:
> the nvidia-drm.modeset=1 kernel module parameter seems ok.

What is the output of

```
# cat /sys/module/nvidia_drm/parameters/modeset
```
?

RobertoAlda, 2024-09-20:
The output of `sudo cat /sys/module/nvidia_drm/parameters/modeset` is:
`Y`
i added `sudo`  because without it it gives me  a permission error:

`cat: /sys/module/nvidia_drm/parameters/modeset: Permission denied`

thesword53, 2024-09-20:
Try launching Firefox with `MOZ_DRM_DEVICE=/dev/dri/renderD128` as a workaround

RobertoAlda, 2024-09-23:
# HW ACCEL WORKS IN FIREFOX, FOLLOW THESE STEPS: # 
TESTED ON:
Ubuntu Desktop 24.04.1 LTS
Kernale:6.8.0-45-generic
Nvidia RTX4000 ADA SFF 20GB
Nvidia drivers:Driver Version: 550.107.02     CUDA Version: 12.4
Firefox Nightly 132.0a1 (2024-09-22) (64-bit)

## Install Firefox Nigthly Using apt

1. Download and install the repo key:
```
wget -q https://packages.mozilla.org/apt/repo-signing-key.gpg -O- | gpg --dearmor | sudo tee /etc/apt/keyrings/packages.mozilla.org.gpg > /dev/null
```

2.add the apt repository by running the single command below:
```
echo "deb [signed-by=/etc/apt/keyrings/packages.mozilla.org.gpg] https://packages.mozilla.org/apt mozilla main" | sudo tee -a /etc/apt/sources.list.d/mozilla.list > /dev/null
```
This command will create the ‘mozilla.list‘ file under /etc/apt/sources.list.d directory and write the source line into it.

3.install the Nightly build of Firefox from that repository by running command:
```
sudo apt update
sudo apt install firefox-nightly

```

## Configure Flags:
Open firefox and go to url about:config
Set these flags:
```
gfx.webrender.all = true
media.ffmpeg.vaapi.enabled = true
media.hardware-video-decoding.enabled = true
```

closed 2024-09-28 by elFarto