elFarto/nvidia-vaapi-driver #327: vainfo fail show libva info while using nvidia libva driver

InnocentFIVE, 2024-09-23:
When I use `vainfo`, the CLI shows;
```
Trying display: wayland
Trying display: x11
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_12         : VAEntrypointVLD

```
It seems libva doesn't exist?
Here are the configuration on my PC, `fastfetch`:
```
OS: Arch Linux x86_64
Host: Dell G15 5511
Kernel: Linux 6.10.10-zen1-1-zen
Uptime: 2 hours, 15 mins
Packages: 1503 (pacman)
Shell: zsh 5.9
Display (BOE0944): 1920x1080 @ 165 Hz in 15″
DE: KDE Plasma 6.1.5
WM: KWin (X11)
WM Theme: Breeze
Theme: Breeze (Dark) [Qt], Breeze-Dark [GTK2], Breeze [GTK3]
Icons: breeze-dark [Qt], breeze-dark [GTK2/3/4]
Font: Noto Sans (10pt) [Qt], Noto Sans (10pt) [GTK2/3/4]
Cursor: Breeze_Light (24px)
Terminal: konsole 24.8.1
CPU: 11th Gen Intel(R) Core(TM) i7-11800H (16) @ 4.60 GHz
GPU 1: NVIDIA GeForce RTX 3060 Mobile / Max-Q [Discrete]
GPU 2: Intel UHD Graphics @ 1.45 GHz [Integrated]
Memory: 5.07 GiB / 15.35 GiB (33%)
Swap: Disabled
Disk (/): 80.80 GiB / 98.12 GiB (82%) - ext4
Local IP (wlp0s20f3): 172.21.31.247/17
Battery (DELL M59JH14): 100% [AC Connected]
Locale: en_US.UTF-8
```
`/etc/environment`:
```
QT_QPA_PLATFORM=xcb

NVD_LOG=1
MOZ_DISABLE_RDD_SANDBOX=1
LIBVA_DRIVER_NAME=nvidia
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json
```

`ffmpeg -hwaccels`:
```
ffmpeg version n7.0.2 Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 14.2.1 (GCC) 20240910
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-frei0r --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libdav1d --enable-libdrm --enable-libdvdnav --enable-libdvdread --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libharfbuzz --enable-libiec61883 --enable-libjack --enable-libjxl --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libplacebo --enable-libpulse --enable-librav1e --enable-librsvg --enable-librubberband --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpl --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-nvdec --enable-nvenc --enable-opencl --enable-opengl --enable-shared --enable-vapoursynth --enable-version3 --enable-vulkan
  libavutil      59.  8.100 / 59.  8.100
  libavcodec     61.  3.100 / 61.  3.100
  libavformat    61.  1.100 / 61.  1.100
  libavdevice    61.  1.100 / 61.  1.100
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
  libpostproc    58.  1.100 / 58.  1.100
Hardware acceleration methods:
vdpau
cuda
vaapi
qsv
drm
opencl
vulkan
```
`pacman -Q | grep nvidia`:
```
libva-nvidia-driver 0.0.12-1
nvidia-dkms 560.35.03-3
nvidia-settings 560.35.03-1
nvidia-utils 560.35.03-3
```
`pacman -Q | grep mesa`:
```
lib32-mesa 1:24.2.3-1
mesa 1:24.2.3-1
mesa-utils 9.0.0-4
```
`pacman -Q | grep libva`
```
libva 2.22.0-1
libva-nvidia-driver 0.0.12-1
libva-utils 2.22.0-1
```
`cat /sys/module/nvidia_drm/parameters/modeset`
```
Y
```
If you need any other info please let me know.

jeevithakannan2, 2024-09-26:
Have same issue !!

elFarto, 2024-09-28:
I'm not sure what the issue is here. libva must exist for vainfo to work. There is nothing in the output that looks incorrect.

closed 2024-10-19 by elFarto