elFarto/nvidia-vaapi-driver #328: HEVC support

RobertoAlda, 2024-10-02:
HI,
i've succesfully make it work with H264,VP9 and AV1 accelerated by my hardware  NVIDIA RTX 4000 ADA SFF.
Now i would like to hw accelerated also HEVC files.
I know that firefox does not support it beacuase of pattent resaons, but do you know any other custom fork of FIrefox that work with HEVC?

Any help would be appreciated, thank you.

elFarto, 2024-10-19:
I'm not sure there are any forks of Firefox that support HEVC. You'll likely have to build it yourself, but I'm not sure what changes are needed to enable support for HEVC.

mirh, 2024-10-25:
Duplicate of #304

thesword53, 2025-01-06:
This driver already supports HEVC and you can check with MPV: `mpv --hwdec=vaapi hevc_video.mp4`.

alastor0325, 2025-02-18:
Firefox will soon support HEVC on Linux, with implementation currently tracked in [bug 1894818](https://bugzilla.mozilla.org/show_bug.cgi?id=1894818).

closed 2026-09-26 by elFarto