elFarto/nvidia-vaapi-driver #329: libva error: /usr/lib/dri/nvidia_drv_video.so init failed

littleblack111, 2024-10-03:
only occurs in wayland. im using a desktop. display plugged in to nvidia gpu

followed arch wiki to install the proper drivers. not a dupe. tried specifying the card but still init failed

AmmoniumX, 2024-10-03:
Hi, I am also having this exact issue. I saw a previous similar issue on https://github.com/elFarto/nvidia-vaapi-driver/issues/160 but none of the solutions suggested there work.

Here's more information:
```
❯ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo
Trying display: wayland
       678.478549093 [3653-3653] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
       678.478563052 [3653-3653] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
       678.478589619 [3653-3653] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
       678.538341590 [3653-3653] ../nvidia-vaapi-driver-0.0.12/src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
       678.538375324 [3653-3653] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2248       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

The linked issue suggested to also set 'NVD_BACKEND=direct', but it has no effect on it for me

littleblack111, 2024-10-03:
theres multiple issues with this. and i do have NVD_BACKEND set to direct

and my vainfo works

just electron doesn't work... with libva error: /usr/lib/dri/nvidia_drv_video.so init failed

littleblack111, 2024-10-04:
yep. same here ```vainfo --display drm``` doesnt work but vainfo does

littleblack111, 2024-10-11:
hmm specifying --device after --display drm seem to work:
```
╰─$ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo --display drm --device /dev/dri/renderD129                ─╯
Trying display: drm
    137368.971182579 [3878543-3878543] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
    137368.971188124 [3878543-3878543] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
    137368.971189253 [3878543-3878543] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
    137368.982933737 [3878543-3878543] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
    137368.982951090 [3878543-3878543] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 560.35.03, major version: 560, minor version: 35
    137368.982954946 [3878543-3878543] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 1 2 6
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
    137369.151778688 [3878543-3878543] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2098              nvTerminate Terminating 0x5695474c68e0
    137369.152201023 [3878543-3878543] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances
```

fknemi, 2024-10-18:
I am having the same issue... anything chromium/electron doesn't seem to work

Icey-Glitch, 2025-01-08:
im getting a similar issue

```
2225.113819019 [166908-166950] ../src/vabackend.c: 168                     init CUDA ERROR 'unknown error' (999)

      2225.113838198 [166908-166950] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
      2225.113839994 [166908-166950] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2225.113841504 [166908-166950] ../src/vabackend.c:2222       __vaDriverInit_1_0 Selecting Direct backend
      2225.121713375 [166908-166950] ../src/direct/nv-driver.c: 295            init_nvdriver Initing nvdriver...
      2225.121728331 [166908-166950] ../src/direct/nv-driver.c: 313            init_nvdriver NVIDIA kernel driver version: 565.77, major version: 565, minor version: 77
      2225.121731203 [166908-166950] ../src/direct/nv-driver.c: 320            init_nvdriver Got dev info: 100 1 2 6
      2225.122296152 [166908-166950] ../src/direct/direct-export-buf.c:  27       findGPUIndexFromFd CUDA ERROR 'initialization error' (3)

      2225.122299926 [166908-166950] ../src/vabackend.c:2252       __vaDriverInit_1_0 CUDA ERROR 'initialization error' (3)
```