elFarto/nvidia-vaapi-driver #330: libva selecting wrong DRM device by default: isNvidiaDrmFd Invalid driver for DRM device: i915

AmmoniumX, 2024-10-03:
Hi, I'm having an issue getting libva to run correctly by default. I have a dual GPU setup with my laptop containing an intel iGPU and an nvidia GPU, and it seems to be trying to use the intel DRM device when it should be using nvidia's instead.

```
❯ cat /etc/environment
QT_QPA_PLATFORM=wayland
VDPAU_DRIVER=nvidia
LIBVA_DRIVER_NAME=nvidia
__NV_PRIME_RENDER_OFFLOAD=1
__GLX_VENDOR_LIBRARY_NAME=nvidia
```

```
❯ NVD_LOG=1 vainfo
Trying display: wayland
       226.796108304 [2258-2258] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
       226.796141537 [2258-2258] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
       226.796156939 [2258-2258] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
       226.889169884 [2258-2258] ../nvidia-vaapi-driver-0.0.12/src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
       226.889199158 [2258-2258] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2248       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

It works if I manually tell it which DRM device to use:
```
❯ NVD_LOG=1 vainfo --display drm --device /dev/dri/renderD128
Trying display: drm
       298.889134944 [2323-2323] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
       298.889166533 [2323-2323] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
       298.889182516 [2323-2323] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
       298.981971645 [2323-2323] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
       298.982077546 [2323-2323] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 560.35.03, major version: 560, minor version: 35
       298.982103456 [2323-2323] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 1 0 fe
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
       299.106684460 [2323-2323] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2098              nvTerminate Terminating 0x5f648d71d8e0
       299.106872810 [2323-2323] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances
```

How can I fix it to use the right drm device by default? I saw a suggestion that it could work by setting the `VAAPI_DEVICE` env var but it doesn't work for me:
```
❯ VAAPI_DEVICE=/dev/dri/by-path/pci-0000:01:00.0-render NVD_LOG=1 vainfo
Trying display: wayland
       567.993644922 [2801-2801] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
       567.993658274 [2801-2801] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
       567.993685181 [2801-2801] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
       568.054314182 [2801-2801] ../nvidia-vaapi-driver-0.0.12/src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
       568.054348233 [2801-2801] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2248       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

AmmoniumX, 2024-10-03:
Update: For some reason, it only happens when trying in `wayland`, x11 and drm work fine:
```
❯ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo --display wayland
Trying display: wayland
      1080.159804159 [13860-13860] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
      1080.159833802 [13860-13860] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1080.159845107 [13860-13860] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
      1080.255231340 [13860-13860] ../nvidia-vaapi-driver-0.0.12/src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
      1080.255268002 [13860-13860] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2248       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit

❯ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo --display x11
Trying display: x11
      1083.000119309 [13892-13892] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      1083.000154076 [13892-13892] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1083.000170411 [13892-13892] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
      1083.096400239 [13892-13892] ../nvidia-vaapi-driver-0.0.12/src/direct/direct-export-buf.c:  68      direct_initExporter Searching for GPU: 0 0 128
      1083.096486713 [13892-13892] ../nvidia-vaapi-driver-0.0.12/src/direct/direct-export-buf.c:  90      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
      1083.096513724 [13892-13892] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
      1083.096686697 [13892-13892] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 560.35.03, major version: 560, minor version: 35
      1083.096718756 [13892-13892] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 1 0 fe
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      1083.228265621 [13892-13892] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2098              nvTerminate Terminating 0x6495febb5980
      1083.228400129 [13892-13892] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances

❯ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo --display drm
Trying display: drm
      1085.218259496 [13926-13926] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2188       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      1085.218292970 [13926-13926] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2197       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1085.218311841 [13926-13926] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2223       __vaDriverInit_1_0 Selecting Direct backend
      1085.314105336 [13926-13926] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 267            init_nvdriver Initing nvdriver...
      1085.314210774 [13926-13926] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 560.35.03, major version: 560, minor version: 35
      1085.314236794 [13926-13926] ../nvidia-vaapi-driver-0.0.12/src/direct/nv-driver.c: 292            init_nvdriver Got dev info: 100 1 0 fe
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      1085.431592634 [13926-13926] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2098              nvTerminate Terminating 0x5e6a31003a30
      1085.431742010 [13926-13926] ../nvidia-vaapi-driver-0.0.12/src/vabackend.c:2112              nvTerminate Now have 0 (0 max) instances
```

AmmoniumX, 2024-10-03:
(Final?) update: I was trying some stuff, and I managed to make vainfo find the right DRM device by modifying this variable on `vabackend.c` to search for it when on wayland instead of trusting the fd given by `ctx->drm_state`:

```
    //drm_state can be passed in with any display type, including X11. But if it's X11, we don't
    //want to use the fd as it'll likely be an Intel GPU, as NVIDIA doesn't support DRI3 at the moment
    // bool isDrm = ctx->drm_state != NULL && ((struct drm_state*) ctx->drm_state)->fd > 0 &&
    //              (((ctx->display_type & VA_DISPLAY_MAJOR_MASK) == VA_DISPLAY_DRM) ||
    //               ((ctx->display_type & VA_DISPLAY_MAJOR_MASK) == VA_DISPLAY_WAYLAND));

    // fix: ignore fd on wayland as well
    bool isDrm = ctx->drm_state != NULL && ((struct drm_state*) ctx->drm_state)->fd > 0 &&
                 ((ctx->display_type & VA_DISPLAY_MAJOR_MASK) == VA_DISPLAY_DRM);
```

I am not sure if this truly is a necessary "fix" or just a band-aid patch to a misconfiguration on my system, so I'm unsure if I should make a PR or not. However, it *does* fix my problem.
```
❯ NVD_LOG=1 vainfo --display wayland
Trying display: wayland
      4318.425692246 [22697-22697] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
      4318.425705492 [22697-22697] ../src/vabackend.c:2199       __vaDriverInit_1_0 isDrm: 0
      4318.425708799 [22697-22697] ../src/vabackend.c:2202       __vaDriverInit_1_0 Now have 0 (0 max) instances
      4318.425714555 [22697-22697] ../src/vabackend.c:2228       __vaDriverInit_1_0 Selecting Direct backend
      4318.448645038 [22697-22697] ../src/direct/direct-export-buf.c:  68      direct_initExporter Searching for GPU: 0 0 128
      4318.448666925 [22697-22697] ../src/direct/direct-export-buf.c:  90      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
      4318.448671215 [22697-22697] ../src/direct/nv-driver.c: 295            init_nvdriver Initing nvdriver...
      4318.448693276 [22697-22697] ../src/direct/nv-driver.c: 313            init_nvdriver NVIDIA kernel driver version: 560.35.03, major version: 560, minor version: 35
      4318.448698426 [22697-22697] ../src/direct/nv-driver.c: 320            init_nvdriver Got dev info: 100 1 0 fe
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      4318.522116745 [22697-22697] ../src/vabackend.c:2095              nvTerminate Terminating 0x5b6b52ed6780
      4318.522207764 [22697-22697] ../src/vabackend.c:2109              nvTerminate Now have 0 (0 max) instances
```

acdcbyl, 2024-11-05:
Hey, I think you're a genius.

AmmoniumX, 2024-11-08:
Fixed on c1a6f1469cc00739baff10a40ac0e583c77c74b4

closed 2024-11-08 by AmmoniumX