elFarto/nvidia-vaapi-driver #332: Unable to enable vaapi in Librewolf despite succesful debugging

newmindarchitech, 2024-10-05:
Hello, I am writing this to let you know that I couldn't enable vaapi in Librewolf(fork of firefox), or at least have any indicating signs that it worked despite many debugging attempts not showing any errors. 

I have installed the necessary dependencies  stated by  the guide(installing ffnvcodec-headers, gst-plugins-bad-libs and gst-plugins-bad from the AUR because I use Arch ), then I installed the libva-nvidia-driver. 

I disabled the Performance option in the browser because it was the main cause of the error( libva error : /usr/lib/dri/nvidia_drv_video.so init failed). Then I followed the guide in enabling vaapi in librewolf(media.ffmpeg.vaapi.enabled = true, widget.dmabuf.force-enabled = true,media.rdd-ffmpeg.enabled=true(by default)). Lastly, I exported some environment variables according to the guide in .bashrc (MOZ_DISABLE_RDD_SANDBOX=1 and LIBVA_DRIVER_NAME=nvidia). 

After all of this, I restarted librewolf and play a video while opening nvidia-settings and there was no video engine utilization section when i clicked the name of my gpu. I checked with nvidia-smi and I didn't see a duplicate of librewolf under C type. I have debugged librewolf by journalctl and nothing seems to indicate an error, I even put NVD_LOG=1 in vainfo and there was no error. Are there any additional steps I need to do because I have installed firefox thinking that these would only apply for firefox and the same thing happened.Here are the pictures of the debugging

Picture 1: vainfo
[image]
Picture 2: Processes running while opening librewolf using btop
[image]
Picture 3: nvidia-smi
[image]
Picture 4: ffmpeg -hwaccels
[image]
Picture 5 : Disabling Performance option
[image]

elFarto, 2024-10-19:
Can you take a look in the `about:support` page?

newmindarchitech, 2024-10-21:
Could these be the problem

newmindarchitech, 2024-11-12:
Does this mean that hardware video decoding is working when I watch youtube videos in HDR and other videos in HD resolutions ?
[image]
 All I did was enable layers acceleration(layers.acceleration.disabled=false) in about:config

elFarto, 2024-11-12:
That last screen shot is showing that there is some hardware acceleration occurring. You could set `NVD_LOG=1` when running the browser to double check.

newmindarchitech, 2024-11-12:
Here is the log when NVD_LOG=1 librewolf:
NVD_LOG=1 librewolf 
       179.660447301 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
       179.660454648 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:2196       __vaDriverInit_1_0 Now have 0 (0 max) instances
       179.660458067 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:2222       __vaDriverInit_1_0 Selecting Direct backend
       179.665778864 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/direct/nv-driver.c: 295            init_nvdriver Initing nvdriver...
       179.665792829 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/direct/nv-driver.c: 313            init_nvdriver NVIDIA kernel driver version: 565.57.01, major version: 565, minor version: 57
       179.665794918 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/direct/nv-driver.c: 320            init_nvdriver Got dev info: 100 1 2 6
       179.754857345 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1443      nvQueryImageFormats In nvQueryImageFormats
       179.862650908 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 672           nvCreateConfig got profile: 6 with 0 attributes
       179.862660942 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) (nil) 0
       179.862663109 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) 0x7ef621b5bf40 8
       179.867373493 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       179.867384999 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 672           nvCreateConfig got profile: 7 with 0 attributes
       179.867389167 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) (nil) 0
       179.867390907 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) 0x7ef621b5b4c0 8
       179.872933006 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       179.872941767 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 672           nvCreateConfig got profile: 13 with 0 attributes
       179.872944908 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) (nil) 0
       179.872946626 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) 0x7ef621b5b4c0 8
       179.878432159 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       179.878440712 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 672           nvCreateConfig got profile: 14 with 0 attributes
       179.878443713 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 9 (8) (nil) 0
       179.878445348 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 9 (8) 0x7ef621b5b4c0 8
       179.883807093 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       179.883814943 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 672           nvCreateConfig got profile: 19 with 0 attributes
       179.883817842 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 10 (8) (nil) 0
       179.883819611 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 10 (8) 0x7ef621b5bac0 8
       179.888163088 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
       179.888170287 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 672           nvCreateConfig got profile: 32 with 0 attributes
       179.888172692 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 735           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
       179.888175222 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 11 (8) (nil) 0
       179.888176821 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 11 (8) 0x7ef621b5bac0 8
       179.893383091 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
       179.893390011 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 672           nvCreateConfig got profile: 21 with 0 attributes
       179.893392701 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 10 (10) (nil) 0
       179.893394420 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 10 (10) 0x7ef621b5b4c0 8
       179.898494461 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
       179.898671439 [4209-4217] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:2095              nvTerminate Terminating 0x7ef648d665c0
[... cut ...]
       215.383404116 [4209-4694] ../nvidia-vaapi-driver-0.0.13/src/direct/direct-export-buf.c: 160 direct_allocateBackingImage Allocate Buffer: 61 99 100
       215.383408853 [4209-4694] ../nvidia-vaapi-driver-0.0.13/src/direct/direct-export-buf.c: 170 direct_allocateBackingImage Importing memory to CUDA

And when I MOZ_LOG="FFmpegVideo:5" librewolf,I do see the VA-API string in the terminal output. So If you think with these results, VAA-API is working on my part you can close this and maybe inform others of another way to check if hardware acceleration on firefox on NVIDIA is working

elFarto, 2024-11-14:
Yes, those log files suggest it's working correctly.

closed 2024-11-14 by elFarto