elFarto/nvidia-vaapi-driver #335: Followed setup instructions but doesn't work (Ubuntu 24.10, Firefox 131.0.3 deb/flatpak)

Novantric, 2024-10-26:
As of the title, I enabled every about:config option I could, followed the setup guide to the letter with env vars and the like in addition to the GRUB launch options.
Vainfo likes my config:
[image]
But Firefox does not.
[image]
What am I doing wrong? Can provide log files.

elFarto, 2025-03-02:
Can you run firefox with `NVD_LOG=1` set, and post the logs here.

Novantric, 2025-03-04:
Sure, but how do I do that? Kinda a noob.

Novantric, 2025-03-04:
Got this.
[image]

Novantric, 2025-03-05:
@elFarto here, managed to get it to work via an Environment Variable.

[image]

```
novantric@Deep-Thought:~$ flatpak run --env=NVD_LOG=1 org.mozilla.firefox
[RDD 95: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in RDD process
[RDD 95: Main Thread]: D/PlatformDecoderModule FFMPEG: version: 0x80, macro: 60, micro: 100, isFFMpeg: yes
[RDD 95: Main Thread]: D/PlatformDecoderModule FFMPEG: version: 0x100, macro: 61, micro: 100, isFFMpeg: yes
[RDD 95: Main Thread]: D/PlatformDecoderModule FFVPX: Link result: Success
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/avc'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/hevc'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/hevc'
[RDD 95: Main Thread]: D/PlatformDecoderModule Agnostic decoder rejects requested type 'video/hevc'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/mp4a-latm'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/opus'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/opus'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/vorbis'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/vorbis'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/flac'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/flac'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/x-wav'
[RDD 95: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/x-wav'
[Parent 2: Main Thread]: D/PlatformDecoderModule Broadcast support from 'RDD', support=H264 SW
VP9 SW
VP8 SW
AV1 SW
HEVC NONE
AAC SW
MP3 SW
Opus SW
Vorbis SW
FLAC SW
Wave SW
```

Novantric, 2025-03-05:
Went through the setup instructions again after adding MOZ_DISABLE_RDD_SANDBOX=1 to the Flatpak environment variables and this output seems more verbose but still doesn't work.

<details>
<summary>Code output</summary>

### You can add a header

```
novantric@Deep-Thought:~$ flatpak run --env=NVD_LOG=1 org.mozilla.firefox
[RDD 86: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in RDD process
[RDD 86: Main Thread]: D/PlatformDecoderModule FFMPEG: version: 0x80, macro: 60, micro: 100, isFFMpeg: yes
[RDD 86: Main Thread]: D/PlatformDecoderModule FFMPEG: version: 0x100, macro: 61, micro: 100, isFFMpeg: yes
[RDD 86: Main Thread]: D/PlatformDecoderModule FFVPX: Link result: Success
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/avc'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/hevc'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/hevc'
[RDD 86: Main Thread]: D/PlatformDecoderModule Agnostic decoder rejects requested type 'video/hevc'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/mp4a-latm'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/opus'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/opus'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/vorbis'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/vorbis'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/flac'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/flac'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/x-wav'
[RDD 86: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/x-wav'
[Parent 2: Main Thread]: D/PlatformDecoderModule Broadcast support from 'RDD', support=H264 SW
VP9 SW
VP8 SW
AV1 SW
HEVC NONE
AAC SW
MP3 SW
Opus SW
Vorbis SW
FLAC SW
Wave SW
[Child 241: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 241: Main Thread]: D/PlatformDecoderModule Sandbox GPU decoder supports requested type video/avc
[Child 241: Main Thread]: D/PlatformDecoderModule Sandbox GPU decoder supports requested type video/avc
[Child 241: Main Thread]: D/PlatformDecoderModule Sandbox GPU decoder supports requested type video/av1
[Child 241: Main Thread]: D/PlatformDecoderModule Sandbox GPU decoder supports requested type video/av1
[Child 241: Main Thread]: D/PlatformDecoderModule Sandbox GPU decoder supports requested type video/avc
[Child 241: Main Thread]: D/PlatformDecoderModule Sandbox GPU decoder supports requested type video/avc
[Child 241: MediaPDecoder #1]: D/PlatformDecoderModule Sandbox GPU decoder supports requested type video/av1
[Child 241: MediaPDecoder #1]: D/PlatformDecoderModule Sandbox GPU decoder supports requested type video/av1
[Child 241: MediaPDecoder #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/av1
[RDD 86: MediaSupervisor #1]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 86: MediaSupervisor #1]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[Child 241: RemVidChild]: D/PlatformDecoderModule RemoteMediaDataDecoder[779250ee6b80] ::RemoteMediaDataDecoder: 779250ee6b80 is created
[Child 241: RemVidChild]: D/PlatformDecoderModule RemoteMediaDataDecoder[779250ee6b80] ::operator(): 779250ee6b80 RemoteDecoderChild has been initialized - description: ffvpx video decoder (RDD remote), process: rdd, codec: av1
[Child 241: RemVidChild]: D/PlatformDecoderModule RemoteMediaDataDecoder[779250ee6b80] ::~RemoteMediaDataDecoder: 779250ee6b80 is released
[Child 241: Main Thread]: D/PlatformDecoderModule Sandbox GPU decoder supports requested type video/vp9
[Child 241: Main Thread]: D/PlatformDecoderModule Sandbox GPU decoder supports requested type video/vp9
[Child 241: Main Thread]: D/PlatformDecoderModule Sandbox GPU decoder supports requested type video/avc
[... cut ...]
[Child 241: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[77924c2d5e60] ::HandleDecodedResult: HandleDecodedResults: 1 decoded data, 1 trimmers
[Child 241: MediaPDecoder #2]: V/PlatformDecoderModule AudioTrimmer[77924c2d5e60] ::HandleDecodedResult: decoded buffer [{21000,1000000}, {41000,1000000}] doesn't need trimming
[RDD 86: MediaPDecoder #1]: V/PlatformDecoderModule ProcessDecode: mDuration=16000µs ; mTime=67000µs ; mTimecode=67000µs
[Child 241: MediaPDecoder #2]: D/PlatformDecoderModule AudioTrimmer[77924c2d5e60] ::Decode: AudioTrimmer::Decode

```

</details>