elFarto/nvidia-vaapi-driver #336: nvidia-smi doesn't show a C in the process type column for firefox thread despite all other checks looking okay everywhere else.

FAYZER77, 2024-10-27:
Distro: Arch Linux 
Laptop: Asus G733CX (2022 ROG Strix) in Discreet (Ultimate) GPU Mode
NVIDIA Driver: 565.57.01 (Beta Branch)

/etc/environment:
CHROME_EXECUTABLE=google-chrome-stable
PATH=/usr/share/perl6/site/bin/:$PATH
JAVA_HOME=/usr/lib/jvm/default
MOZ_DISABLE_RDD_SANDBOX=1
SDL_SOUNDFONTS="/usr/share/soundfonts/Utopia Audigy 2.0 S.sf2"
LIBVA_DRIVER_NAME=nvidia
VDPAU_DRIVER=nvidia
NVD_BACKEND=direct

firefox config:
gfx.x11-egl.force-enabled	true
widget.dmabuf.force-enabled	true
media.ffmpeg.vaapi.enabled	true

vainfo output:
rying display: wayland
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_12         : VAEntrypointVLD

[image]

nvidia-smi
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 565.57.01              Driver Version: 565.57.01      CUDA Version: 12.7     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3080 ...    Off |   00000000:01:00.0  On |                  N/A |
| N/A   54C    P0             43W /  175W |    1839MiB /  16384MiB |      2%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      1178      G   /usr/lib/Xorg                                  35MiB |
|    0   N/A  N/A      1286      G   /usr/bin/kwalletd6                              3MiB |
|    0   N/A  N/A      1323      G   /usr/bin/kwin_wayland                         322MiB |
|    0   N/A  N/A      1422      G   /usr/bin/Xwayland                              15MiB |
|    0   N/A  N/A      1461      G   /usr/bin/ksmserver                              3MiB |
|    0   N/A  N/A      1463      G   /usr/bin/kded6                                  3MiB |
|    0   N/A  N/A      1477      G   /usr/bin/plasmashell                          212MiB |
|    0   N/A  N/A      1505      G   /usr/lib/kactivitymanagerd                      3MiB |
|    0   N/A  N/A      1540      G   /usr/bin/gmenudbusmenuproxy                     3MiB |
|    0   N/A  N/A      1541      G   /usr/bin/kaccess                                3MiB |
|    0   N/A  N/A      1542      G   ...b/polkit-kde-authentication-agent-1          3MiB |
|    0   N/A  N/A      1547      G   /usr/lib/org_kde_powerdevil                     3MiB |
|    0   N/A  N/A      1548      G   /usr/lib/xdg-desktop-portal-kde                 3MiB |
|    0   N/A  N/A      1549      G   /usr/bin/xembedsniproxy                         3MiB |
|    0   N/A  N/A      1773      G   /usr/bin/kclockd                                3MiB |
|    0   N/A  N/A      1776      G   /usr/bin/kdeconnectd                            3MiB |
|    0   N/A  N/A      1780    C+G   /usr/bin/xwaylandvideobridge                   16MiB |
|    0   N/A  N/A      1782      G   /usr/bin/yakuake                                3MiB |
|    0   N/A  N/A      1805      G   /usr/lib/DiscoverNotifier                       3MiB |
|    0   N/A  N/A      1808      G   /usr/bin/kalendarac                             3MiB |
|    0   N/A  N/A      3724      G   /usr/lib/firefox/firefox                      346MiB |
|    0   N/A  N/A      3925      G   ...bin/plasma-browser-integration-host          3MiB |
|    0   N/A  N/A      7472      G   ...erProcess --variations-seed-version         94MiB |
|    0   N/A  N/A     30763      G   /usr/bin/konsole                                3MiB |
|    0   N/A  N/A     35394      G   /usr/lib/kf6/kioworker                          3MiB |
|    0   N/A  N/A     35486      G   /usr/bin/spectacle                             37MiB |
+-----------------------------------------------------------------------------------------+

thesword53, 2024-10-28:
Did you set `media.ffmpeg.vaapi.enabled` to `true` in `about:config` ?

FAYZER77, 2024-10-28:
Yes, that is listed in my post (yes, I realize there is a lot there).   If I boot into hybrid mode and use the iHD intel vaapi driver, hardware acceleration is working in the browser, but in discreet mode using this driver, it's not working in the browser anymore.

I'm not even sure it's a problem with this project, it could the video driver, or it could even be a change arch linux to it's config, because I've seen it run fine under Fedora, but I can't get it to work in Arch Linux anymore.  I've tested under Nvidia 560 driver and the new beta 565 driver and neither work (in Arch Linux).  I just don't know how to dig deeper.  I'm not even getting console output from Firefox to give me a clue.

FAYZER77, 2024-12-07:
Retested with the latest version with CachyOS (Arch based) and 565.77 driver and a separate firefox thread thread now shows up and GPU-Util number is now back to what I would expect.  Closing issue.

closed 2024-12-07 by FAYZER77