elFarto/nvidia-vaapi-driver #337: Upgraded Nvidia driver, no longer working.

ddouglas87, 2024-11-02:
I've been using nvidia-vaapi-driver for years and it has worked wonderfully in Firefox on youtube.  On Nvidia 550 drivers it worked well.

I did a video driver update today to driver version 565.57.01 and now I'm no longer getting hardware acceleration on Youtube videos.  It's the only change so I'm pretty sure it has something to do with the drivers.

My GPU is a GeForce 980.  I'm using the plugin h264ify to force Youtube to play h264 (avc1) so I get video hardware decoding.

My OS is OpenSUSE Tumbleweed, which is a rolling distro with newest libraries and apps. 

```
bunny@happy:~> zy info nvidia-vaapi-driver
Loading repository data...
Reading installed packages...

Information for package nvidia-vaapi-driver:
--------------------------------------------
Repository     : home:Dead_Mozay
Name           : nvidia-vaapi-driver
Version        : 0.0.12-1.11
Arch           : x86_64
Vendor         : obs://build.opensuse.org/home:Dead_Mozay
Installed Size : 114.8 KiB
Installed      : Yes
Status         : up-to-date
Source package : nvidia-vaapi-driver-0.0.12-1.11.src
Upstream URL   : https://github.com/elFarto/nvidia-vaapi-driver
Summary        : A VA-API implemention using NVIDIA's NVDEC
Description    : 
    This is an VA-API implementation that uses NVDEC as a backend.
    This implementation is specifically designed to be used by Firefox for accelerated decode of web content,
    and may not operate correctly in other applications.
```

```
bunny@happy:~> firefox --version
Mozilla Firefox 131.0.3
```

elFarto, 2024-11-02:
I've just installed that driver and decoding is working fine here. Can you try running the latest version of this driver from master?

ddouglas87, 2024-11-03:
I uninstalled my OS' version of nvidia-vaapi-driver.

```
bunny@happy:~/software> git clone https://github.com/elFarto/nvidia-vaapi-driver.git
Cloning into 'nvidia-vaapi-driver'...
remote: Enumerating objects: 1785, done.
remote: Counting objects: 100% (603/603), done.
remote: Compressing objects: 100% (159/159), done.
remote: Total 1785 (delta 511), reused 450 (delta 444), pack-reused 1182 (from 1)
Receiving objects: 100% (1785/1785), 595.35 KiB | 3.50 MiB/s, done.
Resolving deltas: 100% (1229/1229), done.
bunny@happy:~/software> cd nvidia-vaapi-driver/
bunny@happy:~/software/nvidia-vaapi-driver> meson setup build
The Meson build system
Version: 1.6.0
Source dir: /home/bunny/software/nvidia-vaapi-driver
Build dir: /home/bunny/software/nvidia-vaapi-driver/build
Build type: native build
Project name: nvidia-vaapi-driver
Project version: 0.1
C compiler for the host machine: cc (gcc 14.2.1 "cc (SUSE Linux) 14.2.1 20241007 [revision 4af44f2cf7d281f3e4f3957efce10e8b2ccb2ad3]")
C linker for the host machine: cc ld.bfd 2.43.1.20240828-2
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library m found: YES
Library dl found: YES
Found pkg-config: YES (/usr/bin/pkg-config) 2.2.0
Run-time dependency egl found: YES 1.5
Run-time dependency ffnvcodec found: YES 12.0.16.1
Run-time dependency libdrm found: YES 2.4.123
Run-time dependency threads found: YES
Run-time dependency libva found: YES 1.22.0
Run-time dependency gstreamer-codecparsers-1.0 found: YES 1.24.8
Compiler for C supports arguments -Wno-missing-field-initializers: YES 
Compiler for C supports arguments -Wno-unused-parameter: YES 
Compiler for C supports arguments -Werror=format: YES 
Compiler for C supports arguments -Werror=format-security: NO 
Compiler for C supports arguments -Werror=incompatible-pointer-types: YES 
Compiler for C supports arguments -Werror=init-self: YES 
Compiler for C supports arguments -Werror=int-conversion: YES 
Compiler for C supports arguments -Werror=missing-declarations: YES 
Compiler for C supports arguments -Werror=missing-prototypes: YES 
Compiler for C supports arguments -Werror=pointer-arith: YES 
Compiler for C supports arguments -Werror=undef: YES 
Compiler for C supports arguments -Werror=vla: YES 
Compiler for C supports arguments -Wsuggest-attribute=format: YES 
Compiler for C supports arguments -Wwrite-strings: YES 
Build targets in project: 1

Found ninja-1.12.1 at /usr/bin/ninja
bunny@happy:~/software/nvidia-vaapi-driver> meson install -C build
ninja: Entering directory `/home/bunny/software/nvidia-vaapi-driver/build'
[16/16] Linking target nvidia_drv_video.so
Installing nvidia_drv_video.so to /usr/lib64/dri
Installation failed due to insufficient permissions.
Attempt to use /usr/bin/sudo to gain elevated privileges? [y/n] y
Installing nvidia_drv_video.so to /usr/lib64/dri
```

It works! 🥳

Apologies.  I should have tried the master version first.  I don't know if it's an issue with the way my system's package manager installs it, or the older version doesn't work with the newer drivers.  Maybe it's getting time to release a new version so this potential issue does not happen to anyone else?

elFarto, 2024-11-03:
It's not your fault, I should have got around to releasing a new version a while ago.

closed 2024-11-03 by elFarto