elFarto/nvidia-vaapi-driver #340: Some Twitter video GIFs display green bars

ToadKing, 2024-11-20:
Like the title says. I've only encountered this issue on Twitter video GIFs, all other sites like YouTube and normal Twitter videos work fine.

Driver: 565.57.01
GPU: GeForce RTX 2070

Example: https://video.twimg.com/tweet_video/GcylK7VaYAAS-RO.mp4

What I see:
[image]

What it should look like:
[image]

thesword53, 2024-12-04:
I can reproduce the issue with MPV using VAAPI.

rakslice, 2024-12-05:
I'm getting a similar thing, not just on twitter, but other very small resolution videos used in place of gifs these days; codec doesn't seem to matter

https://t.gifs.bsky.app/MsP-20r81QYAAAP3/ewok-starwars.webm
320x122 vp8
https://files.mastodon.social/cache/media_attachments/files/113/557/839/461/281/307/original/af1f5eee037a898a.mp4
200x122 avc1

rakslice, 2024-12-05:
For clarification of @ToadKing 's example, that second frame is from later in the video; the grass and sky colours in the non-green bands in the first frame are mostly accurate, except as you can see the ghost of the reddish thing sticking out of the ground about a green band's width to the right of where it should be... a chunk of the chroma is being shifted horizontally?

rakslice, 2024-12-06:
By some printf debugging, all of these examples are in the case suggested in `calculate_image_size` where one plane has `planeHeight` < 88 and another not, and the old code would have used different `log2GobsPerBlockY` values but that can't be done anymore
https://github.com/elFarto/nvidia-vaapi-driver/blob/259b7b7b7c6891805fbfb8f799d12ea03bd260f7/src/direct/nv-driver.c#L498

rakslice, 2024-12-06:
Is there a way it could error out of this case that could possibly fall back to something else rather than just going ahead with parameter vals that are definitely wrong for the image format

rakslice, 2024-12-06:
Okay so what I'm seeing on my GTX 1070 where this is NV12, in whatever situation `calculate_image_size` is leaving for these cases where `planeHeight < 88` for the UV, the organization of the destination arrays that the direct `copyFrameToSurface` is copying to becomes weird: a bunch of non-contiguous 64-pixel-wide strips, only using half the x space and double the y space as the `CUDA_MEMCPY2D` sees it.

If this striping is somehow some transformation you could deal with by just fiddling with the parameters of the `CUDA_MEMCPY2D` I can't figure it out but I can just copy them each separately.

Here's what my hacky workaround for it looks like; I have no idea how generally working this is:

https://github.com/elFarto/nvidia-vaapi-driver/compare/master...rakslice:nvidia-vaapi-driver:nv12_short_images_fix

elFarto, 2024-12-07:
I've reverted the single-plane export in master. Can you retest and make sure it fixes it for you?

rakslice, 2024-12-07:
@elFarto Yup, works in current master 4a0eb9e1f3759030c5d2ab124601af1c2d3f9548

ToadKing, 2024-12-19:
Can also confirm this now works under latest master.

closed 2024-12-19 by ToadKing

rakslice, 2024-12-20:
@elFarto 
Just found another video that is still giving a similar looking artifact with current master:
https://t.gifs.bsky.app/qgTXJF6Bgz0AAAP3/jeff-the-land-shark-marvel-rivals.webm
(vp8, 214x172)
I haven't dug into this yet but I see that compared to my branch the regions with the UV missing are in a different place:
[image]
master 9c4a3c70b68467343262d295ce56a56d510ec342
my nv12_short_images_fix:
[image]

ToadKing, 2024-12-24:
@rakslice It probably makes sense to make a new issue for that since it's a (similar but) different issue than the one I reported.