elFarto/nvidia-vaapi-driver #343: Decode error when playing looped video in Firefox.

Kam1k4dze, 2024-12-09:
When looping is enabled and video starts playing again, a decoding error occurs. It can happen on the first loop or on later ones.
[image]
Example video with issue: [https://leonardo.osnova.io/be20671e-af21-5324-b36a-78597022c012/-/format/mp4/](video)
[image]
`Media resource https://leonardo.osnova.io/be20671e-af21-5324-b36a-78597022c012/-/format/mp4/ could not be decoded, error: Error Code: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005)`

elFarto, 2024-12-10:
I'm not seeing the same issue here. I have just pushed a few more fixes to master, could you retest?

Kam1k4dze, 2024-12-10:
Nothing has changed. If I reproduce it inside MPV with hwdec=VAAPI and the --loop flag. I also get errors:
```
$ mpv --loop Downloads/ieM4Vz7Z.mp4
● Video  --vid=1  (h264 498x498 14.2857 fps) [default]
Using hardware decoding (vaapi).
VO: [gpu-next] 498x498 vaapi[nv12]
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
```
Also, when I play without the --loop flag and go to some timestamp:
```
mpv cat.mp4 
● Video  --vid=1  (h264 568x480 25 fps) [default]
● Audio  --aid=1  (aac 2ch 44100 Hz 237 kbps) [default]
Using hardware decoding (vaapi).
AO: [pipewire] 44100Hz stereo 2ch floatp
VO: [gpu-next] 568x480 vaapi[nv12]
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[... 22 lines cut ...]
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] h264: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] h264: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
Attempting next decoding method after failure of h264-vaapi.
```
This also happens when I try to watch Twitch VODs in Firefox.

elFarto, 2024-12-10:
Can you run mpv again with `NVD_LOG=1` set?

Kam1k4dze, 2024-12-10:
```
arch% NVD_LOG=1 mpv --loop Downloads/ieM4Vz7Z.mp4
● Video  --vid=1  (h264 498x498 14.2857 fps) [default]
      3770.201644889 [7024-7036] ../src/vabackend.c:2185       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
      3770.201653318 [7024-7036] ../src/vabackend.c:2193       __vaDriverInit_1_0 Got DRM FD: 1 28
      3770.201659063 [7024-7036] ../src/vabackend.c:2205       __vaDriverInit_1_0 Now have 0 (0 max) instances
      3770.201661844 [7024-7036] ../src/vabackend.c:2232       __vaDriverInit_1_0 Selecting Direct backend
      3770.241186862 [7024-7036] ../src/direct/nv-driver.c: 292            init_nvdriver Initing nvdriver...
      3770.241213878 [7024-7036] ../src/direct/nv-driver.c: 310            init_nvdriver NVIDIA kernel driver version: 565.77, major version: 565, minor version: 77
      3770.241218689 [7024-7036] ../src/direct/nv-driver.c: 317            init_nvdriver Got dev info: 100 1 2 6
      3770.348738773 [7024-7036] ../src/vabackend.c:1441      nvQueryImageFormats In nvQueryImageFormats
      3770.464795330 [7024-7036] ../src/vabackend.c: 670           nvCreateConfig got profile: 0 with 0 attributes
      3770.464807884 [7024-7036] ../src/vabackend.c:1797 nvQuerySurfaceAttributes with 1 (8) (nil) 0
      3770.464811392 [7024-7036] ../src/vabackend.c:1797 nvQuerySurfaceAttributes with 1 (8) 0x72dcb3080500 8
      3770.470460210 [7024-7036] ../src/vabackend.c:1864 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4080, height: 16 - 4080
      3770.470477070 [7024-7036] ../src/vabackend.c: 953        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x72dcb307c1e0)
      3770.470482369 [7024-7036] ../src/vabackend.c:1535            nvDeriveImage In nvDeriveImage
      3770.470487154 [7024-7036] ../src/vabackend.c: 953        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x72dcb307ccf0)
      3770.470490393 [7024-7036] ../src/vabackend.c:1535            nvDeriveImage In nvDeriveImage
      3770.470503399 [7024-7036] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x72dcb3081c20 128x128
      3770.477836478 [7024-7036] ../src/vabackend.c: 987        nvDestroySurfaces Destroying surface -1 (0x72dcb307ccf0)
      3770.477970475 [7024-7036] ../src/vabackend.c: 953        nvCreateSurfaces2 Creating surface 128x128, format 100 (0x72dcb307efa0)
      3770.477977432 [7024-7036] ../src/vabackend.c:1535            nvDeriveImage In nvDeriveImage
      3770.477986705 [7024-7036] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x72dcb307ee80 128x128
      3770.479401920 [7024-7036] ../src/vabackend.c: 987        nvDestroySurfaces Destroying surface -1 (0x72dcb307efa0)
      3770.479510233 [7024-7036] ../src/vabackend.c: 953        nvCreateSurfaces2 Creating surface 128x128, format 1000 (0x72dcb307cca0)
      3770.479516256 [7024-7036] ../src/vabackend.c:1535            nvDeriveImage In nvDeriveImage
      3770.479523948 [7024-7036] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x72dcb307e1d0 128x128
      3770.480930045 [7024-7036] ../src/vabackend.c: 987        nvDestroySurfaces Destroying surface -1 (0x72dcb307cca0)
      3770.481042108 [7024-7036] ../src/vabackend.c: 953        nvCreateSurfaces2 Creating surface 128x128, format 4 (0x72dcb307cc40)
      3770.481047828 [7024-7036] ../src/vabackend.c:1535            nvDeriveImage In nvDeriveImage
      3770.481055565 [7024-7036] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x72dcb307ef60 128x128
      3770.490328871 [7024-7036] ../src/vabackend.c: 987        nvDestroySurfaces Destroying surface -1 (0x72dcb307cc40)
      3770.490493835 [7024-7036] ../src/vabackend.c: 953        nvCreateSurfaces2 Creating surface 2x2, format 1 (0x72dcb307e200)
      3770.490500247 [7024-7036] ../src/vabackend.c:1535            nvDeriveImage In nvDeriveImage
      3770.490511053 [7024-7036] ../src/vabackend.c:1477            nvCreateImage created image id: 8
      3770.490516926 [7024-7036] ../src/vabackend.c:1650               nvPutImage In nvPutImage
      3770.490522322 [7024-7036] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x72dcb307e580 2x2
      3770.492115776 [7024-7036] ../src/vabackend.c: 987        nvDestroySurfaces Destroying surface -1 (0x72dcb307e200)
      3770.492226237 [7024-7036] ../src/vabackend.c: 953        nvCreateSurfaces2 Creating surface 2x2, format 1 (0x72dcb2d73aa0)
      3770.492232072 [7024-7036] ../src/vabackend.c:1535            nvDeriveImage In nvDeriveImage
      3770.492240465 [7024-7036] ../src/vabackend.c:1477            nvCreateImage created image id: 11
      3770.492245729 [7024-7036] ../src/vabackend.c:1650               nvPutImage In nvPutImage
      3770.492251085 [7024-7036] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x72dcb000cac0 2x2
      3770.493690716 [7024-7036] ../src/vabackend.c: 987        nvDestroySurfaces Destroying surface -1 (0x72dcb2d73aa0)
      3770.493805390 [7024-7036] ../src/vabackend.c: 953        nvCreateSurfaces2 Creating surface 2x2, format 1 (0x72dcb3081340)
      3770.493811169 [7024-7036] ../src/vabackend.c:1535            nvDeriveImage In nvDeriveImage
      3770.493819968 [7024-7036] ../src/vabackend.c:1477            nvCreateImage created image id: 14
[... cut ...]
      3776.809302634 [7024-7086] ../src/vabackend.c: 412          resolveSurfaces [RT] Resolve thread for 0x6290f3d17240 started
(...) V: 00:00:00 / 00:00:02 (0%)      3776.809794782 [7024-7086] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x72dc64002ca0 512x512
      3776.811700780 [7024-7086] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x72dc64009c60 512x512
(Paused) V: 00:00:00 / 00:00:02 (0%)
```

elFarto, 2024-12-11:
I can't seem to reproduce this, either with the linked video in the initial post or any of my test clips. Could you upload a file somewhere that causes the issue for you?

Kam1k4dze, 2024-12-11:
https://drive.google.com/drive/folders/1YHhL5Vve5rcoEhi-Bu7W_IxW8ubyYwpo?usp=sharing
This only happens with H.264 files, and not consistently. good.mp4 works about 90% of the time, but sometimes fails, and I'm not sure why. After re-encoding it as bad.mp4, I thought it would always fail, but it behaves similarly to good.mp4-except it fails about 90% of the time and works only 10% of the time.

elFarto, 2024-12-11:
I can't get any decode errors using that bad.mp4 file. I can get some very bad visual artefacts but it never errors. This might just be due to our different GPUs (your 3060 vs my 1060) treating bad data differently.

I'll keep poking around and see if I can find anything.

elFarto, 2024-12-12:
I'm running 565.77.

thesword53, 2024-12-12:
I'm also using the 565.77 proprietary driver without GSP.
I can reproduce the error on both GTX 1060 Mobile and RTX 2080 SUPER.

Kam1k4dze, 2024-12-12:
`MOZ_LOG="PlatformDecoderModule:5 Dmabuf:5" NVD_LOG=1 firefox "https://leonardo.osnova.io/be20671e-af21-5324-b36a-78597022c012/-/format/mp4/" &> logs.txt`
[logs.txt](https://github.com/user-attachments/files/18117603/logs.txt)

elFarto, 2024-12-13:
Unfortunately there's nothing in the log file that suggests where the error is. You could try raising this Firefox bug, and see if they have more insight into what could cause the issue. Given that the video decodes fine on my machine, it's likely related to how Firefox/FFMPEG handle seeking and some issue there is causing invalid data to be sent to VA-API.

thesword53, 2024-12-13:
I also have errors on H.264 videos when seeking in MPV with arrow keys

thesword53, 2025-01-09:
I found the cause of the issue. After the second loop `ffmpeg` creates 2 additional surfaces with `vaCreateSurfaces` after `vaCreateContext`, so the decode error is due to picture index overflow.

closed 2025-01-18 by Kam1k4dze