elFarto/nvidia-vaapi-driver #344: more small vid chroma error regions even without single plane

rakslice, 2024-12-26:
Just found another video that is still giving a similar looking missing chroma region with current master 9c4a3c70b68467343262d295ce56a56d510ec342:
https://t.gifs.bsky.app/qgTXJF6Bgz0AAAP3/jeff-the-land-shark-marvel-rivals.webm
(vp8, 214x172)

[image]

_Originally posted by @rakslice in https://github.com/elFarto/nvidia-vaapi-driver/issues/340#issuecomment-2556079238_

rakslice, 2024-12-26:
Another:
https://t.gifs.bsky.app/C90CNOkocIIAAAP3/serious-seriously.webm

Also height 172 (-> chroma 86)

rakslice, 2024-12-26:
Using `log2GobsPerBlockY` 3 for these works.

ToadKing, 2024-12-29:
Another example if it helps: https://t.gifs.bsky.app/VKre7tEUsssAAAP3/my-cousin-vinny-shocking.webm

PR #345 (2024-12-31): dial in the log2GobsPerBlockY threshold a bit more
Plane height 86 appears to need `log2GobsPerBlockY` 4.

Fixes #344.
closed 2024-12-31 by elFarto