elFarto/nvidia-vaapi-driver #346: nvidia-smi and nvtop shows different results. is libva-nvidia-driver truly working?

Fxzzi, 2024-12-29:
as shown by `nvtop` here: 
[image]

there seems to be no decoder usage by any program. Even those which are called N/A (I think the compute N/A is librewolf)

Under `nvidia-smi` it seems to be a similar story:
[image]

no compute process is shown here, which is strange.

Under `nvidia-smi pmon` there also seems to be no decoder usage shown:
[image]

However, under `nvidia-smi dmon` there is actually decoder usage, does this mean that the driver is working?
[image]

elFarto, 2024-12-31:
That's certainly odd. I would say no there's no hardware video decode happening, at least not in librewolf or electron. But as to what is causing the GPU to report that there is decode occurring I have no idea.

thesword53, 2024-12-31:
Hardware decoding is working:
[image]
[image]

Fxzzi, 2024-12-31:
> Hardware decoding is working: [...]

yeah, you're right. That 9% is the same shown in `nvidia-smi pmon`. And it does go to 0% when the video is paused.

Just a little strange that both `nvidia-smi` and `nvtop` are failing to specify that librewolf is the C process

fxzzi, 2025-01-20:
I will close this since it does seem like it is making a change to decode usage.

closed 2025-01-20 by fxzzi

fxzzi, 2025-01-24:
Yeah, looks like 570 drivers fixed the monitoring issue. Nvidia smi and nvtop both correctly show the compute process now:

[image]

[image]