elFarto/nvidia-vaapi-driver #349: Odd heights

rakslice, 2025-01-13:
I don't know how much this really matters, but for posterity:

When the height of the video is an odd number, we get some lines chroma error at the top, and a single line at the bottom.

[image]

Also for some odd heights but not others it seems like the bottom line is truncated in the pre-scaling video (this is easier to see at lower resolutions with something like this caption with a gap underneath at the bottom):

[image]

I suspect playing back videos where the resolution is not a multiple of the subsampling is probably a bad time generally because of all the instances in the code of stuff like:
`height >> p[i].ss.y` 
and
`((width * height) >> (p[i].ss.x + p[i].ss.y))`
i.e. right shifting by the subsampling bits, in other words rounding down by the subsampling multiple, when you need to round up -- the actual plane obviously has to have another line to cover the last less-than-a-full-subsample's-worth lines of the overall image.

But just fixing instances of those that I could find isn't enough on its own. There's something that needs the extra multiple of two lines in the output image, I guess. The hack of just doing e.g.:
```
    height = ROUND_UP(height, 2);
```
at the top of `nvCreateSurface2()` and 
```
    picture_height = ROUND_UP(picture_height, 2);
```
at the top of `nvCreateContext()` does make all the problems go away, at least testing from `mpv`.

elFarto, 2025-01-13:
I think that would be fine to add those `ROUND_UP`s in. It has no effect on 'normal' evenly sized videos so it shouldn't be an issue.

thesword53, 2025-01-14:
The issue is in `CUVIDDECODECREATEINFO` where `ulTargetWidth` and `ulTargetHeight` should be multiple of 2. In `ffmpeg` `ulTargetWidth` and `ulTargetHeight` are rounded up to 2.

`cuviddec.c:167` FFmpeg n7.1
```c
    // target width/height need to be multiples of two
    cuinfo.ulTargetWidth = avctx->width = (avctx->width + 1) & ~1;
    cuinfo.ulTargetHeight = avctx->height = (avctx->height + 1) & ~1;
```

rakslice, 2025-01-17:
I mean, sure, there are lots of different places you can put a hacky work around, but like you have code supporting planes of arbitrary subsampling values (well arbitrary powers of 2)... is there a reason for that or what

rakslice, 2025-01-18:
In fact rounding up the `.ulTargetHeight` and `.ulTargetWidth` doesn't make any difference.
Only rounding up the sizes in the original `nvCreateSurfaces2` makes the chroma problems go away, and rounding up the other size params in `CUVIDDECODECREATEINFO`, the ones for `.display_area`, fix the line truncation for some sizes.

PR #352 (2025-01-18): Round up surface sizes and display area for subsampled chroma videos
Fixes #349
closed 2025-01-18 by elFarto