elFarto/nvidia-vaapi-driver #353: Av1 flickering and green

ManuLinares, 2025-02-08:
#### System Archlinux: (Nvidia RTX 3070)

**Packages:**

- nvidia-dkms 570.86.16-2
- cuda 12.8.0-1
- linux 6.13.1.arch1-1
- libva-nvidia-driver-git 0.0.13.r17.gc519e97-1

---

#### To reproduce:

1. Open this video on firefox. https://ftp.belnet.be/mirror/FOSDEM/video/2025/k3401/fosdem-2025-4694-scaling-to-12k-live-streams.av1.webm
2. Green flickering

---

#### Video info (ffprobe):
```
  Duration: 00:36:13.76, start: -0.007000, bitrate: 391 kb/s
  Stream #0:0: Video: av1 (libdav1d) (Main), yuv420p(tv, progressive), 2560x1440, SAR 1:1 DAR 16:9, 30 fps, 30 tbr, 1k tbn
      Metadata:
        ENCODER         : Lavc59.37.100 libsvtav1
        DURATION        : 00:36:13.740000000
  Stream #0:1: Audio: opus, 48000 Hz, mono, fltp
      Metadata:
        ENCODER         : Lavc59.37.100 libopus
        DURATION        : 00:36:13.757000000
```

elFarto, 2025-02-14:
AV1 issue are difficult for me to debug as my 1060 doesn't support it. Can you post the output of firefox with `NVD_LOG=1` set?

ManuLinares, 2025-02-14:
Of course

<details>
<summary>Firefox Log</summary>

```
~ $ NVD_LOG=1 firefox https://ftp.belnet.be/mirror/FOSDEM/video/2025/k3401/fosdem-2025-4694-scaling-to-12k-live-streams.av1.webm
    139337.647662137 [1121658-1121692] ../src/vabackend.c:2239       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
    139337.647670507 [1121658-1121692] ../src/vabackend.c:2247       __vaDriverInit_1_0 Got DRM FD: 1 23
    139337.647676962 [1121658-1121692] ../src/vabackend.c:2259       __vaDriverInit_1_0 Now have 0 (20 max) instances
    139337.647679836 [1121658-1121692] ../src/vabackend.c:2286       __vaDriverInit_1_0 Selecting Direct backend
    139337.655333184 [1121658-1121692] ../src/direct/nv-driver.c: 292            init_nvdriver Initing nvdriver...
    139337.655354427 [1121658-1121692] ../src/direct/nv-driver.c: 310            init_nvdriver NVIDIA kernel driver version: 570.86.16, major version: 570, minor version: 86
    139337.655358118 [1121658-1121692] ../src/direct/nv-driver.c: 317            init_nvdriver Got dev info: 100 1 2 6
    139337.702027395 [1121658-1121692] ../src/vabackend.c:1495      nvQueryImageFormats In nvQueryImageFormats
    139337.732194220 [1121658-1121692] ../src/vabackend.c: 682           nvCreateConfig got profile: 6 with 0 attributes
    139337.732204912 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 4 (8) (nil) 0
    139337.732207067 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 4 (8) 0x7f45c11c14c0 8
    139337.733657950 [1121658-1121692] ../src/vabackend.c:1918 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
    139337.733670912 [1121658-1121692] ../src/vabackend.c: 682           nvCreateConfig got profile: 7 with 0 attributes
    139337.733673589 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 4 (8) (nil) 0
    139337.733675027 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 4 (8) 0x7f45c11c1d00 8
    139337.735135943 [1121658-1121692] ../src/vabackend.c:1918 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
    139337.735143147 [1121658-1121692] ../src/vabackend.c: 682           nvCreateConfig got profile: 13 with 0 attributes
    139337.735145713 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 4 (8) (nil) 0
    139337.735147121 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 4 (8) 0x7f45c11c14c0 8
    139337.736463577 [1121658-1121692] ../src/vabackend.c:1918 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
    139337.736470295 [1121658-1121692] ../src/vabackend.c: 682           nvCreateConfig got profile: 14 with 0 attributes
    139337.736472605 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 9 (8) (nil) 0
    139337.736474032 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 9 (8) 0x7f45c11c1700 8
    139337.737931428 [1121658-1121692] ../src/vabackend.c:1918 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
    139337.737938542 [1121658-1121692] ../src/vabackend.c: 682           nvCreateConfig got profile: 19 with 0 attributes
    139337.737940815 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 10 (8) (nil) 0
    139337.737942221 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 10 (8) 0x7f45c11c14c0 8
    139337.739269363 [1121658-1121692] ../src/vabackend.c:1918 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
    139337.739276414 [1121658-1121692] ../src/vabackend.c: 682           nvCreateConfig got profile: 32 with 0 attributes
    139337.739278220 [1121658-1121692] ../src/vabackend.c: 745           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
    139337.739280403 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 11 (8) (nil) 0
    139337.739281868 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 11 (8) 0x7f45c11c1f40 8
    139337.740600250 [1121658-1121692] ../src/vabackend.c:1918 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
    139337.740606945 [1121658-1121692] ../src/vabackend.c: 682           nvCreateConfig got profile: 21 with 0 attributes
    139337.740609219 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 10 (10) (nil) 0
    139337.740610675 [1121658-1121692] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 10 (10) 0x7f45c11c1700 8
    139337.741932546 [1121658-1121692] ../src/vabackend.c:1918 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
    139337.761953878 [1121658-1122131] ../src/vabackend.c: 682           nvCreateConfig got profile: 32 with 0 attributes
    139337.761961998 [1121658-1122131] ../src/vabackend.c: 745           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
    139337.761966099 [1121658-1122131] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 11 (8) (nil) 0
    139337.761967890 [1121658-1122131] ../src/vabackend.c:1851 nvQuerySurfaceAttributes with 11 (8) 0x7f45c11c1640 8
    139337.762873657 [1121658-1122131] ../src/vabackend.c:1918 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
[... cut ...]
    140220.531859122 [1134151-1134162] ../src/vabackend.c:1013        nvDestroySurfaces Destroying surface 6 (0x5a5b746bfa60)
    140220.532010609 [1134151-1134162] ../src/vabackend.c:2147              nvTerminate Terminating 0x7652645f5cd0
    140220.532085164 [1134151-1134162] ../src/vabackend.c:2161              nvTerminate Now have 0 (20 max) instances
[vo/gpu] flushing shader cache
[vo/gpu/x11] uninit ...

```

</details>

Screen recording of the mpv test for reference: 

https://github.com/user-attachments/assets/57405401-3808-44af-9f3c-5b532d524554

Anything more I can do to help debug this 👍

elFarto, 2025-02-15:
Interesting, it's playing the frames completely out of order, including surfaces that haven't had anything decoded to them (the green ones). I haven't seen that sort of issue since I fixed the surface reuse bug ages ago.

That's a very odd bug to have, and I'm not entirely sure why it would happen. We get passed in which surface to decode to and we decode the next frame to it. It's meant to be up to ffmpeg in this case to work out when to show each frame. This likely means that we're not interpreting something in the frame correctly, which will be a pain to debug.

I'll see if I find some time to debug this.

elFarto, 2025-03-01:
I've taken a look at it, but I can't seem to spot the problem. There are some differences, but nothing that seems to point to the issue.

ManuLinares, 2025-03-06:
> I've taken a look at it, but I can't seem to spot the problem. There are some differences, but nothing that seems to point to the issue.

I have an rtx 3070, would it help debugging this on my machine?
I can prepare a clean install for remote accessing my pc.

closed 2026-09-26 by elFarto