elFarto/nvidia-vaapi-driver #354: Firefox (135.0) errors when opening nvidia VA-API driver in Fedora 41.

linuxguy123, 2025-02-12:
Firefox (135.0) is throwing libva errors when I enable hardware decoding.

Setup:

```
$ kinfo
Operating System: Fedora Linux 41
KDE Plasma Version: 6.3.0
KDE Frameworks Version: 6.10.0
Qt Version: 6.8.2
Kernel Version: 6.12.10-200.fc41.x86_64 (64-bit)
Graphics Platform: X11
Processors: 24 × AMD Ryzen 9 5900X 12-Core Processor
Memory: 62.7 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1070
```
Relevant installed packages:

```
$dnf list firefox
Installed packages
firefox.x86_64 135.0-1.fc41 updates

libva-nvidia-driver.x86_64
xorg-x11-drv-nvidia.x86_64
xorg-x11-drv-nvidia-cuda.x86_64  
```

Testing libva from the command line:

```
$ vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.22.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri-nonfree/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
...
```
This is what happens when I run Firefox:

```
$firefox
...
libva info: VA-API version 1.22.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri-nonfree/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva error: /usr/lib64/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
- repeats over and over and whenever a webpage with a video is opened
```

libva is present where it should be:

```
$/ls /usr/lib64/dri/nvidia_drv_video.so
/usr/lib64/dri/nvidia_drv_video.so
```
The environmental var I have set for libva is: `LIBVA_DRIVER_NAME=nvidia`
It is stored in ~/.config/environment.d/LIBVA.conf

Some resources I have been using:

https://wiki.archlinux.org/title/Firefox <-- I'm running Fedora

https://fedoraproject.org/wiki/Firefox_Hardware_acceleration

https://github.com/elFarto/nvidia-vaapi-driver/

What am I missing ?

How do I get Firefox to open libva without erroring ?

elFarto, 2025-02-14:
Can you run firefox with `NVD_LOG=1` set, and paste the output here?

barmiro, 2025-03-17:
I'm getting the same with f41 on gnome, here's my NVD_LOG=1 output:

libva info: VA-API version 1.22.0
libva info: Trying to open /usr/lib64/dri-nonfree/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
     10948.353360593 [26850-26887] ../src/vabackend.c:2185       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
     10948.353368633 [26850-26887] ../src/vabackend.c:2193       __vaDriverInit_1_0 Got DRM FD: 1 26
     10948.353376401 [26850-26887] ../src/vabackend.c:2205       __vaDriverInit_1_0 Now have 25 (0 max) instances
libva error: /usr/lib64/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1

seems fine with NVD_LOG=1 vainfo:

Trying display: wayland
libva info: VA-API version 1.22.0
libva info: Trying to open /usr/lib64/dri-nonfree/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
     10912.957183331 [27151-27151] ../src/vabackend.c:2185       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
     10912.957187363 [27151-27151] ../src/vabackend.c:2193       __vaDriverInit_1_0 Got DRM FD: 1 4
     10912.957194801 [27151-27151] ../src/vabackend.c:2205       __vaDriverInit_1_0 Now have 0 (0 max) instances
     10912.957197172 [27151-27151] ../src/vabackend.c:2232       __vaDriverInit_1_0 Selecting Direct backend
     10912.977949916 [27151-27151] ../src/direct/nv-driver.c: 292            init_nvdriver Initing nvdriver...
     10912.977985015 [27151-27151] ../src/direct/nv-driver.c: 310            init_nvdriver NVIDIA kernel driver version: 570.124.04, major version: 570, minor version: 124
     10912.977989674 [27151-27151] ../src/direct/nv-driver.c: 317            init_nvdriver Got dev info: 100 1 0 fe
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
     10913.053694842 [27151-27151] ../src/vabackend.c:2093              nvTerminate Terminating 0x559375e27e70
     10913.053742858 [27151-27151] ../src/vabackend.c:2107              nvTerminate Now have 0 (0 max) instances

chenziliang0725, 2025-07-28:
> Firefox (135.0) is throwing libva errors when I enable hardware decoding.
> 
> Setup:
> 
> ```
> $ kinfo
> Operating System: Fedora Linux 41
> KDE Plasma Version: 6.3.0
> KDE Frameworks Version: 6.10.0
> Qt Version: 6.8.2
> Kernel Version: 6.12.10-200.fc41.x86_64 (64-bit)
> Graphics Platform: X11
> Processors: 24 × AMD Ryzen 9 5900X 12-Core Processor
> Memory: 62.7 GiB of RAM
> Graphics Processor: NVIDIA GeForce GTX 1070
> ```
> 
> Relevant installed packages:
> 
> ```
> $dnf list firefox
> Installed packages
> firefox.x86_64 135.0-1.fc41 updates
> 
> libva-nvidia-driver.x86_64
> xorg-x11-drv-nvidia.x86_64
> xorg-x11-drv-nvidia-cuda.x86_64  
> ```
> 
> Testing libva from the command line:
> 
> ```
> $ vainfo
> Trying display: wayland
> Trying display: x11
> libva info: VA-API version 1.22.0
> libva info: User environment variable requested driver 'nvidia'
> libva info: Trying to open /usr/lib64/dri-nonfree/nvidia_drv_video.so
> libva info: Trying to open /usr/lib64/dri-freeworld/nvidia_drv_video.so
> libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
> libva info: Found init function __vaDriverInit_1_0
> libva info: va_openDriver() returns 0
> ...
> ```
> 
> This is what happens when I run Firefox:
> 
> ```
> $firefox
> ...
> libva info: VA-API version 1.22.0
> libva info: User environment variable requested driver 'nvidia'
> libva info: Trying to open /usr/lib64/dri-nonfree/nvidia_drv_video.so
> libva info: Trying to open /usr/lib64/dri-freeworld/nvidia_drv_video.so
> libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
> libva info: Found init function __vaDriverInit_1_0
> libva error: /usr/lib64/dri/nvidia_drv_video.so init failed
> libva info: va_openDriver() returns 1
> - repeats over and over and whenever a webpage with a video is opened
> ```
> 
> libva is present where it should be:
> 
> ```
> $/ls /usr/lib64/dri/nvidia_drv_video.so
> /usr/lib64/dri/nvidia_drv_video.so
> ```
> 
> The environmental var I have set for libva is: `LIBVA_DRIVER_NAME=nvidia`我为 libva 设置的环境变量是：`LIBVA_DRIVER_NAME=nvidia` It is stored in ~/.config/environment.d/LIBVA.conf它存储在 ~/.config/environment.d/LIBVA.conf 中
> 
> Some resources I have been using:
> 
> https://wiki.archlinux.org/title/Firefox <-- I'm running Fedora
> 
> https://fedoraproject.org/wiki/Firefox_Hardware_acceleration
> 
> https://github.com/elFarto/nvidia-vaapi-driver/
> 
> What am I missing ?
> 
> How do I get Firefox to open libva without erroring ?

libva error: /usr/lib64/dri/nvidia_drv_video.so init failed
i get the same error, had you solved it? bro

closed 2026-09-26 by elFarto