elFarto/nvidia-vaapi-driver #358: Nvidia hardware acceleration not working on Ubuntu 24.10 and Firefox

cataluna84, 2025-03-26:
I can't make Nvidia GPU HW decoding work in Firefox

My system is:
```
## Hardware Information:
- **Hardware Model:**                              ASUSTeK COMPUTER INC. M5A99FX PRO R2.0
- **Memory:**                                      32.0 GiB
- **Processor:**                                   AMD FX™-8350 × 8
- **Graphics:**                                    NVIDIA GeForce RTX™ 2070
- **Disk Capacity:**                               11.0 TB

## Software Information:
- **Firmware Version:**                            2501
- **OS Name:**                                     Ubuntu 24.10
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               47
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.11.0-21-generic
- ** Nvidia Driver Version:**                      565.77 Open Kernel
```

I have succesfully built nvidia-vaapi-driver project and file `/usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so` is created.

I have set these enviroment variables in /etc/environment file:
```
export NVD_BACKEND=direct
export MOZ_DISABLE_RDD_SANDBOX=1
export LIBVA_DRIVER_NAME=nvidia
```

ffmpeg has vaapi in `ffmpeg -hwaccels` output.

After i've tried with no success the official Firefox(via Snap), i've installed Firefox Nightly using APT

## Firefox Nightly using APT

1. Download and install the repo key:
```
wget -q https://packages.mozilla.org/apt/repo-signing-key.gpg -O- | gpg --dearmor | sudo tee /etc/apt/keyrings/packages.mozilla.org.gpg > /dev/null
```
2. Add the apt repository by running the single command below:
```
echo "deb [signed-by=/etc/apt/keyrings/packages.mozilla.org.gpg] https://packages.mozilla.org/apt mozilla main" | sudo tee -a /etc/apt/sources.list.d/mozilla.list > /dev/null
```
This command will create the ‘mozilla.list‘ file under /etc/apt/sources.list.d directory and write the source line into it.
3. Install the Nightly build of Firefox from that repository by running command:
```
sudo apt update
sudo apt install firefox-nightly
```

## Configure Flags:

Open Firefox-nightly and go to url `about:config`
Set these flags:
```
media.ffmpeg.vaapi.enabled = true
media.rdd-ffmpeg.enabled = true
media.hardware-video-decoding.enabled = true
media.av1.enabled = true
gfx.webrender.all = true
gfx.x11-egl.force-enabled = false (i am using wayland)
widget.dmabuf.force-enabled = true
```

In /etc/environment:
```
export NVD_BACKEND=direct
export MOZ_DISABLE_RDD_SANDBOX=1
export LIBVA_DRIVER_NAME=nvidia
```

## Build the nvidia-vaapi-driver
- https://github.com/elFarto/nvidia-vaapi-driver

## Run Firefox Nightly
Start with these flags
```
$ LIBVA_MESSAGING_LEVEL=1 MOZ_ENABLE_WAYLAND=1 NVD_LOG=1 firefox-nightly
```

nvtop
[image]

[image]

[image]

[image]

[image]

[image]

cataluna84, 2025-04-02:
Thanks @j4reporting! 

`media.hardware-video-decoding.force-enabled` was indeed the config option for Firefox 137 and later

I guess @elFarto or some other contributor should update the `Readme.md` to reflect this change.

Hardware Decoding is supported on these codecs -
[image]

See the pink GPU0 decode% line which is at 20% playing 4k-60FPS YouTube VP9
[image]

ManuLinares, 2025-04-02:
`media.hardware-video-decoding.force-enabled = true` worked for me, thanks (I was going nuts)

nvidia-dkms 570.133.07-1
linux 6.13.8.arch1-1
firefox 137.0-1
libva-nvidia-driver-git 0.0.13.r17.gc519e97-1

Archlinux, Nvidia RTX 3070.

closed 2026-09-26 by elFarto