elFarto/nvidia-vaapi-driver #359: Firefox media decoder sometimes crashes since version 137

thesword53, 2025-04-02:
When watching videos on Twitch with Firefox 137, the content decoder process sometimes crashes with segmentation fault.

```
[ 1512.947882] NVRM: VM: invalid mmap
[ 1512.947902] NVRM: VM: invalid mmap
[ 1512.947932] NVRM: VM: invalid mmap
[ 1512.947957] NVRM: VM: invalid mmap
[ 1512.948067] NVRM: VM: invalid mmap
[ 1512.948113] NVRM: VM: invalid mmap
[ 1512.948728] NVRM: VM: invalid mmap
[ 1512.948999] NVRM: VM: invalid mmap
[ 1512.949097] NVRM: VM: invalid mmap
[ 1512.949122] NVRM: VM: invalid mmap
[ 1512.949147] NVRM: VM: invalid mmap
[ 1512.949207] NVRM: VM: invalid mmap
[ 1512.949246] NVRM: VM: invalid mmap
[ 1512.949267] NVRM: VM: invalid mmap
[ 1512.949324] NVRM: VM: invalid mmap
[ 1512.949410] NVRM: VM: invalid mmap
[ 1512.949480] NVRM: VM: invalid mmap
[ 1512.949497] NVRM: VM: invalid mmap
[ 1512.949716] NVRM: VM: invalid mmap
[ 1512.949734] NVRM: VM: invalid mmap
[ 1512.949839] NVRM: VM: invalid mmap
[ 1512.949853] NVRM: VM: invalid mmap
[ 1512.950112] NVRM: VM: invalid mmap
[ 1512.950124] NVRM: VM: invalid mmap
[ 1512.950156] NVRM: VM: invalid mmap
[ 1512.950190] NVRM: VM: invalid mmap
[ 1512.950199] NVRM: VM: invalid mmap
[ 1512.950225] NVRM: VM: invalid mmap
[ 1512.950234] NVRM: VM: invalid mmap
[ 1512.950264] NVRM: VM: invalid mmap
[ 1512.950335] NVRM: VM: invalid mmap
[ 1512.950473] NVRM: VM: invalid mmap
[ 1512.950847] NVRM: VM: invalid mmap
[ 1512.950967] NVRM: VM: invalid mmap
[ 1512.951026] NVRM: VM: invalid mmap
[ 1512.951691] NVRM: VM: invalid mmap
[ 1512.951849] NVRM: VM: invalid mmap
[ 1512.951887] NVRM: VM: invalid mmap
[ 1512.958611] MediaPD~oder #4[20052]: segfault at 0 ip 000063e9e1074ed3 sp 00007010a3dfddc0 error 6 in firefox[40ed3,63e9e1056000+9f000] likely on CPU 0 (core 0, socket 0)
[ 1512.958620] Code: 89 05 21 59 08 00 31 c0 b9 24 12 00 00 48 89 08 e8 e1 dc 04 00 48 8d 05 11 a3 fc ff 48 89 05 04 59 08 00 31 c0 b9 b5 0b 00 00 <48> 89 08 e8 c4 dc 04 00 41 83 e1 01 4d 89 4b 08 48 8d 05 47 c4 fc
```

```
#0  0x000063e9e1074ed3 in MOZ_CrashSequence (aAddress=0x0, aLine=2997) at /usr/src/debug/firefox/firefox-137.0/obj/dist/include/mozilla/Assertions.h:267
#1  arena_run_reg_dalloc (run=0x7010c53c9000, bin=0x7010d9600900, ptr=<optimized out>, size=3328) at /usr/src/debug/firefox/firefox-137.0/memory/build/mozjemalloc.cpp:2996
#2  arena_t::DallocSmall (this=0x7010d9600000, aChunk=<optimized out>, aPtr=<optimized out>, aMapElm=<optimized out>) at /usr/src/debug/firefox/firefox-137.0/memory/build/mozjemalloc.cpp:4572
#3  arena_dalloc (aPtr=<optimized out>, aOffset=<optimized out>, aArena=0x0) at /usr/src/debug/firefox/firefox-137.0/memory/build/mozjemalloc.cpp:4647
#4  BaseAllocator::free (aPtr=<optimized out>, this=<optimized out>) at /usr/src/debug/firefox/firefox-137.0/memory/build/mozjemalloc.cpp:5545
#5  MozJemalloc::free (arg1=<optimized out>) at /usr/src/debug/firefox/firefox-137.0/memory/build/malloc_decls.h:54
#6  PageFree (aArenaId=..., aPtr=<optimized out>) at /usr/src/debug/firefox/firefox-137.0/memory/build/PHC.cpp:1793
#7  MozJemallocPHC::free (aPtr=<optimized out>) at /usr/src/debug/firefox/firefox-137.0/memory/build/PHC.cpp:1796
#8  free (arg1=<optimized out>) at /usr/src/debug/firefox/firefox-137.0/memory/build/malloc_decls.h:54
#9  0x00007010c6a5c453 in ?? ()
#10 0x00007010a139e740 in ?? ()
#11 0x00007010d95575f0 in ?? ()
#12 0x00007010c8967024 in ?? ()
#13 0x00007010c9657be0 in ?? ()
#14 0x00007010a3dfdf10 in ?? ()
#15 0x00007010c893dd4b in ?? ()
#16 0x00007010d95575f0 in ?? ()
#17 0x8910156610273000 in ?? ()
#18 0x00007010a139e740 in ?? ()
#19 0x00077232d95575f0 in ?? ()
#20 0x0000701000000000 in ?? ()
#21 0x00000000000270b7 in ?? ()
#22 0x00007010a3dfdf50 in ?? ()
#23 0x00007010c8943362 in ?? ()
#24 0x0000000000000000 in ?? ()

```

lu0se, 2025-04-03:
this is my crash log
```
MOZ_DISABLE_RDD_SANDBOX=1 NVD_LOG=1 MOZ_LOG="FFmpegVideo:5" firefox
Gtk-Message: 21:27:11.566: Failed to load module "appmenu-gtk-module"
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG: FFmpegVideoDecoder::FFmpegVideoDecoder MIME video/avc Codec ID 27
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG: Initialising VA-API FFmpeg decoder
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:   codec h264 : H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
       534.709273462 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
       534.709288161 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:2196       __vaDriverInit_1_0 Now have 0 (0 max) instances
       534.709299509 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:2222       __vaDriverInit_1_0 Selecting Direct backend
       534.749826361 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/direct/nv-driver.c: 295            init_nvdriver Initing nvdriver...
       534.749856395 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/direct/nv-driver.c: 313            init_nvdriver NVIDIA kernel driver version: 570.133.07, major version: 570, minor version: 133
       534.749864893 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/direct/nv-driver.c: 320            init_nvdriver Got dev info: 100 1 2 6
       534.951332245 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1443      nvQueryImageFormats In nvQueryImageFormats
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG: FFmpegVideoDecoder::GetAcceleratedFormats()
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:   Profile H264Main:
       535.231468213 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 672           nvCreateConfig got profile: 6 with 0 attributes
       535.231481911 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) (nil) 0
       535.231488189 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) 0x7aab0c722c40 8
       535.238009533 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:     codec h264 format nv12              3         12
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:     codec h264 format p010le            3         15
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:     codec h264 format p012le            3         18
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:   Profile H264High:
       535.238047754 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 672           nvCreateConfig got profile: 7 with 0 attributes
       535.238057561 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) (nil) 0
       535.238063664 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) 0x7aab0c722c40 8
       535.245566758 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:     codec h264 format nv12              3         12
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:     codec h264 format p010le            3         15
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:     codec h264 format p012le            3         18
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:   Profile H264ConstrainedBaseline:
       535.245603424 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 672           nvCreateConfig got profile: 13 with 0 attributes
       535.245609822 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) (nil) 0
       535.245616044 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) 0x7aab0c722c40 8
       535.252294134 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:     codec h264 format nv12              3         12
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:     codec h264 format p010le            3         15
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:     codec h264 format p012le            3         18
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:   Profile HEVCMain:
       535.252329049 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 672           nvCreateConfig got profile: 17 with 0 attributes
       535.252335180 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 8 (8) (nil) 0
       535.252339173 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1799 nvQuerySurfaceAttributes with 8 (8) 0x7aab0c722c40 8
       535.259516214 [16344-16386] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 144 - 8192, height: 144 - 8192
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:     codec hevc format nv12              3         12
[RDD 16344: MediaSupervisor #1]: D/FFmpegVideo FFMPEG:     codec hevc format p010le            3         15
[... cut ...]
ExceptionHandler::GenerateDump attempting to generate:/home/lu0se/.mozilla/firefox/t9n87yi5.default-release/minidumps/269055a2-2259-0c01-e968-9109a5d63d47.dmp
ExceptionHandler::GenerateDump cloned child 16783
ExceptionHandler::WaitForContinueSignal waiting for continue signal...
ExceptionHandler::SendContinueSignalToChild sent continue signal to child
ExceptionHandler::GenerateDump minidump generation succeeded
Exiting due to channel error.
Exiting due to channel error.
Exiting due to channel error.
Exiting due to channel error.
Exiting due to channel error.
Exiting due to channel error.
```

thesword53, 2025-04-11:
I created the pull request https://github.com/elFarto/nvidia-vaapi-driver/pull/366. Maybe it can help.

thesword53, 2025-04-19:
The media decoder crashes with segmentation fault when playing 2 twitch live stream:

```
[ 9525.520239] NVRM: GPU at PCI:0000:01:00: GPU-38495d49-2ff6-7fc9-e4c4-a8b4550b1e01
[ 9525.520245] NVRM: Xid (PCI:0000:01:00): 31, pid=5429, name=MediaPD~oder #2, Ch 0000008c, intr 10000000. MMU Fault: ENGINE NVDEC HUBCLIENT_NVDEC faulted @ 0x1_097f0000. Fault is of type FAULT_PTE ACCESS_TYPE_READ
[ 9525.522611] NVRM: VM: invalid mmap
[ 9525.522763] NVRM: VM: invalid mmap
[ 9525.522818] NVRM: VM: invalid mmap
[ 9525.522924] NVRM: VM: invalid mmap
[ 9525.523313] NVRM: VM: invalid mmap
[ 9525.523338] NVRM: VM: invalid mmap
[ 9525.523355] NVRM: VM: invalid mmap
[ 9525.523377] NVRM: VM: invalid mmap
[ 9525.523548] NVRM: VM: invalid mmap
[ 9525.523610] NVRM: VM: invalid mmap
[ 9525.523646] NVRM: VM: invalid mmap
[ 9525.523682] NVRM: VM: invalid mmap
[ 9525.523714] NVRM: VM: invalid mmap
[ 9525.523742] NVRM: VM: invalid mmap
[ 9525.523820] NVRM: VM: invalid mmap
[ 9525.524092] NVRM: VM: invalid mmap
[ 9525.524107] NVRM: VM: invalid mmap
[ 9525.524146] NVRM: VM: invalid mmap
[ 9525.524155] NVRM: VM: invalid mmap
[ 9525.524183] NVRM: VM: invalid mmap
[ 9525.524191] NVRM: VM: invalid mmap
[ 9525.524199] NVRM: VM: invalid mmap
[ 9525.524223] NVRM: VM: invalid mmap
[ 9525.524232] NVRM: VM: invalid mmap
[ 9525.524255] NVRM: VM: invalid mmap
[ 9525.524264] NVRM: VM: invalid mmap
[ 9525.524397] NVRM: VM: invalid mmap
[ 9525.524433] NVRM: VM: invalid mmap
[ 9525.524504] NVRM: VM: invalid mmap
[ 9525.524631] NVRM: VM: invalid mmap
[ 9525.524764] NVRM: VM: invalid mmap
[ 9525.524889] NVRM: VM: invalid mmap
[ 9525.524952] NVRM: VM: invalid mmap
[ 9525.525664] NVRM: VM: invalid mmap
[ 9525.525730] NVRM: VM: invalid mmap
[ 9525.525776] NVRM: VM: invalid mmap
[ 9525.531699] MediaPD~oder #1[5823]: segfault at 74d049b0c5e5 ip 000074cfab4542cd sp 000074cf979aad48 error 4 in libc.so.6[16c2cd,74cfab30c000+171000] likely on CPU 6 (core 2, socket 0)
[ 9525.531709] Code: 00 00 00 00 00 66 66 2e 0f 1f 84 00 00 00 00 00 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 f3 0f 1e fa 48 89 f8 48 83 fa 20 72 23 <c5> fe 6f 06 48 83 fa 40 0f 87 a5 00 00 00 c5 fe 6f 4c 16 e0 c5 fe
```
```
[Child 5696, MediaDecoderStateMachine #1] WARNING: Decoder=784bfa3fea00 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005): file /usr/src/debug/firefox/firefox-137.0.2/dom/media/MediaDecoderStateMachineBase.cpp:168
[Child 5548, MediaDecoderStateMachine #1] WARNING: Decoder=7ba8e3e5b400 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005): file /usr/src/debug/firefox/firefox-137.0.2/dom/media/MediaDecoderStateMachineBase.cpp:168
[GFX1-]: VideoBridgeParent receives IPC close with reason=AbnormalShutdown
```
Firefox 137.0.2
Nvidia 570.133.07

thesword53, 2025-04-19:
The commit https://github.com/elFarto/nvidia-vaapi-driver/commit/0ef97c1849fac3e07cdf39ffe4cd9aa9f352896f#diff-f727c889ef4325f1cf6fda5e7376d6c5c1275e503c9a00c0560a7a0171bc25a1R551 seems to fix the mmu fault issue, but I still have random decode errors on Twitch with `cuvidDecodePicture` failing with `CUDA_ERROR_INVALID_VALUE` (1). I'm currently trying to investigate why this is happening.

ManuLinares, 2025-04-20:
I found some segfaults on my log:

firefox 137.0.2-1
nvidia-dkms 570.133.07-1
linux 6.14.2.arch1-1

<details>
<summary>journal log</summary>

```bash
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
abr 20 20:12:49 thought kernel: NVRM: VM: invalid mmap
[... 364 lines cut ...]
                                                 #3  0x00007bcd46baac0e pthread_cond_wait (libc.so.6 + 0x94c0e)
                                                 #4  0x00007bcd36442faf n/a (libavcodec.so.61 + 0x642faf)
                                                 #5  0x00005c2d06570e83 n/a (/usr/lib/firefox/firefox + 0x4de83)
                                                 #6  0x00007bcd46bab70a n/a (libc.so.6 + 0x9570a)
                                                 #7  0x00007bcd46c2faac n/a (libc.so.6 + 0x119aac)

                                                 Stack trace of thread 13933:
                                                 #0  0x00007bcd46bb3be2 n/a (libc.so.6 + 0x9dbe2)
                                                 #1  0x00007bcd46ba7e33 n/a (libc.so.6 + 0x91e33)
                                                 #2  0x00007bcd46ba84bc n/a (libc.so.6 + 0x924bc)
                                                 #3  0x00007bcd46baae11 pthread_cond_timedwait (libc.so.6 + 0x94e11)
                                                 #4  0x00005c2d065894d5 _ZN7mozilla6detail21ConditionVariableImpl8wait_forERNS0_9MutexImplERKNS_16BaseTimeDurationINS_27TimeDurationValueCalculatorEEE (/usr/lib/firefox/firefox + 0x664d5)
                                                 #5  0x00007bcd3a513f3a n/a (libxul.so + 0x2d13f3a)
                                                 #6  0x00007bcd3a48cc83 n/a (libxul.so + 0x2c8cc83)
                                                 #7  0x00007bcd3a49200a n/a (libxul.so + 0x2c9200a)
                                                 #8  0x00007bcd3a4919e6 n/a (libxul.so + 0x2c919e6)
                                                 #9  0x00007bcd470e968f n/a (libnspr4.so + 0x2a68f)
                                                 #10 0x00005c2d06570e83 n/a (/usr/lib/firefox/firefox + 0x4de83)
                                                 #11 0x00007bcd46bab70a n/a (libc.so.6 + 0x9570a)
                                                 #12 0x00007bcd46c2faac n/a (libc.so.6 + 0x119aac)
                                                 ELF object binary architecture: AMD x86-64
abr 20 20:12:50 thought systemd[1]: systemd-coredump@2-15089-0.service: Deactivated successfully.
abr 20 20:12:50 thought systemd[1]: systemd-coredump@2-15089-0.service: Consumed 585ms CPU time, 436.2M memory peak.
```
</details>

thesword53, 2025-05-01:
> I found some segfaults on my log:
> 
> firefox 137.0.2-1 nvidia-dkms 570.133.07-1 linux 6.14.2.arch1-1
> journal log

In your can i seems to crash here https://github.com/elFarto/nvidia-vaapi-driver/blob/c2860cc621ae1ef515c003d43315c63a41529ff4/src/vabackend.c#L1609

In my tests, it also crashes in `nvRenderPicture`. I think it's an issue caused by invalid memory access from multiple threads.

thesword53, 2025-05-03:
I modified https://github.com/elFarto/nvidia-vaapi-driver/pull/366 and it should solve crashes

ManuLinares, 2025-05-03:
> I modified [#366](https://github.com/elFarto/nvidia-vaapi-driver/pull/366) and it should solve crashes

with patch #366 there are still crashes.

Firefox NVD_LOG=1 -> [firefox_nvd.log](https://github.com/user-attachments/files/20024277/firefox_nvd.log)

<details><summary>Crashdump log</summary>

```
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: NVRM: VM: invalid mmap
may 03 13:58:12 thought kernel: traps: MediaPD~oder #1[1228016] general protection fault ip:7cbf76b7db41 sp:7cbf7c75a080 error:0 in libnvcuvid.so.570.144[17db41,7cbf76a00000+f71000]
may 03 13:58:12 thought systemd-coredump[1228314]: Process 1227030 (RDD Process) of user 1000 terminated abnormally with signal 11/SEGV, processing...
may 03 13:58:12 thought systemd[1]: Started Process Core Dump (PID 1228314/UID 0).
may 03 13:58:13 thought systemd-coredump[1228315]: [🡕] Process 1227030 (RDD Process) of user 1000 dumped core.

                                                   Stack trace of thread 1228016:
                                                   #0  0x00007cbf76b7db41 n/a (libnvcuvid.so.1 + 0x17db41)
                                                   #1  0x00007cbf76b7f98e n/a (libnvcuvid.so.1 + 0x17f98e)
                                                   #2  0x00007cbf76a47468 n/a (libnvcuvid.so.1 + 0x47468)
                                                   #3  0x00007cbf76a2874b n/a (libnvcuvid.so.1 + 0x2874b)
                                                   #4  0x00007cbf76a2f96e n/a (libnvcuvid.so.1 + 0x2f96e)
                                                   #5  0x00007cbf76a2f26f n/a (libnvcuvid.so.1 + 0x2f26f)
                                                   #6  0x00007cbf967e68dc n/a (nvidia_drv_video.so + 0xc8dc)
                                                   #7  0x00007cbf967e6a69 n/a (nvidia_drv_video.so + 0xca69)
                                                   #8  0x00007cbf8b173fbb vaDestroyContext (libva.so.2 + 0xafbb)
                                                   #9  0x00007cbf8b95a75d n/a (libavcodec.so.61 + 0x75a75d)
                                                   #10 0x00007cbf8b45baba n/a (libavcodec.so.61 + 0x25baba)
                                                   #11 0x00007cbf8b560d38 n/a (libavcodec.so.61 + 0x360d38)
                                                   #12 0x00007cbf8b56a205 n/a (libavcodec.so.61 + 0x36a205)
                                                   #13 0x00007cbf8b5822b6 n/a (libavcodec.so.61 + 0x3822b6)
                                                   #14 0x00007cbf8b585706 n/a (libavcodec.so.61 + 0x385706)
                                                   #15 0x00007cbf8b45eaac n/a (libavcodec.so.61 + 0x25eaac)
                                                   #16 0x00007cbf8b45f305 n/a (libavcodec.so.61 + 0x25f305)
                                                   #17 0x00007cbf8b45f97c avcodec_send_packet (libavcodec.so.61 + 0x25f97c)
                                                   #18 0x00007cbf92a89506 n/a (libxul.so + 0x5e89506)
                                                   #19 0x00007cbf92a3bb46 n/a (libxul.so + 0x5e3bb46)
                                                   #20 0x00007cbf92a81fd9 n/a (libxul.so + 0x5e81fd9)
                                                   #21 0x00007cbf929f375f n/a (libxul.so + 0x5df375f)
                                                   #22 0x00007cbf8f93b109 n/a (libxul.so + 0x2d3b109)
                                                   #23 0x00007cbf8f936a32 n/a (libxul.so + 0x2d36a32)
[... 242 lines cut ...]

                                                   Stack trace of thread 1228012:
                                                   #0  0x00007cbf9c2ade22 n/a (libc.so.6 + 0x9de22)
                                                   #1  0x00007cbf9c2a1fda n/a (libc.so.6 + 0x91fda)
[... cut ...]
                                                   #9  0x00007cbf9c32918c n/a (libc.so.6 + 0x11918c)
                                                   ELF object binary architecture: AMD x86-64
may 03 13:58:13 thought systemd[1]: systemd-coredump@6-1228314-0.service: Deactivated successfully.
may 03 13:58:13 thought systemd[1]: systemd-coredump@6-1228314-0.service: Consumed 346ms CPU time, 243.9M memory peak.
```

</details>

---

Archlinux
libva-nvidia-driver-git with #366 patch
firefox 138.0.1-1
nvidia-dkms 570.144-3

thesword53, 2025-05-03:
I am unable to reproduce this crash with the latest patch. Do you have an example of video which crashes Firefox ?

ManuLinares, 2025-05-03:
> I am unable to reproduce this crash with the latest patch. Do you have an example of video which crashes Firefox ?

Can reproduce when opening several tabs of the site 

[image]

thesword53, 2025-05-09:
Hello @ManuLinares,

Do you still have crashes after applying the following patch on top of `master` ?
```diff
diff --git a/src/vabackend.c b/src/vabackend.c
index 8d70e43..a138853 100644
--- a/src/vabackend.c
+++ b/src/vabackend.c
@@ -316,8 +316,6 @@ static void deleteObject(NVDriver *drv, VAGenericID id) {
 }
 
 static bool destroyContext(NVDriver *drv, NVContext *nvCtx) {
-    CHECK_CUDA_RESULT_RETURN(cu->cuCtxPushCurrent(drv->cudaContext), false);
-
     LOG("Signaling resolve thread to exit");
     struct timespec timeout;
     clock_gettime(CLOCK_REALTIME, &timeout);
@@ -333,14 +331,15 @@ static bool destroyContext(NVDriver *drv, NVContext *nvCtx) {
 
     bool successful = true;
     if (nvCtx->decoder != NULL) {
+      CHECK_CUDA_RESULT_RETURN(cu->cuCtxPushCurrent(drv->cudaContext), false);
       CUresult result = cv->cuvidDestroyDecoder(nvCtx->decoder);
+      CHECK_CUDA_RESULT_RETURN(cu->cuCtxPopCurrent(NULL), false);
       if (result != CUDA_SUCCESS) {
           LOG("cuvidDestroyDecoder failed: %d", result);
           successful = false;
       }
     }
     nvCtx->decoder = NULL;
-    CHECK_CUDA_RESULT_RETURN(cu->cuCtxPopCurrent(NULL), false);
 
     return successful;
 }

```

ManuLinares, 2025-05-09:
> 
> Do you still have crashes after applying the following patch on top of `master` ?
> 

Yes. Can reproduce when opening several tabs of the same site, switching tabs. Here are the logs

[journalctl_crash.log](https://github.com/user-attachments/files/20127688/journalctl_crash.log)
[NVD_LOG_firefox.log](https://github.com/user-attachments/files/20127689/NVD_LOG_firefox.log)

thesword53, 2025-05-09:
Hello @ManuLinares,

I modified the diff in my previous comment. I added missing `cuCtxPushCurrent` and `cuCtxPopCurrent`. Can you try again ? If it still doesn't work, can also build and install the `-debug` package to have more information in stack trace ?

ManuLinares, 2025-05-09:
> I modified the diff in my previous comment. I added missing `cuCtxPushCurrent` and `cuCtxPopCurrent`. Can you try again ? If it still doesn't work, can also build and install the `-debug` package to have more information in stack trace ?

I captured two crashes.

[journalctl_crash.log](https://github.com/user-attachments/files/20128898/journalctl_crash.log)
[NVD_LOG_firefox.log](https://github.com/user-attachments/files/20128897/NVD_LOG_firefox.log)

`/usr/lib/dri/nvidia_drv_video.so: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=adfcbaf06f2037eb487102aebc2532f9ff774e64, with debug_info, not stripped`

thesword53, 2025-05-23:
The issue was fixed with https://github.com/elFarto/nvidia-vaapi-driver/commit/effa3af05da4525d850a525ae2dedf5eb80351d7